coolsnowwolf/lede #5746: 下载时报错

ramondelee, 2020-11-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：下载时报错

### (2) 路由器型号和固件版本

A：R2S

### (3) 详细日志

A：Wed Nov  4 14:27:26 2020 daemon.err nlbwmon[1533]: Netlink receive failure: Object busy
Wed Nov  4 14:27:26 2020 daemon.err nlbwmon[1533]: Unable to dump conntrack: I/O error
Wed Nov  4 14:54:29 2020 daemon.err nlbwmon[1533]: Netlink receive failure: Out of memory
Wed Nov  4 14:54:29 2020 daemon.err nlbwmon[1533]: Unable to dump conntrack: No buffer space available
Wed Nov  4 16:41:36 2020 daemon.err nlbwmon[1533]: Netlink receive failure: Out of memory
Wed Nov  4 16:41:36 2020 daemon.err nlbwmon[1533]: Unable to dump conntrack: No buffer space available
Wed Nov  4 18:26:42 2020 daemon.err nlbwmon[1533]: Netlink receive failure: Out of memory
Wed Nov  4 18:26:42 2020 daemon.err nlbwmon[1533]: Unable to dump conntrack: No buffer space available
Wed Nov  4 20:13:48 2020 daemon.err nlbwmon[1533]: Netlink receive failure: Out of memory
Wed Nov  4 20:13:48 2020 daemon.err nlbwmon[1533]: Unable to dump conntrack: No buffer space available

Dimitri2020007, 2020-11-04:
你的flash或者ram空间不够了。

ramondelee, 2020-11-05:
> 你的flash或者ram空间不够了。

这个是R2S，1G内存，闪存8G

martin-xj, 2020-11-06:
同样问题，经常会出现此类提示。

Fyred, 2020-11-14:
同样问题：
Fri Nov 13 07:38:56 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Out of memory
Fri Nov 13 07:38:56 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: No buffer space available
Fri Nov 13 12:33:24 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Out of memory
Fri Nov 13 12:33:24 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: No buffer space available
Fri Nov 13 12:42:56 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Object busy
Fri Nov 13 12:42:56 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: I/O error
Fri Nov 13 12:55:29 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Object busy
Fri Nov 13 12:55:29 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: I/O error
Fri Nov 13 13:22:34 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Object busy
Fri Nov 13 13:22:34 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: I/O error
Fri Nov 13 13:38:07 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Object busy
Fri Nov 13 13:38:07 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: I/O error
Fri Nov 13 18:47:13 2020 daemon.notice softethervpn-server[4305]: Writing vpn_server.config (AutoSaveConfigSpan set to < 86400 > seconds)
Fri Nov 13 20:05:23 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Out of memory
Fri Nov 13 20:05:23 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: No buffer space available
Fri Nov 13 21:19:08 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Object busy
Fri Nov 13 21:19:08 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: I/O error
Fri Nov 13 21:58:45 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Object busy
Fri Nov 13 21:58:45 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: I/O error
Sat Nov 14 02:00:00 2020 cron.err crond[2453]: USER root pid 14052 cmd /usr/share/shadowsocksr/ssrplusupdate.sh >>/tmp/ssrplus.log 2>&1
Sat Nov 14 02:08:34 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Out of memory
Sat Nov 14 02:08:34 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: No buffer space available
Sat Nov 14 03:02:45 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Object busy
Sat Nov 14 03:02:45 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: I/O error
Sat Nov 14 03:24:20 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Object busy
Sat Nov 14 03:24:20 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: I/O error
Sat Nov 14 03:46:54 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Object busy
Sat Nov 14 03:46:54 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: I/O error
Sat Nov 14 04:04:27 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Object busy
Sat Nov 14 04:04:27 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: I/O error
Sat Nov 14 06:00:00 2020 cron.err crond[2453]: USER root pid 16405 cmd /usr/share/adbyby/adblock.sh > /tmp/adupdate.log 2>&1
Sat Nov 14 06:00:24 2020 cron.err crond[16490]: crond (busybox 1.31.1) started, log level 8
Sat Nov 14 06:00:24 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Nov 14 06:00:24 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Nov 14 06:00:27 2020 daemon.info dnsmasq[16585]: Connected to system UBus
Sat Nov 14 06:00:28 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Nov 14 06:00:28 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Nov 14 06:00:31 2020 daemon.info dnsmasq[16754]: Connected to system UBus
Sat Nov 14 07:18:05 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Out of memory
Sat Nov 14 07:18:05 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: No buffer space available
Sat Nov 14 11:02:50 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Out of memory
Sat Nov 14 11:02:50 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: No buffer space available
Sat Nov 14 12:17:05 2020 daemon.err nlbwmon[3332]: Netlink receive failure: Out of memory
Sat Nov 14 12:17:05 2020 daemon.err nlbwmon[3332]: Unable to dump conntrack: No buffer space available

WYC-2020, 2021-02-23:
16G一样报错，初步看了下源代码,好像时nlbwmon在解析数据结构，获取数据长度，并且malloc内存空间的时候出错了，应该是数据结构解析后获取到了一个错误的长度，导致申请空间失败，具体是那个发送的数据库能力有限无法定位，等大佬修复

WYC-2020, 2021-02-23:
libnl-tiny 在这个库里面 的msg.c里面又解析数据获取长度，又能力的可以去干

closed 2021-06-20 by coolsnowwolf