coolsnowwolf/lede #5747: cifsd共享服务无法启动

keeno1982, 2020-11-04:
报错日志如下：
Wed Nov  4 22:51:28 2020 daemon.notice ksmbd: Stopping Ksmbd userspace service.
Wed Nov  4 22:51:28 2020 daemon.err modprobe: failed to find a module named ksmbd
Wed Nov  4 22:51:28 2020 daemon.err ksmbd: modprobe of ksmbd module failed, can\'t start ksmbd!

Phicomm N1
ARMv8 Processor rev 4 (v8l) x 4
固件版本 | OpenWrt  R20.10.20 / 			
LuCI Master (git-20.256.12360-1a54222)
内核版本 | 5.4.73-flippy-47+o

Dimitri2020007, 2020-11-04:
寻找ksmbd模块失败

modprobe加载ksmbd模块失败，无法启动ksmbd

keeno1982, 2020-11-05:
> 
> 
> 寻找ksmbd模块失败
> 
> modprobe加载ksmbd模块失败，无法启动ksmbd

你这只是翻译吧，如何解决？

Dimitri2020007, 2020-11-05:
进入后台搜索软件包ksmbd和cifsd

看看有没有ksmbd-server这个软件（一般你选了cifsd服务就会自动选上ksmbd-server软件的）

卸载了cifsd然后重装一下

keeno1982, 2020-11-05:
> 
> 
> 进入后台搜索软件包ksmbd和cifsd
> 
> 看看有没有ksmbd-server这个软件（一般你选了cifsd服务就会自动选上ksmbd-server软件的）
> 
> 卸载了cifsd然后重装一下

ksmbd开头的软件都装上了，不过启动还是报错。
然后我卸载，打算重新安装时提示

Installing kmod-fs-ksmbd (5.4.74+3.2.5-1) to root...
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/targets/armvirt/64/packages/kmod-fs-ksmbd_5.4.74%2b3.2.5-1_aarch64_generic.ipk

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fs-ksmbd:
 * 	kernel (= 5.4.74-1-3d48bdbbdbb94845a068812a2ad7d14e)
 * opkg_install_cmd: Cannot install package kmod-fs-ksmbd.

Dimitri2020007, 2020-11-05:
重新编译吧，不过记得在menuconfig>kernel modules>filesystem support下全部选择。
其中就有一个ksmbd-kmod依赖项目

hyird, 2020-11-06:
N1杂交的内核 得看 杂交内核是否支持， op roots的kmod 只能提供配置去激活这个功能

keeno1982, 2020-11-06:
> 
> 
> 重装的时候kmod-fs-ksmbd没装成功。
> 
> 重新编译吧，不过记得在menuconfig>kernel modules>filesystem下全部选择，放心的选没有任何冲突。flash空间不够就只选kmod-ksmbd和这个support for insecure SMB1/CIFS and SMB2.0 protocols
> 其中就有一个ksmbd-kmod依赖项目
> 
> （.config是这个）
> 
> CONFIG_PACKAGE_kmod-fs-ksmbd=y
> 
> （图形配置是这个，注意把support for insecure SMB1/CIFS and SMB2.0 protocols也选上）
> [image]

你说的这编译的内容我都有选上，我怀疑是内核的问题

keeno1982, 2020-11-06:
> 
> 
> N1杂交的内核 得看 杂交内核是否支持， op roots的kmod 只能提供配置去激活这个功能

可能你说的是对的，就是不知道怎么去激活你说的那东西

Dimitri2020007, 2020-11-06:
要是内核问题就有点复杂了，我编译的是x86_64版本，cifsd功能是正常的。

closed 2021-06-20 by coolsnowwolf