coolsnowwolf/lede #5751: Linksys  wrt1900acs  没有交换机界面，高级重启显示无效分区

hwlon, 2020-11-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
Linksys WRT1900ACS
### (1) 具体问题
A：没有交换机界面，
B：高级重启显示无效分区

### (2) 路由器型号和固件版本

A：Linksys WRT1900ACS  11月6日版

### (3) 详细日志

A：没有交换机界面，
B：高级重启显示无效分区

Dimitri2020007, 2020-11-07:
选择了base system下的bridge和swconfig吗？

第二个我是x86_64也会出现无效分区的问题，无解。

coolsnowwolf, 2020-11-08:
5.4 内核的 DSA 模型不再包含交换机

hwlon, 2020-11-09:
这个高级重启本身只支持Linksys 双分区路由器，不支持x86_64。
x86_64本身没有交换机，也不支持内置第二硬盘休眠，刚刚换回Linksys WRT1900ACS ，谁想竟然要用的这两个功能缺无效了。
昨天用老大5月份的源码编译先用着，谢谢您辛苦了。

edmondfr, 2020-11-28:
Linksys WRT1200AC 也是没办法进入双分区切换功能，以前的版本却都正常。

暂时使用下面的命令登陆进去切换分区，希望老大抽空解决一下。感谢。

查看当前分区：
fw_printenv boot_part

如果返回结果是 1，那么切换另一分区命令：
fw_setenv boot_part 2 && reboot

如果是 2，则：
fw_setenv boot_part 1 && reboot

teapethu, 2021-01-25:
Linksys WRT32X，同样的问题没法进入双分区切换功能。

teapethu, 2021-01-25:
> 5.4 内核的 DSA 模型不再包含交换机

这个。。。那需要在交换机上配置Vlan的用什么方法呢？VLAN用来设置IPTV
官方的19.07还是有交换机的。

AmadeusGhost, 2021-02-23:
luci 并入这个 commit 试试：https://github.com/AmadeusGhost/luci/commit/ab851a1c88a170d5938a4749c23f56c7ddeb6fba

CrBoom, 2021-03-15:
官方luci源的advanced reboot是正常工作的
ps:有无计划更新luci/插件啊

closed 2021-06-20 by coolsnowwolf