coolsnowwolf/lede #5756: 谁帮测试下我编译的r2s固件能用不

tl768, 2020-11-08:
用自己编译的固件启动异常。就一个网卡灯亮。也连接不上，不知道是卡刷的问题还是固件的问题。
[openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img.gz](https://github.com/coolsnowwolf/lede/files/5505884/openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img.gz)

hxlls, 2020-11-08:
换电源有惊喜，我用大雕固件，用树莓派的3A原装电源启动不了，换了小米的快充头，加数据线，正常启动，同卡没重新写卡，但是用友善那个杂交的固件就是KLEVER的那个mini什么的固件，树莓派的3A电源正常启动，同卡都测试过了，我也没搞懂是什么问题，应该可以判断是固件的patch 部分有差异，可能是sdio 或者其他控制电压部分的有区别？具体还没找出来是哪里的区别，在研究中

tl768, 2020-11-08:
> 换电源有惊喜，我用大雕固件，用树莓派的3A原装电源启动不了，换了小米的快充头，加数据线，正常启动，同卡没重新写卡，但用友善那个那个交叉的固件就是KLEVER的那个mini什么的固件，树莓派的3A电源正常启动，同卡都测试过了，我也没搞懂是什么问题，应该可以判断是固件的补丁部分有差异，可能是sdio或其他控制电压部分的有区别？具体还没发现来是哪里的区别，在研究中

谢谢，我找个电源试试。

467815891a, 2020-11-08:
我用的就是L大最新的源码，可以正常启动啊。你可以下载我的，也可以自己编译
https://github.com/467815891a/lede/actions/runs/333629627

tl768, 2020-11-08:
> 我用的就是L大最新的源码，可以正常启动啊。你可以下载我的，也可以自己编译
> https://github.com/467815891a/lede/actions/runs/333629627

谢谢，我就想知道本地编译怎么不好使。

tl768, 2020-11-08:
> > 换电源有惊喜，我用大雕固件，用树莓派的3A原装电源启动不了，换了小米的快充头，加数据线，正常启动，同卡没重新写卡，但用友善那个那个那个交叉的固件就是KLEVER的那个mini什么的固件，树莓派的3A电源正常启动，同卡都测试过了，我也没搞懂是什么问题，应该可以判断是固件的补丁部分有差异，可能是sdio或其他控制电压部分的有区别？具体还没发现来是哪里的区别，在研究中
> 
> 谢谢，我找个电源试试。

电源试了两个，不灵呀。晕了！

varios009, 2020-11-08:
> > > 换电源有惊喜，我用大雕固件，用树莓派的3A原装电源启动不了，换了小米的快充头，加数据线，正常启动，同卡没重新写卡，但用友善那个那个那个交叉的固件就是KLEVER的那个mini什么的固件，树莓派的3A电源正常启动，同卡都测试过了，我也没搞懂是什么问题，应该可以判断是固件的补丁部分有差异，可能是sdio或其他控制电压部分的有区别？具体还没发现来是哪里的区别，在研究中
> > 
> > 
> > 谢谢，我找个电源试试。
> 
> 电源试了两个，不灵呀。晕了！

换电源 TF卡 USB线 都会影响 启动~
用LEDE 编译的 只要用东芝32G 卡是 启动不了，用三星 32G 闪迪 8G 创见64G 夏科32G 都可以.
https://github.com/varios009/openwrtLEDE-x64-r2s/actions
这个是我编译,稳定性都可以。你可以试试~

tl768, 2020-11-08:
> > > > 换电源有惊喜，我用大雕固件，用树莓派的3A原装电源启动不了，换了小米的快充头，加数据线，正常启动，同卡没重新写卡，但用友善那个那个那个交叉的固件就是KLEVER的那个mini什么的固件，树莓派的3A电源正常启动，同卡都测试过了，我也没搞懂是什么问题，应该可以判断是固件的补丁部分有差异，可能是sdio或其他控制电压部分的有区别？具体还没发现来是哪里的区别，在研究中
> > > 
> > > 
> > > 谢谢，我找个电源试试。
> > 
> > 
> > 电源试了两个，不灵呀。晕了！
> 
> 换电源 TF卡 USB线 都会影响 启动~
> 用LEDE 编译的 只要用东芝32G 卡是 启动不了，用三星 32G 闪迪 8G 创见64G 夏科32G 都可以.
> 最好用 手机的充电器5V2A，试过用台达的5V3A 摄像头用的启动不了。
> https://github.com/varios009/openwrtLEDE-x64-r2s/actions
> 这个是我编译,稳定性都可以。你可以试试~

非常感谢

closed 2020-11-24 by tl768