coolsnowwolf/lede #5759: ASUS RT-ACRH17编译不通过

maidmeow4, 2020-11-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：似乎是 `firmware-2-ct-full-community-22.bin.lede.019` 此文件不能被正确下载
第一个mirror下载后报hash不符，其余mirror则全都HTTP 404

### (2) 路由器型号和固件版本

A：ASUS RT-ACRH17

### (3) 详细日志

A：
```
...
SHELL= flock /workdir/openwrt/tmp/.QCA988X-firmware-2-ct-full-community-22.bin.lede.019.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "QCA988X-firmware-2-ct-full-community-22.bin.lede.019" "8b4c99253aa309d35f2e060c190091b8db1b84dbda06a6a15c83ac0f9a938126" "firmware-2-ct-full-community-22.bin.lede.019" "https://www.candelatech.com/downloads/"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://www.candelatech.com/downloads/firmware-2-ct-full-community-22.bin.lede.019
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  210k  100  210k    0     0   431k      0 --:--:-- --:--:-- --:--:--  431k
Hash of the downloaded file does not match (file: 072fafbd4d741ba77f5d92b60a644c7360af0c21a3a9ec06f1c299484bf41688, requested: 8b4c99253aa309d35f2e060c190091b8db1b84dbda06a6a15c83ac0f9a938126) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/firmware-2-ct-full-community-22.bin.lede.019
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/firmware-2-ct-full-community-22.bin.lede.019
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/firmware-2-ct-full-community-22.bin.lede.019
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
Makefile:103: recipe for target '/workdir/openwrt/dl/QCA988X-firmware-2-ct-full-community-22.bin.lede.019' failed
make[3]: Leaving directory '/workdir/openwrt/package/firmware/ath10k-ct-firmware'
time: package/firmware/ath10k-ct-firmware/compile#0.35#0.10#2.00
make[3]: *** [/workdir/openwrt/dl/QCA988X-firmware-2-ct-full-community-22.bin.lede.019] Error 2
package/Makefile:111: recipe for target 'package/firmware/ath10k-ct-firmware/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[2]: *** [package/firmware/ath10k-ct-firmware/compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make: *** [world] Error 2
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
Error: Process completed with exit code 2.
```

presisco, 2020-11-09:
openwrt官方代码也有这个问题，ath10k-ct-firmware 2020-7月的版本文件hash码对不上

HoldOnBro, 2020-11-09:
竞斗云 GDock-p2w_r619ac 也出现同样问题

maidmeow4, 2020-11-11:
编译已能成功完成，看起来上游修复了问题。

closed 2020-11-11 by maidmeow4