coolsnowwolf/lede #5761: 编译出错“Makefile:151: recipe for target '/home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img' failed”

CyrixIII, 2020-11-09:
全新安装的Ubuntu 18.04 LTS

编译的时候出现如下错误：

`Makefile:151: recipe for target '/home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img' failed
make[5]: *** [/home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-x86-64-generic-squashfs-rootfs.img] Error 1
make[5]: Leaving directory '/home/ubuntu/lede/target/linux/x86/image'
Makefile:29: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/ubuntu/lede/target/linux/x86'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/target/linux'
time: target/linux/install#1069.85#155.60#1154.36
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
target/Makefile:19: recipe for target '/home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.target_install' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
/home/ubuntu/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2`

重试过多次，始终是出现这个错误，请问这是什么问题导致的呢？

jinglei207, 2020-11-10:
删了 openwrt 文件夹，从新clone.

CyrixIII, 2020-11-12:
> 删了 openwrt 文件夹，从新clone.

也删除了尝试过多次，还是不行，始终是这个错误

TendernessRadif, 2020-11-18:
2个 1.给大一些空间 2.输入 sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget curl swig rsync libtinfo5  在尝试

closed 2021-06-20 by coolsnowwolf