coolsnowwolf/lede #5770: 小白使用的Actions那里编译出来的固件，发现不支持ipv6

WONGHONGXU, 2020-11-11:
昨天自己用虚拟机装了歌Linux系统尝试自己编译，结果完全看不懂里面什么是什么，有谁能给我一个编译了ipv6其他的和Actions那里面没什么区别的固件啊

AmadeusGhost, 2020-11-11:
自行用默认配置加上 ipv6-helper 编译即可

Elletear, 2020-11-11:
action的話改變一下workflow文件即可編譯相應固件；虛擬機可參考一些別人發的範例及教程：eg.
https://www.right.com.cn/forum/thread-3732465-1-1.html【編譯教程】
https://www.right.com.cn/forum/thread-3682029-1-1.html【插件說明】
ps.僅供參考

WONGHONGXU, 2020-11-11:
谢谢，我去试试

Dimitri2020007, 2020-11-12:
按照lede的readme文档下操作，只是不执行make -j1 V=s。

make menuconfig出来后把.config中的内容复制到action根目录下的.config文件里。

使用action编译就是了。

不支持ipv6的可能跟ipv6helper无关，相关软件配置base system>dnsmasq-full（全部选择，不要选dnsmasq和dnsmasq-dhcpv6）

插件说明针对luci-app

功能重复的：

dns：

dnscrypt-proxy（时间太长不更新，不要选）

dnsforward（dns转发器，企业级）

https-dns-proxy（支持doh访问的dns客户端，设置不好上不了网，不会操作就不要装了。）

unbound（强烈推荐，支持dot访问，可以与dnsmasq分流结合使用，屏蔽广告。网上有教程。）

nat加速：

sfe、flowoffload（二选一、推荐sfe，两个都装会刷屏。#5735有提到这个错误。）

屏蔽广告：

adguard、adblock、adbyby-plus（三选一）

网络共享：

cifsd、cifs-mount、smb（已经强制选择，无法取消，与smb4有冲突）、smb4（选了你也装不上，与smb有冲突）

只装cifsd，功能用不了就只装cifs-mount

音频：

shairplay、airplay2（装airplay2）

流量统计：

vnstat（使用libgd）、statistics（使用libgd-full）、netdata（使用libgd-full）、nlbwmon、wrtbwmon

vnstat有冲突建议不装（选了libgd提示被libgd-full替代装不上）

流量控制：

qos、nft-qos、sqm（三选一）

功能未知（不经常使用的软件，根据个人需求选择。没有要求请略过，避免不必要冲突。）：

cshark、bmx6、bmx7、bird1-ipv4/ipv6、olsr、ahcpd、hnet、siitwizard（配置ipv6和v4的软件、与ahcpd一样的）、simple-adblock（屏蔽广告已经提到这些。）、ss-libev（用不上）asterisk（电话服务器？不常使用的软件。）、cjdns、clamav（杀毒软件）、freifunk系列软件（看不懂可以wiki）、noddos（ddos防护软件）、privoxy（代理软件）、squid（cdn加速）、travelmate（3g/4g网络用的软件，功能未知。）、tinyproxy、rp-pppoe-server、polipo

常用软件（必选）：

dnsmasq-full（屏蔽广告、域名泛解析、强大的分流可以配合unbound对抗dns错误。）

base system>busybox>networking utilitles>udhcpc和udhcpc6（取消了无法接收上游dhcp分配地址。）

network>iputils（全部选择，用于诊断网络）、iperf3（端口测速用）、ifstat和iftop（检查网络流量用）、ipset和ipset-dns（给ssr-plus用）、net-tools-route（内含ifconfig，强烈推荐）、odhcp6c（可应用在wan口和lan口，接收上游isp的ipv6地址）

luci>application>vlmcsd（用于激活windows，强烈推荐的必选软件，网上有教程。）、ssr-plus（你懂得，这里不多说了）

有argument list too long错误的（保持默认不要乱选）：

menuconfig

luci：proto、libraries

network：除了常用软件和time synchronization和tcprelay、wen server\proxies以外不要乱选。

firmware：（含显卡、有/无线网卡固件驱动，有存在冲突的选项。）

kernel modules：wireless drivers（包含冲突选项）

以上是软件编译列表（包含ipv6），是有点啰嗦了，凑合看吧。

closed 2020-11-15 by WONGHONGXU