coolsnowwolf/lede #5774: lxc页面显示不正常

4860575, 2020-11-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译了lxc后luci的页面显示不出“ Template”如下图：
[image]

### (2) 路由器型号和固件版本

A：rpi-4 
OpenWrt R20.10.20 / LuCI Master (git-20.256.12360-1a54222)

### (3) 详细日志

A：

NagaseKouichi, 2020-11-15:
给小娱路由器编译的lede的lxc也有这个问题

NagaseKouichi, 2020-11-17:
发现编译完的固件里面只有lxc-create、lxc-attach、lxc-console几个命令，lxc-start等很多都没有

Dimitri2020007, 2020-11-18:
menuconfig>utilities>lxc下全选了吗？

4860575, 2020-11-19:
应该不是没选lxc-start的问题。

4860575, 2020-11-19:
lxc-create -t download -n alpine -- --server mirrors.tuna.tsinghua.edu.cn/lxc-images  --no-validate
使用命令是可以创建成功的。

NagaseKouichi, 2020-11-23:
请问LZ你是怎么编译的？只要选中luci-app-lxc和menuconfig>utilities>lxc下面的所有组件就可以吗？

http://www.gnuton.org/blog/2016/02/lxc-on-openwrt/
我看这里写还要进行一些别的设置，你有设置这个吗？

4860575, 2020-11-23:
> 请问LZ你是怎么编译的？只要选中luci-app-lxc和menuconfig>utilities>lxc下面的所有组件就可以吗？
> 
> http://www.gnuton.org/blog/2016/02/lxc-on-openwrt/
> 我看这里写还要进行一些别的设置，你有设置这个吗？

1、替换lxc为2.1版本（可以从官方19.07，18.06下载替换），lean包里的lxc是4.0，我测试了4.0不行。2.1没问题；
2、正常编译就可以，选择上内核相应的选项；
3、如果没有/srv/lxc就创建；
4、使用lxc-create -t download -n alpine -- --server mirrors.tuna.tsinghua.edu.cn/lxc-images --no-validate 创建就可以了

NagaseKouichi, 2020-11-23:
> 1、替换lxc为2.1版本（可以从官方19.07，18.06下载替换），lean包里的lxc是4.0，我测试了4.0不行。2.1没问题；
> 2、正常编译就可以，选择上内核相应的选项；
> 3、如果没有/srv/lxc就创建；
> 4、使用lxc-create -t download -n alpine -- --server mirrors.tuna.tsinghua.edu.cn/lxc-images [...]

谢谢，我之后试下

NagaseKouichi, 2020-11-23:
把package/feeds/packages/lxc/里面的文件替换成2.1.1之后，编译报错呢

[image]

4860575, 2020-11-24:
> 把package/feeds/packages/lxc/里面的文件替换成2.1.1之后，编译报错呢
> 
> [image]
> 
> 我从头编译试试

清理一下编译目录。

NagaseKouichi, 2020-11-24:
> > 把package/feeds/packages/lxc/里面的文件替换成2.1.1之后，编译报错呢
> > [image]
> > 我从头编译试试
> 
> 清理一下编译目录。

嗯，make dirclean之后编译成功了，谢谢。我去测试下

NagaseKouichi, 2020-11-24:
已经成功运行了Ubuntu Trusty，和 https://mlapp.cn/375.html 这篇文章说的一样，debian无法启动，提示mount cgroup失败。

另外，我按照LZ说的方法编译后，lxc的luci界面显示正常，下拉菜单可以显示镜像列表。
我用的这个LXC镜像源：mirrors.tuna.tsinghua.edu.cn/lxc-images

4860575, 2020-11-24:
> 已经成功运行了Ubuntu Trusty，和 https://mlapp.cn/375.html 这篇文章说的一样，debian无法启动，提示mount cgroup失败。
> 
> 另外，我按照LZ说的方法编译后，lxc的luci界面显示正常，下拉菜单可以显示镜像列表。
> 我用的这个LXC镜像源：mirrors.tuna.tsinghua.edu.cn/lxc-images

那可能我是arm架构的问题

NagaseKouichi, 2020-11-24:
https://forum.openwrt.org/t/request-for-lxc-support-in-mvebu/58767/4
我找到这个帖子，他通过给lxc-auto.init打补丁，可以解决mount /sys/fs/cgroup/systemd失败的问题。

我试了下，在openwrt启动后再运行下面两行命令也是可以的：
```
mkdir -p /sys/fs/cgroup/systemd
mount -t cgroup -o rw,nosuid,nodev,noexec,relatime,none,name=systemd cgroup /sys/fs/cgroup/systemd
```

运行完就可以正常启动debian和ubuntu高版本的容器了

4860575, 2020-11-24:
> https://forum.openwrt.org/t/request-for-lxc-support-in-mvebu/58767/4
> 我找到这个帖子，他通过给lxc-auto.init打补丁，可以解决mount /sys/fs/cgroup/systemd失败的问题。
> 
> 我试了下，在openwrt启动后再运行下面两行命令也是可以的：
> 
> ```
> mkdir -p /sys/fs/cgroup/systemd
> mount -t cgroup -o rw,nosuid,nodev,noexec,relatime,none,name=systemd cgroup /sys/fs/cgroup/systemd
> ```
> 
> 运行完就可以正常启动debian和ubuntu高版本的容器了
> 
> 不知道这个可不可以让LXC4.0正常运行

我看他是19.07版本。lxc版本应该是2.1

NagaseKouichi, 2020-11-24:
> 我看他是19.07版本。lxc版本应该是2.1

嗯，我试了2.1运行上面的命令就可以正常启动debian和ubuntu高版本的容器了

下次有时间在4.0试试

closed 2021-06-20 by coolsnowwolf