coolsnowwolf/lede #5795: 修改登录账号跟密码

Guxing0321, 2020-11-15:
Hi Team

我想修改登录账号跟密码
参考的教程如下：
https://blog.csdn.net/songshuai0223/article/details/103857650

但是发现一个问题，就是改完后，一旦我这边在页面修改了配置，
点击应用，就会一直加载

[image]

有办法解决这个问题吗，或者 在编译的时候这个可以怎么改

BR
Robin

Guxing0321, 2020-11-15:
Hi Team

我发现我是少改了一个文件
https://www.right.com.cn/forum/thread-469400-1-1.html
/usr/lib/lua/luci/controller/admin/servicectl.lua

如果有修改编译的方法，也可以提供下，谢谢

BR
Robin

doumeki, 2020-11-23:
直接改zzz default 文件就行了, 至于用户名自己试吧，改密码如下：
package/lean/default-settings/files/zzz-default-settings
sed -i 's/root::0:0:99999:7:::/root:$1$V4UetPzk$CYXluq4wUazHjmCDBCqXF.:0:0:99999:7:::/g' /etc/shadow 这行把$1$***替换就OK了
至于密码用的是什么加密方式不知道，自己在虚拟机里改了后复制出来吧。

closed 2021-06-20 by coolsnowwolf