coolsnowwolf/lede #5797: cat: -: No such file or directory

pansila, 2020-11-15:
## 1.`make -j1 V=s`编译错误

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：按照首页上的步骤，一步一步执行，到`make -j1 V=s`这步时编译失败。检查了出问题包的configure文件，发现问题是cat不知为何不认识'-'标准输入符号，导致生成的测试代码文件不完整，从何检查条件不过。

### (2) 路由器型号和固件版本

A：raspberry pi 4B
编译环境是在Windows 10上的WSL2上进行，网上有说明在这个系统下是可以成功的，并且已解决大小写问题。

### (3) 详细日志

A：
<details>
<summary>make失败时的log</summary>
make[2]: Warning: File '/mnt/d/projects/coolsnowwolf-lede/tmp/info/.files-packageinfo.mk' has modification time 0.17 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: Warning: File '/mnt/d/projects/coolsnowwolf-lede/tmp/info/.files-targetinfo.mk' has modification time 0.0059 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: Entering directory '/mnt/d/projects/coolsnowwolf-lede'
make[2]: Entering directory '/mnt/d/projects/coolsnowwolf-lede'
+ mkdir -p /mnt/d/projects/coolsnowwolf-lede/staging_dir/target-aarch64_cortex-a72_musl
+ cd /mnt/d/projects/coolsnowwolf-lede/staging_dir/target-aarch64_cortex-a72_musl
+ mkdir -p bin lib stamp
mkdir -p /mnt/d/projects/coolsnowwolf-lede/build_dir/target-aarch64_cortex-a72_musl/stamp
touch /mnt/d/projects/coolsnowwolf-lede/staging_dir/target-aarch64_cortex-a72_musl/.prepared
+ mkdir -p /mnt/d/projects/coolsnowwolf-lede/staging_dir/host
+ cd /mnt/d/projects/coolsnowwolf-lede/staging_dir/host
+ mkdir -p bin lib stamp
mkdir -p /mnt/d/projects/coolsnowwolf-lede/build_dir/host/stamp /mnt/d/projects/coolsnowwolf-lede/staging_dir/host/include/sys
install -m0644 /mnt/d/projects/coolsnowwolf-lede/tools/include/*.h /mnt/d/projects/coolsnowwolf-lede/staging_dir/host/include/
install -m0644 /mnt/d/projects/coolsnowwolf-lede/tools/include/sys/*.h /mnt/d/projects/coolsnowwolf-lede/staging_dir/host/include/sys/
ln -snf lib /mnt/d/projects/coolsnowwolf-lede/staging_dir/host/lib64
touch /mnt/d/projects/coolsnowwolf-lede/staging_dir/host/.prepared
make[3]: Entering directory '/mnt/d/projects/coolsnowwolf-lede/tools/flock'
make[3]: Leaving directory '/mnt/d/projects/coolsnowwolf-lede/tools/flock'
time: tools/flock/compile#0.09#0.01#0.42
make[3]: Entering directory '/mnt/d/projects/coolsnowwolf-lede/tools/xz'
(cd /mnt/d/projects/coolsnowwolf-lede/build_dir/host/xz-5.2.5/; if [ -x configure ]; then cp -fpR /mnt/d/projects/coolsnowwolf-lede/scripts/config.{guess,sub} /mnt/d/projects/coolsnowwolf-lede/build_dir/host/xz-5.2.5// && CC="gcc" CFLAGS="-O2 -I/mnt/d/projects/coolsnowwolf-lede/staging_dir/host/include " CXX="g++" CPPFLAGS="-I/mnt/d/projects/coolsnowwolf-lede/staging_dir/host/include " LDFLAGS="-L/mnt/d/projects/coolsnowwolf-lede/staging_dir/host/lib " CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/mnt/d/projects/coolsnowwolf-lede/staging_dir/host --exec-prefix=/mnt/d/projects/coolsnowwolf-lede/staging_dir/host --sysconfdir=/mnt/d/projects/coolsnowwolf-lede/staging_dir/host/etc --localstatedir=/mnt/d/projects/coolsnowwolf-lede/staging_dir/host/var --sbindir=/mnt/d/projects/coolsnowwolf-lede/staging_dir/host/bin --enable-static=yes --enable-shared=no --disable-doc --disable-nls --with-pic ; fi )
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory

XZ Utils 5.2.5

System type:
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu

Configure options:
checking if debugging code should be compiled... no
checking which encoders to build... lzma1 lzma2 delta x86 powerpc ia64 arm armthumb sparc
checking which decoders to build... lzma1 lzma2 delta x86 powerpc ia64 arm armthumb sparc
checking which match finders to build... hc3 hc4 bt2 bt3 bt4
checking which integrity checks to build... crc32 crc64 sha256
checking if external SHA-256 should be used... no
checking if assembler optimizations should be used... x86_64
checking if small size is preferred over speed... no
checking if threading support is wanted... yes, posix
checking how much RAM to assume if the real amount is unknown... 128 MiB
cat: -: No such file or directory
checking if library symbol versioning should be used... yes
checking if sandboxing should be used... maybe (autodetect)

checking for a shell that conforms to POSIX... /bin/sh

Initializing Automake:
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... cat: -: No such file or directory
no
checking whether make supports nested variables... yes
cat: -: No such file or directory
cat: -: No such file or directory
checking whether ln -s works... yes
checking whether make supports the include directive... cat: -: No such file or directory
no
checking for x86_64-pc-linux-gnu-gcc... gcc
cat: -: No such file or directory
checking whether the C compiler works... no
configure: error: in `/mnt/d/projects/coolsnowwolf-lede/build_dir/host/xz-5.2.5':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[3]: *** [Makefile:37: /mnt/d/projects/coolsnowwolf-lede/build_dir/host/xz-5.2.5/.configured] Error 77
make[3]: Leaving directory '/mnt/d/projects/coolsnowwolf-lede/tools/xz'
time: tools/xz/compile#0.48#0.07#1.98
make[2]: *** [tools/Makefile:165: tools/xz/compile] Error 2
make[2]: Leaving directory '/mnt/d/projects/coolsnowwolf-lede'
make[1]: *** [tools/Makefile:161: /mnt/d/projects/coolsnowwolf-lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyynyyyyynyynnyyyynyyyyyyyyyyyyyyynynynnyyyynynyyyy] Error 2
make[1]: Leaving directory '/mnt/d/projects/coolsnowwolf-lede'
make: *** [/mnt/d/projects/coolsnowwolf-lede/include/toplevel.mk:224: world] Error 2
</details>

<details>
<summary>出错包的`config.log`文件</summary>
configure:5300: checking whether the C compiler works
configure:5322: gcc -O2 -I/mnt/d/projects/coolsnowwolf-lede/staging_dir/host/include  -I/mnt/d/projects/coolsnowwolf-lede/staging_dir/host/include  -L/mnt/d/projects/coolsnowwolf-lede/staging_dir/host/lib  conftest.c  >&5
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/8/../../../x86_64-linux-gnu/Scrt1.o: in function `_start':
(.text+0x20): undefined reference to `main'
collect2: error: ld returned 1 exit status
configure:5326: $? = 1
configure:5364: result: no
configure: failed program was:
| /* confdefs.h */
| #define NDEBUG 1
| #define HAVE_ENCODER_LZMA1 1
| #define HAVE_ENCODER_LZMA2 1
| #define HAVE_ENCODER_DELTA 1
| #define HAVE_ENCODER_X86 1
| #define HAVE_ENCODER_POWERPC 1
| #define HAVE_ENCODER_IA64 1
| #define HAVE_ENCODER_ARM 1
| #define HAVE_ENCODER_ARMTHUMB 1
| #define HAVE_ENCODER_SPARC 1
| #define HAVE_ENCODERS 1
| #define HAVE_DECODER_LZMA1 1
| #define HAVE_DECODER_LZMA2 1
| #define HAVE_DECODER_DELTA 1
| #define HAVE_DECODER_X86 1
| #define HAVE_DECODER_POWERPC 1
| #define HAVE_DECODER_IA64 1
| #define HAVE_DECODER_ARM 1
| #define HAVE_DECODER_ARMTHUMB 1
| #define HAVE_DECODER_SPARC 1
| #define HAVE_DECODERS 1
| #define HAVE_MF_HC3 1
| #define HAVE_MF_HC4 1
| #define HAVE_MF_BT2 1
| #define HAVE_MF_BT3 1
| #define HAVE_MF_BT4 1
| #define HAVE_CHECK_CRC32 1
| #define HAVE_CHECK_CRC64 1
| #define HAVE_CHECK_SHA256 1
configure:5369: error: in `/mnt/d/projects/coolsnowwolf-lede/build_dir/host/xz-5.2.5':
configure:5371: error: C compiler cannot create executables
See `config.log' for more details
</details>

pansila, 2020-11-19:
看起来不能在windows目录编译，在WSL主目录编译，问题消失。

closed 2020-11-19 by pansila

1368129224, 2021-03-24:
请问你是怎么解决的？把代码放到/home吗？会不会导致C盘占用太大

pansila, 2021-04-19:
@1368129224 对，放到wsl实例里面，怕占用大就把镜像文件移到其它硬盘，wsl  --export/--import，网上有教程

sheep94lion, 2021-09-18:
我也遇到了类似问题，在Unraid虚拟机里挂在了Unraid share，代码放在share里出现了同样的错误。