coolsnowwolf/lede #5813: HiWifi HC5962 固件编译，分区问题（怀疑）

uoow, 2020-11-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：11月编译的固件，编译不报错，文件生成正常；但刷到路由器后出现不停重启现象，这情况以前某个版本好像也遇到过，好像是内核更新了，分区太小的原因，后来更新过DTS相关的文件后解决

### (2) 路由器型号和固件版本

A：HiWiFi B70 HC5962

### (3) 详细日志

A：

到lede-project上看过，DTS相关的文件是9月份的版本，比该项目下的要新，不知更新后能否解决此问题？

coolsnowwolf, 2020-11-23:
可以提交个 PR

uoow, 2020-11-23:
> 
> 
> 可以提交个 PR

大佬我不会github的PR，点开了也不知道要填什么，创建PR的按钮都是不能按状态

hyird, 2020-12-02:
15号编译的，表示并没有任何问题。
[image]
[image]
 分区是会使用全部NAND的

uoow, 2020-12-02:
> 
> 
> 15号编译的，表示并没有任何问题。
> [image]
> [image]
> 分区是会使用全部NAND的

这就奇怪了，你的DTS文件用的是默认的吗，没有自行修改过？boot是用breed还是uboot？

hyird, 2020-12-02:
> > 15号编译的，表示并没有任何问题。
> > [image]
> > [image]
> > 分区是会使用全部NAND的
> 
> 这就奇怪了，你的DTS文件用的是默认的吗，没有自行修改过？boot是用breed还是uboot？

breed 这个机器 breed要刷facotry固件。 sysupgrade只能在线升级

uoow, 2020-12-02:
> 
> 
> > > 15号编译的，表示并没有任何问题。
> > > 分区是会使用全部NAND的
> > 
> > 
> > 这就奇怪了，你的DTS文件用的是默认的吗，没有自行修改过？boot是用breed还是uboot？
> 
> breed 这个机器 breed要刷facotry固件。 sysupgrade只能在线升级
难道是breed的原因？我的是uboot。你的固件生成后大小是多少，factory与sysupgrade的。能把你的DTS文件发给我对比下吗？

hyird, 2020-12-02:
> > > > 15号编译的，表示并没有任何问题。
> > > > 分区是会使用全部NAND的
> > > 
> > > 
> > > 这就奇怪了，你的DTS文件用的是默认的吗，没有自行修改过？boot是用breed还是uboot？
> > 
> > 
> > breed 这个机器 breed要刷facotry固件。 sysupgrade只能在线升级
> 
> 难道是breed的原因？我的是uboot。你的固件生成后大小是多少，factory与sysupgrade的。能把你的DTS文件发给我对比下吗？

刷个breed先

closed 2021-06-20 by coolsnowwolf