coolsnowwolf/lede #5819: 安装了acme之后，整个luci界面都用不了了。

soulxiangmu, 2020-11-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：安装了acme之后，整个luci界面都乱了，用不了。换了别的主题也一样。

### (2) 路由器型号和固件版本

A：Linksys WRT1900ACS
OpenWrt R9.11.23 / LuCI Master (git-19.331.11944-c1039b9)

### (3) 详细日志

A：
root@OpenWrt:~# logread
Sun Nov 22 04:41:15 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:41:18 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:41:18 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:41:27 2020 daemon.err uhttpd[2775]: /usr/lib/lua/luci/dispatcher.lua:382: Failed to execute template dispatcher target for entry '/admin/status/overview'.
Sun Nov 22 04:41:27 2020 daemon.err uhttpd[2775]: The called action terminated with an exception:
Sun Nov 22 04:41:27 2020 daemon.err uhttpd[2775]: /usr/lib/lua/luci/template.lua:55: Failed to execute template 'admin_status/index'.
Sun Nov 22 04:41:27 2020 daemon.err uhttpd[2775]: A runtime error occured: /usr/lib/lua/luci/model/network.lua:349: attempt to call field 'checkmac' (a nil value)
Sun Nov 22 04:41:27 2020 daemon.err uhttpd[2775]: stack traceback:
Sun Nov 22 04:41:27 2020 daemon.err uhttpd[2775]:       [C]: in function 'assert'
Sun Nov 22 04:41:27 2020 daemon.err uhttpd[2775]:       /usr/lib/lua/luci/dispatcher.lua:382: in function 'dispatch'
Sun Nov 22 04:41:27 2020 daemon.err uhttpd[2775]:       /usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>
Sun Nov 22 04:41:47 2020 cron.info crond[12889]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 04:41:50 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:42:19 2020 cron.info crond[13658]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 04:42:20 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:42:20 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:42:23 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:42:23 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:42:52 2020 cron.info crond[14403]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 04:42:52 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:42:52 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:43:24 2020 cron.info crond[15100]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 04:43:24 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:43:24 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:43:27 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:43:27 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:43:56 2020 cron.info crond[15813]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 04:43:59 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:43:59 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:44:28 2020 cron.info crond[16533]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 04:45:02 2020 cron.info crond[17228]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 04:45:02 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:45:02 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:45:05 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:45:05 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:45:34 2020 cron.info crond[17972]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 04:45:34 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:45:34 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:45:38 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:45:38 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:46:07 2020 cron.info crond[18671]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 04:46:07 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:46:07 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:46:12 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:46:12 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 04:46:41 2020 cron.info crond[19386]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 04:46:41 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 04:46:41 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
[... 579 lines cut ...]
Sun Nov 22 05:10:30 2020 cron.info crond[27059]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 05:10:34 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 05:10:34 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 05:11:01 2020 authpriv.info dropbear[21133]: Exit (root): Exited normally
Sun Nov 22 05:11:02 2020 cron.info crond[27899]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 05:11:03 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 05:11:03 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 05:11:07 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 05:11:07 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 05:11:31 2020 authpriv.info dropbear[30238]: Exit (root): Exited normally
Sun Nov 22 05:11:36 2020 cron.info crond[28760]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 05:11:36 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 05:11:36 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 05:11:37 2020 authpriv.info dropbear[28819]: Child connection from 192.168.2.150:6834
Sun Nov 22 05:11:37 2020 authpriv.notice dropbear[28819]: Password auth succeeded for 'root' from 192.168.2.150:6834
Sun Nov 22 05:11:39 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 05:11:39 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 05:11:41 2020 authpriv.info dropbear[29382]: Child connection from 192.168.2.150:6837
Sun Nov 22 05:11:41 2020 authpriv.notice dropbear[29382]: Password auth succeeded for 'root' from 192.168.2.150:6837
Sun Nov 22 05:12:08 2020 cron.info crond[29612]: crond (busybox 1.30.0) started, log level 5
Sun Nov 22 05:12:08 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 05:12:08 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 05:12:11 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 05:12:11 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
root@OpenWrt:~#

hxlls, 2020-11-22:
R9 11.23？都20年了啊。。R20版本起

soulxiangmu, 2020-11-23:
> R9 11.23？都20年了啊。。R20版本起

新版没有科学上网插件了啊 所以没换

Ljzkirito, 2020-11-23:
> 新版没有科学上网插件了啊 所以没换

火星？？？
在源码目录下运行以下代码
```
sed -i 's/^#\(.*helloworld\)/\1/' feeds.conf.default
./scripts/feeds update -a && ./scripts/feeds install -a
```
然后在`make menuconfig` `luci`中就能找到

soulxiangmu, 2020-11-23:
> > 新版没有科学上网插件了啊 所以没换
> 
> 火星？？？
> 在源码目录下运行以下代码
> 
> ```
> sed -i 's/^#\(.*helloworld\)/\1/' feeds.conf.default
> ./scripts/feeds update -a && ./scripts/feeds install -a
> ```
> 
> 然后在`make menuconfig` `luci`中就能找到

谢谢大佬，今天把lede和openwrt这两个都编译成功了，目前上机测试，openwrt这个目录编译出来的固件，初步使用之后发现有两个问题：
1.科学上网插件，更新了v2ray订阅之后，显示ShadowsocksR Plus+ 未运行，输入 /etc/init.d/shadowsocksr start 后才运行。
2.路由重启之后，所有的设置都恢复到原始状态，无法保存任何设置。

lede编译的那个固件目前还没刷进去测试。。
随便问下大佬，lede这个目录编译出来的固件和openwrt这个目录编译出来的固件有什么不同的地方？

Ljzkirito, 2020-11-24:
> 1.科学上网插件，更新了v2ray订阅之后，显示ShadowsocksR Plus+ 未运行，输入 /etc/init.d/shadowsocksr start 后才运行。
我猜与此问题[#225](https://github.com/fw876/helloworld/issues/225)有关，等待coolsnowwolf大佬合并[pr](https://github.com/fw876/helloworld/pull/235)或自行用[Mattraks大佬的测试分支](https://github.com/Mattraks/helloworld)编译ssrplus。
> 2.路由重启之后，所有的设置都恢复到原始状态，无法保存任何设置。
可能此[pr](https://github.com/fw876/helloworld/pull/231)有关，和上一个问题一样，等待coolsnowwolf大佬合并。
> 随便问下大佬，lede这个目录编译出来的固件和openwrt这个目录编译出来的固件有什么不同的地方？
lede源码有coolsnowwolf大佬的一些更改，如禁止DNS返回IPv6解析；[packages](https://github.com/coolsnowwolf/packages)与[luci](https://github.com/coolsnowwolf/luci)使用自己维护源，虽更新慢一点，胜在稳定。
openwrt指的是https://github.com/openwrt/openwrt的话，openwrt更激进。
其他的不同自行体会，喜欢用哪个就用哪个：）

closed 2021-06-20 by coolsnowwolf