coolsnowwolf/lede #5823: 大佬们请问编译coolsnowwolf / openwrt 报错了 如何解决，谢谢🙏

blacknesswing, 2020-11-23:
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
Makefile:102: recipe for target '/workdir/openwrt/dl/QCA99X0-firmware-5-ct-full-community-12.bin-lede.013' failed
make[3]: Leaving directory '/workdir/openwrt/package/firmware/ath10k-firmware'
time: package/firmware/ath10k-firmware/compile#0.96#0.38#2.82
make[3]: *** [/workdir/openwrt/dl/QCA99X0-firmware-5-ct-full-community-12.bin-lede.013] Error 2
make[2]: *** [package/firmware/ath10k-firmware/compile] Error 2
package/Makefile:111: recipe for target 'package/firmware/ath10k-firmware/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
/workdir/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
Error: Process completed with exit code 2.

LiYao703198594, 2020-11-23:
服务器在国内吗？404警告
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.

blacknesswing, 2020-11-23:
> 服务器在国内吗？404警告
> 卷曲：（22）请求的URL返回错误：404未找到
> 下载失败。
> 没有更多的镜子可以尝试-放弃。

您好 我是用Actions编译的

9l, 2020-11-23:
看了一下你 X86_stable.yml 的 action 错误

+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://www.candelatech.com/downloads/ath10k-10-4b/firmware-5-ct-full-community-12.bin-lede.013
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
 20  548k   20  112k    0     0   197k      0  0:00:02 --:--:--  0:00:02  197k
100  548k  100  548k    0     0   759k      0 --:--:-- --:--:-- --:--:--  758k
Hash of the downloaded file does not match (file: 6a95c4323e008499af8d3549513c8a128c69b97296888a4c4b642e6c14a65508, requested: 6fa74a3fc87cba97dbc4a7213b760f8d997cd9c5f11900d47d387b23764cf20a) - deleting download.

ath10k  固件下载回来的文件 hash 校验错误，这个文件的 sha256 hash 变了，你要么把配置文件里的 ath10k 配置关了，要么自己写个脚本改一下这个 hash，在这个文件里 https://github.com/coolsnowwolf/openwrt/blob/lede-17.01/package/firmware/ath10k-firmware/Makefile

blacknesswing, 2020-11-23:
> 看了一下你X86_stable.yml的动作错误
> 
> * curl -f-连接超时20-重试5-位置-不安全[https://www.candelatech.com/downloads/ath10k-10-4b/firmware-5-ct-full-community-12.bin -lede.013](https://www.candelatech.com/downloads/ath10k-10-4b/firmware-5-ct-full-community-12.bin-lede.013)
>   总计百分比已接收百分比Xferd平均速度时间时间时间当前
>   Dload上传总已花费左速度
>   0 0 0 0 0 0 0 0 0-：-：--：-：--：-：-0 
>   20 548k 20 112k 0 0 197k 0 0:00:02-：- -：-0:00:02 197k 
>   100 548k 100 548k 0 0 759k 0-：-：--：-：--：-：-758k
>   下载文件的哈希不匹配（文件：6a95c4323e008499af8d3549513c8a128c69b97296888a4c4b642e6c14a65508，请求：6fa74a3fc87cba97dbc4a7213b760f8d997cd9c5f11900d47d387b23764cf20a）-删除下载。
> 
> ath10k固件下载回来的文件hash校验错误，这个文件的sha256 hash变了，你可以把配置文件里的ath10k配置关了，可以自己写个脚本改一下这个hash，在这个文件里[https://github.com .com / coolsnowwolf / openwrt / blob / lede-17.01 / package / firmware / ath10k-firmware / Makefile](https://github.com/coolsnowwolf/openwrt/blob/lede-17.01/package/firmware/ath10k-firmware/Makefile)

谢谢大佬 脚本不会写，我去把ath10k相关的东西关了

blacknesswing, 2020-11-23:
> 看了一下你 X86_stable.yml 的 action 错误
>
> * curl -f --connect-timeout 20 --retry 5 --location --insecure https://www.candelatech.com/downloads/ath10k-10-4b/firmware-5-ct-full-community-12.bin-lede.013
>   % [...]

大佬在请问一下，关闭ath10k 对固件有影响吗？

AmadeusGhost, 2020-11-23:
> 大佬在请问一下，关闭ath10k 对固件有影响吗？

ath10k 是高通无线网卡用的

blacknesswing, 2020-11-23:
> > 大佬在请问一下，关闭ath10k 对固件有影响吗？
> 
> ath10k 是高通无线网卡用的

谢谢大佬 明白了

closed 2021-06-20 by coolsnowwolf