coolsnowwolf/lede #5825: 编译小娱路由器固件报错

NagaseKouichi, 2020-11-23:
[image]

bash有报错bash: COLON: command not found
浏览器打开 https://chrome-infra-packages.appspot.com/dl/gn/gn/linux-amd64/+/git_revision:e327ffdc503815916db2543ec000226a8df45163 这个地址可以下载

编译的时候luci里面没有选择include naiveproxy。
请问大佬们这个如何解决呢？之前可以编译成功，最近不知道为什么不行了

sxx1314, 2020-11-24:
- 梯子挂全局   
- 404 多明显的报错。
- 不是你浏览器可以下载 shell里的就一定能下载。

NagaseKouichi, 2020-11-24:
> * 梯子挂全局
> * 404 多明显的报错。
> * 不是你浏览器可以下载 shell里的就一定能下载。

梯子已经挂了全局呢。那个bash报错是正常的吗？

Ljzkirito, 2020-11-24:
反正有GitHub账号，直接学学用Github Actions编译，本地编译挂全局还下载困难，云编译就不用考虑这些问题了。
可参考：
1.[P3TERX/Actions-OpenWrt](https://github.com/P3TERX/Actions-OpenWrt)，易上手，[中文教程](https://p3terx.com/archives/build-openwrt-with-github-actions.html)。
2.[KFERMercer/OpenWrt-CI](https://github.com/KFERMercer/OpenWrt-CI)，灵活性大。
同时我也是XY-C5使用者，这是我的[Github Actions编译](https://github.com/Ljzkirito/Actions-Openwrt-XY-C5)，可参考：）

NagaseKouichi, 2020-11-24:
> 反正有GitHub账号，直接用Github Actions编译，本地编译挂全局还下载困难，云编译就不用考虑这些问题了。
> 可参考：
> 1.[P3TERX/Actions-OpenWrt](https://github.com/P3TERX/Actions-OpenWrt)，易上手，[中文教程](https://p3terx.com/archives/build-openwrt-with-github-actions.html)。
> 2.[KFERMercer/OpenWrt-CI](https://github.com/KFERMercer/OpenWrt-CI)，灵活性大。
> 同时我也是XY-C5使用者，这是我的[Github Actions编译](https://github.com/Ljzkirito/Actions-Openwrt-XY-C5)，可参考：）

谢谢，我研究下

closed 2020-12-15 by NagaseKouichi