coolsnowwolf/lede #5826: 路由重启之后所有的设置丢失

soulxiangmu, 2020-11-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：路由重启之后所有的设置都恢复到原始状态了。包括上网账号也无法保存。

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] OF: fdt: Machine model: Linksys WRT1900ACS
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Thu Aug 27 16:11:16 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 131072
Thu Aug 27 16:11:16 2020 kern.debug kernel: [    0.000000]   Normal zone: 1152 pages used for memmap
Thu Aug 27 16:11:16 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Thu Aug 27 16:11:16 2020 kern.debug kernel: [    0.000000]   Normal zone: 131072 pages, LIFO batch:31
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x7c/0x4f8 with crng_init=0
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] percpu: Embedded 11 pages/cpu s15436 r8192 d21428 u45056
Thu Aug 27 16:11:16 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s15436 r8192 d21428 u45056 alloc=11*4096
Thu Aug 27 16:11:16 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 root=/dev/mtdblock7 ro rootdelay=1 rootfstype=jffs2 earlyprintk mtdparts=armada-nand:2048K(uboot)ro,256K(u_env),256K(s_env),1m@9m(devinfo),40m@10m(kernel),34m@16m(rootfs),40m@50m(alt_kernel),34m@56m(alt_rootfs),80m@10m(ubifs),-@90m(syscfg)
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000] Bootloader command line (ignored): console=ttyS0,115200 root=/dev/mtdblock7 ro rootdelay=1 rootfstype=jffs2 earlyprintk mtdparts=armada-nand:2048K(uboot)ro,256K(u_env),256K(s_env),1m@9m(devinfo),40m@10m(kernel),34m@16m(rootfs),40m@50m(alt_kernel),34m@56m(alt_rootfs),80m@10m(ubifs),-@90m(syscfg)
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] Memory: 509392K/524288K available (6144K kernel code, 214K rwdata, 1516K rodata, 1024K init, 231K bss, 14896K reserved, 0K cma-reserved, 0K highmem)
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000] Virtual kernel memory layout:
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (7136 kB)
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 215 kB)
Thu Aug 27 16:11:16 2020 kern.notice kernel: [    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 232 kB)
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] L2C-310 enabling early BRESP for Cortex-A9
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] L2C-310 D prefetch enabled, offset 1 lines
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] L2C-310 Coherent cache controller enabled, 16 ways, 1024 kB
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000000] L2C-310 Coherent: CACHE_ID 0x410054c9, AUX_CTRL 0x56070001
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000006] sched_clock: 64 bits at 800MHz, resolution 1ns, wraps every 4398046511103ns
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000015] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0xb881274fa3, max_idle_ns: 440795210636 ns
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000027] Switching to timer-based delay loop, resolution 1ns
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000144] Ignoring duplicate/late registration of read_current_timer delay
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000151] clocksource: armada_370_xp_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 76450417870 ns
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000282] Calibrating delay loop (skipped), value calculated using timer frequency.. 1600.00 BogoMIPS (lpj=8000000)
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000289] pid_max: default: 32768 minimum: 301
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000352] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000357] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000659] CPU: Testing write buffer coherency: ok
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000670] CPU0: Spectre v2: using BPIALL workaround
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000782] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Thu Aug 27 16:11:16 2020 kern.info kernel: [    0.000948] Setting up static identity map for 0x100000 - 0x100060
[... 645 lines cut ...]
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: lock engine: pthread robust mutexes
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: thunder lock: enabled
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: uwsgi socket 0 inherited UNIX address /var/run//luci-webui.socket fd 0
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: uWSGI running as root, you can use --uid/--gid/--chroot options
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: *** WARNING: you are running uWSGI as root !!! (use the --uid flag) ***
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: uWSGI running as root, you can use --uid/--gid/--chroot options
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: *** WARNING: you are running uWSGI as root !!! (use the --uid flag) ***
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: your server socket listen backlog is limited to 100 connections
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: your mercy for graceful operations on workers is 60 seconds
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: mapped 487200 bytes (475 KB) for 9 cores
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: *** Operational MODE: preforking+threaded ***
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: initialized CGI path: /www/
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: *** no app loaded. going in full dynamic mode ***
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: uWSGI running as root, you can use --uid/--gid/--chroot options
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: *** WARNING: you are running uWSGI as root !!! (use the --uid flag) ***
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: *** uWSGI is running in multiple interpreter mode ***
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: spawned uWSGI master process (pid: 9064)
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: cheap mode enabled: waiting for socket connection...
[... cut ...]
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: Tue Nov 24 01:59:06 2020 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is ready to accept requests
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: announcing my loyalty to the Emperor...
Tue Nov 24 01:59:06 2020 daemon.info uwsgi-emperor: Tue Nov 24 01:59:06 2020 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Tue Nov 24 01:59:09 2020 daemon.info pppd[7403]: System time change detected.
Tue Nov 24 01:59:31 2020 cron.err crond[4526]: time disparity of 127298 minutes detected

sxx1314, 2020-11-24:
### df -h 看占用
- 原因：设置还原大概率 overlay空间占满/异常
- 修改：不要编译太多东西/自行删除overlay中不需要的东西。如果硬件允许可以自行修改扩容overlay空间。

soulxiangmu, 2020-11-24:
> ### df -h看占用
> * 原因：设置还原大概率overlay空间占满/异常
> * 修改：不要编译过多的东西/自行删除overlay中不需要的东西。如果硬件允许可以自行修改扩容overlayspace。

我看了一下 确实是空间满了，后来编译的时候，选少一点插件，编译出来安装后可以保存设置了。但是overlay还是告急，得想办法增加overlay空间。
[image]

pangfeng161, 2020-11-25:
> > ### df -h看占用
> > 
> > * 原因：设置还原大概率overlay空间占满/异常
> > * 修改：不要编译过多的东西/自行删除overlay中不需要的东西。如果硬件允许可以自行修改扩容overlayspace。
> 
> 我看了一下 确实是空间满了，后来编译的时候，选少一点插件，编译出来安装后可以保存设置了。但是overlay还是告急，得想办法增加overlay空间。
> [image]

pangfeng161, 2020-11-25:
> > > ### df -h看占用
> > > 
> > > * 原因：设置还原大概率overlay空间占满/异常
> > > * 修改：不要编译过多的东西/自行删除overlay中不需要的东西。如果硬件允许可以自行修改扩容overlayspace。
> > 
> > 
> > 我看了一下 确实是空间满了，后来编译的时候，选少一点插件，编译出来安装后可以保存设置了。但是overlay还是告急，得想办法增加overlay空间。
> > [image]

> > > ### df -h看占用
> > > 
> > > * 原因：设置还原大概率overlay空间占满/异常
> > > * 修改：不要编译过多的东西/自行删除overlay中不需要的东西。如果硬件允许可以自行修改扩容overlayspace。
> > 
> > 
> > 我看了一下 确实是空间满了，后来编译的时候，选少一点插件，编译出来安装后可以保存设置了。但是overlay还是告急，得想办法增加overlay空间。
> > [image]

编译固件时加大空间不久ok了

JohnGodzilla, 2020-12-04:
> 编译固件时加大空间不久ok了

我也遇到了同样的问题, 重启之后会恢复出厂设置.

查看过了空间占用看起来没有问题.
[image]

sxx1314, 2020-12-05:
> 我也遇到了同样的问题, 重启之后会恢复出厂设置.
>
> 查看过了空间占用看起来没有问题.
> [image]

XD,这叫没问题？你的 overlay都没挂载，东西都没地方写怎么给你保存设置！

JohnGodzilla, 2020-12-05:
> > > > > > ### df -h看占用
> > > > > > 
> > > > > > * 原因：设置还原大概率overlay空间占满/异常
> > > > > > * 修改：不要编译过多的东西/自行删除overlay中不需要的东西。如果硬件允许可以自行修改扩容overlayspace。
> > > > > 
> > > > > 
> > > > > 我看了一下确实是空间满了，后来编译的时候，选少一点插件，编译出来安装后可以保存设置了。但是overlay还是告急，得想办法增加overlay空间。
> > > > > [image]
> > > 
> > > 
> > > > > > ### df -h看占用
> > > > > > 
> > > > > > * 原因：设置还原大概率overlay空间占满/异常
> > > > > > * 修改：不要编译过多的东西/自行删除overlay中不需要的东西。如果硬件允许可以自行修改扩容overlayspace。
> > > > > 
> > > > > 
> > > > > 我看了一下确实是空间满了，后来编译的时候，选少一点插件，编译出来安装后可以保存设置了。但是overlay还是告急，得想办法增加overlay空间。
> > > > > [image]
> > > 
> > > 
> > > 编译固件时加大空间不久ok了
> > 
> > 
> > 我也遇到了同样的问题, 重启之后会恢复出厂设置.
> > 查看过了空间占用看起来没有问题.
> > [image]
> 
> XD,这叫没问题？你的 overlay都没挂载到硬盘上，东西都没地方写怎么给你保存设置！ tmp目录仅仅是ram掉电清空。

学到了,学到了.

woalbb, 2021-02-02:
我的wnr612v2现在只能用bb14.07的固件，3.3M大小的，安装完了剩余100多k，再加个upnp，中文都不能加，否则时间长了就完蛋了。更高版本的固件都大于3.3M，安装后也是无法保存任何设置，一重启就无效。

closed 2021-06-20 by coolsnowwolf