coolsnowwolf/lede #5829: Aria2不能保存下载记录，重启下载就没了

jxph1123, 2020-11-24:
配置和下载目录都改到硬盘了，也设置force-save=true了，记得以前的版本可以保存，同样的配置，目前的版本不能保存下载记录，重启下载就没了，有大佬碰到吗，能解决吗，谢谢！

gsblay, 2020-11-24:
我最新的git的版本没有问题，和以前的一样，都能保存的，重启也在，我是把配置文件写在files文件夹里进行编译的，自己看看权限对不对。

jxph1123, 2020-11-24:
> 我最新的git的版本没有问题，和以前的一样，都能保存的，重启也在，我是把配置文件写在files文件夹里进行编译的，自己看看权限对不对。

配置文件我是在设置界面设置到外挂硬盘的，请问下权限在哪里看呢？谢谢！

gsblay, 2020-11-24:
我是内置硬盘的，你可以用 WinSCP 来看！

jxph1123, 2020-11-25:
> 我是内置硬盘的，你可以用 WinSCP 来看！
用 WinSCP看了，权限是0644,下载列表只有在aria2.session.main__temp里面有，aria2.session.main里面没有东西，空的。我重启路由器和重新开启aria2软件时，aria2.session.main__temp自动清空了，下载列表也没了。也设置force-save=true了，aria2.session.main里面始终是空的，请问大侠该如何操作呢？谢谢！

gsblay, 2020-11-25:
外置硬盘里面应该有两个目录，一个是 aria2，我这里设置的权限是755；一个是用于存放下载文件的 Downloads，我设置的权限是777，注意这两个文件夹的拥有者应相同，并在 OP 里面存在。

jxph1123, 2020-11-25:
> 外置硬盘里面应该有两个目录，一个是 aria2，我这里设置的权限是755，里面应该有三个文件 ；一个是用于存放下载文件的 Downloads，我设置的权限是777，注意这两个文件夹的拥有者应相同，并在 OP 里面存在，并在 luci 界面配置好。 另外，在我的设置里，没有设置force-save=true。

完全按照你的操作方式，也去掉了force-save=true，还是不行。。。我准备重新编译一个试试。

gsblay, 2020-11-25:
原来内置的 ftp 最好选上并启用，我的两个文件夹的拥有者是内置的 ftp。

jxph1123, 2020-11-25:
ftp都是启用了。我以前的版本是可以的，目前的不行，配置完全一样，我也不知道哪里出问题了

jxph1123, 2020-11-25:
有个这个日志：
Wed Nov 25 10:53:12 2020 daemon.info aria2: Aria2 will run with user 'aria2'.
Wed Nov 25 10:53:12 2020 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /mnt/sda5/aria/downloads
Wed Nov 25 10:53:14 2020 daemon.err aria2c[25000]: Could not open /dev/crypto: Permission denied
Wed Nov 25 10:55:22 2020 daemon.info aria2: Instance "main" disabled.
Wed Nov 25 10:55:57 2020 daemon.info aria2: Aria2 will run with user 'aria2'.
Wed Nov 25 10:55:57 2020 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /mnt/sda5/aria/downloads
Wed Nov 25 10:55:57 2020 daemon.err aria2c[25658]: Could not open /dev/crypto: Permission denied
Wed Nov 25 11:01:01 2020 daemon.info aria2: Instance "main" disabled.
Wed Nov 25 11:01:11 2020 daemon.info aria2: Aria2 will run with user 'aria2'.
Wed Nov 25 11:01:11 2020 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /mnt/sda5/aria/downloads
Wed Nov 25 11:01:11 2020 daemon.err aria2c[26505]: Could not open /dev/crypto: Permission denied
Wed Nov 25 11:01:50 2020 daemon.info aria2: Instance "main" disabled.
Wed Nov 25 11:02:17 2020 daemon.info aria2: Aria2 will run with user 'aria2'.
Wed Nov 25 11:02:17 2020 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /mnt/sda5/aria/downloads
Wed Nov 25 11:02:18 2020 daemon.err aria2c[27036]: Could not open /dev/crypto: Permission denied
Wed Nov 25 11:07:44 2020 daemon.info aria2: Instance "main" disabled.
Wed Nov 25 11:08:24 2020 daemon.info aria2: Aria2 will run with user 'aria2'.
Wed Nov 25 11:08:24 2020 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /mnt/sda5/aria/downloads
Wed Nov 25 11:08:24 2020 daemon.err aria2c[27988]: Could not open /dev/crypto: Permission denied

gsblay, 2020-11-25:
你的 OP 里面有没有 aria2 这个用户，看日志好像这个用户不存在！

jxph1123, 2020-11-25:
直接选择root用户行不行？

gsblay, 2020-11-25:
你可以试试看选 root  如何，但不建议这样干！

jxph1123, 2020-11-25:
用root用户都不行，不知道出啥问题了。。。

gsblay, 2020-11-25:
[image]
这我就不清楚了，这是我的用户和权限设置，你参考一下！

jxph1123, 2020-11-25:
经过多次实验，问题解决，原因是配置文件设置不能设置到挂载的硬盘上，如果设置到硬盘上，会生成aria2.session.main_temp文件，下载记录保存在aria2.session.main_temp里，重启后消失。
将配置文件设置到路由器里，不会生成aria2.session.main_temp文件，下载记录会保存在aria2.session.main里，问题他解决，重启路由器下载列表还在。

gsblay, 2020-11-26:
pls closed!

closed 2020-12-07 by jxph1123

AdorableJiang, 2022-01-07:
问题修复
coolsnowwolf/packages#350