coolsnowwolf/lede #5831: 编译出错

Btivan, 2020-11-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
[x] 没有类似的issue

## 2. 详细叙述

find /home/ivan/lede/build_dir/target-x86_64_musl/root-x86/ -mindepth 1 -execdir touch -hcd "@1606108463" "{}" +
find: The relative path 'Files/WindowsApps/CanonicalGroupLimited.Ubuntu18.04onWindows_1804.2020.824.0_x64__79rhkp1fndgsc' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH

A：

### (2) 路由器型号和固件版本

A：软路由86

### (3) 详细日志

A：sed -i "s/Installed-Time: .*/Installed-Time: 1606108463/" /home/ivan/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/status
rm -rf /home/ivan/lede/build_dir/target-x86_64_musl/root-x86/tmp/*
rm -f /home/ivan/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/lists/*
rm -f /home/ivan/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/info/*.postinst*
rm -f /home/ivan/lede/build_dir/target-x86_64_musl/root-x86/var/lock/*.lock
rm -rf /home/ivan/lede/build_dir/target-x86_64_musl/root-x86/boot
find /home/ivan/lede/build_dir/target-x86_64_musl/root-x86/ -mindepth 1 -execdir touch -hcd "@1606108463" "{}" +
find: The relative path 'Files/WindowsApps/CanonicalGroupLimited.Ubuntu18.04onWindows_1804.2020.824.0_x64__79rhkp1fndgsc' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 1
make[2]: Leaving directory '/home/ivan/lede'
package/Makefile:108: recipe for target '/home/ivan/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/ivan/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/ivan/lede'
/home/ivan/lede/include/toplevel.mk:222: recipe for target 'world' failed

jjit, 2020-11-25:
微软的 Linux子系统 吗
op官网：https://openwrt.org/docs/guide-developer/build-system/wsl
在 Linux子系统 里的 etc 目录里 创建 wsl.conf 文件，在文件里添加 2行
[interop]
appendWindowsPath = false
然后 改一下 win系统的 注册表
HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Lxss\{GUID}\Flags
把 7 改成 5
然后重启一下win系统

Btivan, 2020-11-25:
> 微软的 Linux子系统 吗
> op官网：https://openwrt.org/docs/guide-developer/build-system/wsl
> 在 Linux子系统 里的 etc 目录里 创建 wsl.conf 文件，在文件里添加 2行
> [interop]
> appendWindowsPath = false
> 然后 改一下 win系统的 注册表
> HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Lxss{GUID}\Flags
> 把 7 改成 5
> 然后重启一下win系统

好，谢谢，试试看。

Tzwcard, 2020-11-26:
> 微软的 Linux子系统 吗
> op官网：https://openwrt.org/docs/guide-developer/build-system/wsl
> 在 Linux子系统 里的 etc 目录里 创建 wsl.conf 文件，在文件里添加 2行
> [interop]
> appendWindowsPath = false
> 然后 改一下 win系统的 注册表
> HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Lxss{GUID}\Flags
> 把 7 改成 5
> 然后重启一下win系统

实测可用
我不想重启 就把LxssManager服务重新启动了一下也没问题

zzzal, 2020-11-26:
同样的报错，路由器是 k2p，系统是 Ubuntu18.04 server（aws）

jaynfs, 2021-02-26:
原来是wsl的问题

closed 2021-06-20 by coolsnowwolf

Bearlele, 2022-08-23:
改注册表之后子系统自动重置了咋办qwq