coolsnowwolf/lede #5832: 选择ffmpeg编译时出错

Tzwcard, 2020-11-26:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：默认选项编译通过，Multimedia > ffmpeg 选上之后再编译会出错。

### (2) 路由器型号和固件版本

A：NanoPi-R2S / gitver: bcbce88ad3ca6c03f915acb8a5a36b435b07ff15

### (3) 详细日志

A：
```Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ffmpeg:
 *      alsa-lib
 *      libgnutls
 *      lame-lib
 *      fdk-aac
 * opkg_install_cmd: Cannot install package ffmpeg.
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavcodec.so.58
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavcodec.so.58.91.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavdevice.so.58
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavdevice.so.58.10.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavformat.so.58
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavformat.so.58.45.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavutil.so.56
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavutil.so.56.51.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * opkg_install_cmd: Cannot install package ffprobe.
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavcodec.so.58
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavcodec.so.58.91.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavdevice.so.58
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavdevice.so.58.10.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavformat.so.58
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavformat.so.58.45.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavutil.so.56
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavutil.so.56.51.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * opkg_install_cmd: Cannot install package libffmpeg-full.
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavcodec.so.58
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavcodec.so.58.91.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavdevice.so.58
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavdevice.so.58.10.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavformat.so.58
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavformat.so.58.45.100
[... 6 lines cut ...]
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavcodec.so.58
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavcodec.so.58.91.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavdevice.so.58
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavdevice.so.58.10.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavformat.so.58
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavformat.so.58.45.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavutil.so.56
        But that file is already provided by package  * libffmpeg-audio-dec
 * check_data_file_clashes: Package libffmpeg-full wants to install file /home/user/lede/build_dir/target-aarch64_generic_musl/root-rockchip/usr/lib/libavutil.so.56.51.100
        But that file is already provided by package  * libffmpeg-audio-dec
 * opkg_install_cmd: Cannot install package libffmpeg-full.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/user/lede'
package/Makefile:108: recipe for target '/home/user/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_install' failed
make[1]: *** [/home/user/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/user/lede'
/home/user/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2```

minlang112, 2020-11-26:
But that file is already provided by package  * libffmpeg-audio-dec

yonggedebaqi, 2020-11-29:
But that file is already provided by package  * 
libffmpeg-audio-dec 
libffmpeg-full
二选一，我是选libffmpeg-full

Tzwcard, 2020-11-29:
> But that file is already provided by package *
> libffmpeg-audio-dec
> libffmpeg-full
> 二选一，我是选libffmpeg-full

原来是这样 我还以为ffmpeg需要libffmpeg所以没敢去动他

closed 2020-11-29 by Tzwcard