coolsnowwolf/lede #5833: 高并发请求下，提示DNS异常，请问是否有类似防火墙机制？

aaron021512, 2020-11-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我用10台手机跑高并发测试大概20分钟，请求同一个域名（已排除域名本身可能的问题，而且在其他网络下尝试过10台同样的机子测试是通过的），跑到3分钟左右（3分钟25000次请求左右），就全部报错显示“Unable to resolve host xxxx.com,No address associated with hostname”，像是DNS解析出了问题，多次尝试复现同样的问题。

已排除之前的电信天翼路由器问题，之前怀疑是电信前端路由有防火墙，然后让LEDE直接pppoe拨号，同样出现。
已排除AP wifi问题，使用ubnt AC-PRO，带10台机子完全没压力，咨询过官方技术，没有任何这方面的限制。

所以目前只能排查LEDE是否存在类似的防火墙了？

还发现一个现象，看看能不能作为线索，LEDE首页 “活跃连接数” 没有出错的时候正常维持在5500个左右，CPU负荷在0.6-0.8，一旦提示错误后，连接数会慢慢飙升到3w多，导致整个路由器无法上网，推测应该是由于请求失败，导致请求积累越来越多，然后把路由器给拖死了。

### (2) 路由器型号和固件版本

A：软路由Intel(R) Celeron(R) CPU 3865U @ 1.80GHz 4G内存，
固件版本：Openwrt Koolshare mod V2.27 r10054-b7cccc1136
内核版本：4.14.96

### (3) 详细日志

A：没有发现有异常的日志

minlang112, 2020-11-26:
你们抓的是周树人,和我鲁迅有什么关系? Koolshare当然应该去问Koolshare了……

Cathgao, 2020-11-27:
Koolshare魔改来lean问我是没想到的

Myoko, 2020-11-27:
跑错场了

aaron021512, 2020-12-08:
> 你们抓的是周树人,和我鲁迅有什么关系? Koolshare当然应该去问Koolshare了……

emmm...请问Koolshare的iuuse我没有搜到，在哪里可以提问啊？所以我找到了openwrt..我想是一家的么..

LGA1150, 2020-12-23:
网络 - DHCP/DNS 有个最大并发 DNS 请求数，默认是每秒 150 个

closed 2021-06-20 by coolsnowwolf