coolsnowwolf/lede #5838: 有没有办法把zerotier的流量转发到旁路由？

NagaseKouichi, 2020-11-28:
现在是用硬路由刷openwrt做主路由，NAS上虚拟了openwrt做旁路由跑ssr等服务
但是并没有用旁路由做默认网关（因为旁路由做网关的话有些设备的upnp会有问题），只是用主路由的SSR套娃连接旁路由的SSR

如果用性能比较差的主路由提供zerotier服务，会发现连接zerotier虚拟网络里面的外网电脑的速度比较慢，ping值能到300多ms，但是如果旁路由提供zerotier服务并做网关，则ping外网电脑只有5、6ms。
那么有没有办法类似SSR套娃那样，旁路由提供zerotier的虚拟组网服务，然后把走zerotier的流量通过主路由转发到旁路由上呢？因为不想用旁路由做设备的网关，如果旁路由挂了会影响设备联网

coolsnowwolf, 2020-11-28:
什么是旁路由？

NagaseKouichi, 2020-11-28:
> 什么是旁路由？

大佬你好，像SSR套娃这种方式，实际运行V2RAY之类服务的路由器，如果不做网关的话，不叫旁路由吗？

> 大雕今天更新了什么
> @全体成员 今天的源代码已更新： XXR Plus+ 奇怪的技能又增加了！
> 
> 支持分流重定向流量到某个特定的网络接口上（例如 WireGuard, 思科 Any connect， Zerotier ， IPSEC 。。。等等）
> 
> 这些会生成一个虚拟的 Tun 隧道接口，现在 XXR Plus+ 可以根据规则将例如国外的流量分到这个接口上啦~~
> 
> 应该有人用的上的，Enjoy~~

看更新日志里面有一条这个，这个功能可以实现吗？

mjyhj, 2020-11-29:
300多ms是因为你打洞失败了，重新连接一次就可以了

NagaseKouichi, 2020-11-29:
> 300多ms是因为你打洞失败了，重新连接一次就可以了

不是的，同样的网络，用硬路由连zerotier就很慢（但是也是直连，没有走国外服务器），用虚拟的软路由连就很快，我猜是zerotier的流量也需要用AES加密

ZqinKing, 2020-12-11:
> 现在是用硬路由刷openwrt做主路由，NAS上虚拟了openwrt做旁路由跑ssr等服务
> 但是并没有用旁路由做默认网关（因为旁路由做网关的话有些设备的upnp会有问题），只是用主路由的SSR套娃连接旁路由的SSR [...]

你直接在旁路有上开zerotier不就行了吗？再做几条静态路由。

NagaseKouichi, 2020-12-15:
> 300多ms是因为你打洞失败了，重新连接一次就可以了

找到原因了，确实应该是打洞失败了，需要在防火墙里面把WAN口入站和转发改为接受，才能打洞直连成功

closed 2020-12-15 by NagaseKouichi

cdliaozhi, 2022-04-22:
> > 300多ms是因为你打洞失败了，重新连接一次就可以了
> 
> 找到原因了，确实应该是打洞失败了，需要在防火墙里面把WAN口入站和转发改为接受，才能打洞直连成功

我是通过硬路由直连快，通过旁路由就慢。能请你详细说下如何设置防火墙吗？