coolsnowwolf/lede #5849: nlbwmon报错

ramondelee, 2020-11-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：nlbwmon报错

### (2) 路由器型号和固件版本

A：R2S，R20.10.20

### (3) 详细日志

A：
Mon Nov 30 11:26:15 2020 daemon.err nlbwmon[1527]: Netlink receive failure: Out of memory
Mon Nov 30 11:26:15 2020 daemon.err nlbwmon[1527]: Unable to dump conntrack: No buffer space available
Mon Nov 30 14:13:59 2020 daemon.err nlbwmon[1527]: Netlink receive failure: Out of memory
Mon Nov 30 14:13:59 2020 daemon.err nlbwmon[1527]: Unable to dump conntrack: No buffer space available
Mon Nov 30 17:01:40 2020 daemon.err nlbwmon[1527]: Netlink receive failure: Out of memory
Mon Nov 30 17:01:40 2020 daemon.err nlbwmon[1527]: Unable to dump conntrack: No buffer space available
Mon Nov 30 19:49:22 2020 daemon.err nlbwmon[1527]: Netlink receive failure: Out of memory
Mon Nov 30 19:49:22 2020 daemon.err nlbwmon[1527]: Unable to dump conntrack: No buffer space available

martin-xj, 2020-12-01:
同样的问题，不知什么原因。

ZqinKing, 2020-12-03:
netlink报错的话就是驱动kmalloc内存的时候超了。老版本内核中驱动申请内存是由上限的。不过照理说现在至少也是4.x的内核，上限很高才对。

ramondelee, 2020-12-04:
> 同样的问题，不知什么原因。

老兄，你是X86，还是与我一样ARM或R2S？

Neverends4, 2021-01-12:
遇到过，我是MT7621，搜了下貌似是配置问题，ssh进后台/etc/config/nlbwmon 添加：
`	option netlink_buffer_size '5242880'`

WYC-2020, 2021-02-21:
就是这个错误一直都有在哪里不断的刷，一会一次，坐等大佬修复

tanst, 2021-02-28:
> > 同样的问题，不知什么原因。
> 
> 老兄，你是X86，还是与我一样ARM或R2S？

我是 x86 物理内存 2g，也有这种日志，统计大概 2 个月后出现

closed 2021-06-20 by coolsnowwolf