coolsnowwolf/lede #5854: Intel网卡的E1000e驱动需要更新

xinyl189, 2020-12-01:
1.关于你要提交的问题
路由器硬件为联想M93，物理机直接跑openwrt。网卡为Intel 的I217-LM网卡，在经过大流量的时候这个网卡就会down掉，并且出现e1000e 0000:00:19.0 eth0: Reset adapter unexpectedly日志。

Q：是否搜索了issue (使用 "x" 选择)

* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
Intel 的I217-LM网卡在流量大的时候，就出现就会出现e1000e 0000:00:19.0 eth0: Reset adapter unexpectedly日志。
goole之后，得知，该驱动有问题。
源的版本为3.2.6-k，最新的驱动版本为3.8。

在issue里有人提到I219-V的网卡也依赖新的e1000e的驱动。

### (2) 路由器型号和固件版本

A：联想M93p
编译的固件版本为OpenWrt R20.10.20 / LuCI Master (git-20.256.12360-1a54222)
### (3) 详细日志

A：
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192] e1000e 0000:00:19.0 eth0: Detected Hardware Unit Hang:
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192]   TDH                  <11>
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192]   TDT                  <18>
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192]   next_to_use          <18>
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192]   next_to_clean        <f>
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192] buffer_info[next_to_clean]:
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192]   time_stamp           <ffffcf56>
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192]   next_to_watch        <11>
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192]   jiffies              <ffffd1f0>
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192]   next_to_watch.status <0>
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192] MAC Status             <80083>
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192] PHY Status             <796d>
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192] PHY 1000BASE-T Status  <7800>
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192] PHY Extended Status    <3000>
Tue Dec  1 16:54:59 2020 kern.err kernel: [  252.733192] PCI Status             <10>
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952] e1000e 0000:00:19.0 eth0: Detected Hardware Unit Hang:
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952]   TDH                  <11>
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952]   TDT                  <18>
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952]   next_to_use          <18>
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952]   next_to_clean        <f>
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952] buffer_info[next_to_clean]:
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952]   time_stamp           <ffffcf56>
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952]   next_to_watch        <11>
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952]   jiffies              <ffffd3e8>
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952]   next_to_watch.status <0>
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952] MAC Status             <80083>
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952] PHY Status             <796d>
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952] PHY 1000BASE-T Status  <7800>
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952] PHY Extended Status    <3000>
Tue Dec  1 16:55:01 2020 kern.err kernel: [  254.748952] PCI Status             <10>
Tue Dec  1 16:55:02 2020 daemon.notice netifd: Network device 'eth0' link is down
Tue Dec  1 16:55:02 2020 kern.err kernel: [  255.868457] e1000e 0000:00:19.0 eth0: Reset adapter unexpectedly
Tue Dec  1 16:55:02 2020 kern.info kernel: [  255.869823] br-lan: port 1(eth0) entered disabled state
Tue Dec  1 16:55:03 2020 daemon.notice netifd: bridge 'br-lan' link is down
Tue Dec  1 16:55:03 2020 daemon.notice netifd: Interface 'lan' has link connectivity loss
Tue Dec  1 16:55:05 2020 kern.info kernel: [  259.314372] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
Tue Dec  1 16:55:05 2020 kern.info kernel: [  259.315776] br-lan: port 1(eth0) entered blocking state
Tue Dec  1 16:55:05 2020 kern.info kernel: [  259.317104] br-lan: port 1(eth0) entered forwarding state
Tue Dec  1 16:55:05 2020 daemon.notice netifd: Network device 'eth0' link is up
Tue Dec  1 16:55:06 2020 uucp.err /usr/sbin/noddos[5516]: Noddos: Epoll event error for FD 8
Tue Dec  1 16:55:06 2020 uucp.err /usr/sbin/noddos[5516]: Noddos: Closing file description without re-opening it 8
Tue Dec  1 16:55:06 2020 uucp.err /usr/sbin/noddos[5516]: Noddos: Epoll event error for FD 6
Tue Dec  1 16:55:06 2020 uucp.err /usr/sbin/noddos[5516]: Noddos: Closing file description without re-opening it 6
Tue Dec  1 16:55:06 2020 uucp.err /usr/sbin/noddos[5516]: Noddos: Epoll event error for FD 8
Tue Dec  1 16:55:06 2020 uucp.err /usr/sbin/noddos[5516]: Noddos: Closing file description without re-opening it 8
[... 176 lines cut ...]
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416]   TDH                  <aa>
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416]   TDT                  <e4>
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416]   next_to_use          <e4>
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416]   next_to_clean        <a8>
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416] buffer_info[next_to_clean]:
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416]   time_stamp           <10002da70>
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416]   next_to_watch        <aa>
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416]   jiffies              <10002e318>
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416]   next_to_watch.status <0>
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416] MAC Status             <80083>
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416] PHY Status             <796d>
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416] PHY 1000BASE-T Status  <7800>
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416] PHY Extended Status    <3000>
Tue Dec  1 17:08:23 2020 kern.err kernel: [ 1056.716416] PCI Status             <10>
Tue Dec  1 17:08:24 2020 kern.err kernel: [ 1057.900006] e1000e 0000:00:19.0 eth0: Reset adapter unexpectedly
Tue Dec  1 17:08:24 2020 kern.info kernel: [ 1057.902017] br-lan: port 1(eth0) entered disabled state
Tue Dec  1 17:08:24 2020 daemon.notice netifd: Network device 'eth0' link is down
Tue Dec  1 17:08:25 2020 daemon.notice netifd: bridge 'br-lan' link is down
Tue Dec  1 17:08:25 2020 daemon.notice netifd: Interface 'lan' has link connectivity loss
Tue Dec  1 17:08:28 2020 daemon.notice netifd: Network device 'eth0' link is up
Tue Dec  1 17:08:28 2020 kern.info kernel: [ 1062.358411] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
Tue Dec  1 17:08:28 2020 kern.info kernel: [ 1062.360502] br-lan: port 1(eth0) entered blocking state
Tue Dec  1 17:08:28 2020 kern.info kernel: [ 1062.361963] br-lan: port 1(eth0) entered forwarding state
Tue Dec  1 17:08:29 2020 daemon.notice netifd: bridge 'br-lan' link is up
Tue Dec  1 17:08:29 2020 daemon.notice netifd: Interface 'lan' has link connectivity

coolsnowwolf, 2020-12-02:
这个和 EPPROM 版本有关

xinyl189, 2020-12-02:
感谢L大的回复邮件。
那这个问题，除了使用ethtool -K eth0 tso off之外，还有别的解决办法吗？
期盼您的回复。，多谢。

coolsnowwolf, 2020-12-02:
Intel 自己的硬件Bug，从 82574L 开始就这样了

xinyl189, 2020-12-02:
感谢大佬回复。
还有个问题困扰着我：
我有2个小主机，一个用的是intel 的I219LM的网卡，这主机个完全正常，不需要使用ethtool -K eth0 tx off rx off关闭校验完全正常的。
另一个小主机是M93P用的intel 的I217-LM网卡，这个必须要关闭校验才能正常使用。
对比2个网卡，I219的firmware-version是 0.13-4，而I217的firmware-version是 0.12-4。
难道是网卡的firmware-version导致的问题吗？
不知道网卡的firmware-version能不能升级，我尝试用intel的网卡驱动程序，但是提示集成网卡不能升级firmware-version。

多谢。

Dimitri2020007, 2020-12-05:
集成网卡的firmware是已经塞进主板bios里的，你说的方法是刷不进去的。除非能找到机型的bios升级程序（必须大于现在的版本或者同等，太低要编程器。）

1.提取出bios主文件。找到网卡的pci-id，和主bios的网卡pci-id做下对比。

2.使用bios修改工具将新网卡bios和旧网卡bios替换下。（这步最危险，需要非常的仔细小心。我曾经就是这步刷坏了主板。）

3.刷入就好（必须保证不要停电！这步也是危险的。刷的中间停电了就gg了。）

是否有效请看看网卡的firmware版本是不是更新了。

有可能会失败。（因为工厂在主板的bios文件上上锁了，像华硕uefi bios的主板就是上了锁的。不排除品牌机也是这样子。根本还是防止用户修改。）

closed 2021-06-20 by coolsnowwolf