coolsnowwolf/lede #5865: XRay 是 V2RAY 是否合并为单个 BIN 的问题

coolsnowwolf, 2020-12-02:
目前来看， XRAY 是 V2RAY 的增集

但不排除将来 V2RAY 会和 XRAY 分开，就像 SS 和 SSR 的关系一样

大家觉得应该为了节省 ROM 空间合并为单个 BIN，还是保持独立呢？

tinyxingqiu, 2020-12-02:
我认为应该合并，因为两者本身就是同一类的功能，两个项目还有借鉴关系，只是内部实现不一样。

xianren78, 2020-12-02:
实际上路由器ROM可用的肯定是XRAY走在前面了，有XTLS加成，有SPLICE加成。计划中XRAY-NEXT简化增强架构后更为强大。

qqqwqk, 2020-12-02:
目前就使用上来看X完全可以胜任V的功能

qqqwqk, 2020-12-02:
目前就使用上来看X完全可以胜任V的功能

qqqwqk, 2020-12-02:
目前就使用上来看X完全可以胜任V的功能

qqqwqk, 2020-12-02:
目前就使用上来看X完全可以胜任V的功能

hackyes, 2020-12-02:
个人主为不必合并，xray对于低配路由比v2ray提升了不少。再加上提供完整的vless，也就是说未来v2ray只用来跑vmess了，vless还是xray领先

hetao8980823, 2020-12-02:
支持合并xray 有v2ray 所有功能，合并可以节约空间。

gzcrayon, 2020-12-02:
目前已经分开，而且后续的发展方向应该会不一样，建议分开

hxlls, 2020-12-02:
个人认为分开是必要的，现在看来后期发展会不一样，分开更合理，至于容量，用不到的不选就完事了。好像没啥纠结的

eallion, 2020-12-02:
建议分开。
PS：单是为了节省空间也应该分开，只编译自己需要的协议。

Beginner-Go, 2020-12-02:
建议分开 说不定以后xray不兼容v2ray呢(就像谁能想到和v2ray分家呢)，如果不用v2ray大可不编译v2ray

HoldOnBro, 2020-12-02:
分开

sbily1988, 2020-12-02:
建议分开，已经是两个不同的项目了

4860575, 2020-12-02:
xray兼容常用的v2ray功能。作为路由器使用个人觉得合并比较好

lanfenglin, 2020-12-02:
建议分开比较好，喜欢哪个选哪个，不费劲！

xiaobaishu1, 2020-12-02:
既然是V2的进化版本 那它有的功能都有 比V2还强 为什么不直接替换V2就好了？

hxlls, 2020-12-02:
> 既然是V2的进化版本 那它有的功能都有 比V2还强 为什么不直接替换V2就好了？

分支不是进化，分支和主干是并行关系，现在xray 有v2 功能不代表以后都有，v2也会持续发展

ihuaer, 2020-12-03:
目前建议还是分开比较好，就好比lede和openwrt 后续怎么样很难说

lqhorochi, 2020-12-03:
可以添加Trojan go吗???

kslr, 2020-12-03:
不要合并，它们是完全独立的项目。

Fyred, 2020-12-03:
都不是一帮人开发的两个软件，合什么并？谁知道以后会怎样

chpos, 2020-12-03:
建议分开，两个软件应该会有不同的发展方向

jianyun8023, 2020-12-03:
建议分开，两个软件的方向不一样。配置很有可能会变得不兼容。

1715173329, 2020-12-04:
xray 早些时候就表明了
完全兼容 v2ray，并跟随 v2ray 更新
xray 本质上即为 v2ray + xtls support

mcbbsandnz, 2020-12-05:
建议分开,Xray作者已经把xtls给分出来了,很有可能要自己发展成一个独立分支

wsa01378, 2020-12-06:
分开吧，看看ss和ssr。基本互相敌对
日后估计xray不再支持v2的特性。

PandaEgg, 2020-12-07:
已经合并了吗？

mcbbsandnz, 2020-12-07:
> 已经合并了吗?

还没有,现在暂时是两个项目

mjyhj, 2020-12-08:
啥都不建议，直接弄3个自己选。
A B A+B

rrrwwwkkk, 2020-12-09:
支持分开，需要哪个编译哪个。

bilibilistack, 2020-12-10:
支持分开，需要哪个用哪个

dingpengyu, 2020-12-11:
不推荐合并。

S0MEW0LF, 2020-12-11:
Xray项目那里写着完全兼容v2ray的，所以我觉得保留Xray就足够了

mcucore, 2020-12-16:
保留独立，默认使用XRAY！有特别需求的可以自己勾选。

DanielChen008, 2020-12-22:
不合并，未来两家的走向不明朗

31415926pai, 2020-12-24:
不要合并。
其实大部分不会纠结选哪个，二个都选择。有些路由器用户会纠结一下，毕竟存储有点小。

delury, 2020-12-26:
分成2个吧，爱用那个用那个。

xiaocenter, 2021-01-14:
现在一个问题是 16+128的路由上 勾选v2不勾选xray 可以出包（小于16M)  可是勾选XRAY 不勾选V2 编译出来还是包括了v2  包大于16  无法出包   是否BUG  ？

closed 2021-06-20 by coolsnowwolf