coolsnowwolf/lede #5867: 使用Actions编译时出现Haproxy错误，小白求解。

alvazhang79, 2020-12-03:
反馈bug/问题模板，提建议请删除

## 1.使用P3TERX大大的Actions编译Lede，haproxy出错，求解。

已经搜多过Issues，无解。

## 2. 详细叙述

### (1) 具体问题

A：加入了https://github.com/kenzok8/openwrt-packages与https://github.com/kenzok8/small两个源，两次编译都出错，两次时间时12月02日18点，12月03日00点，使用相同的.config与config-5.4的配置文件。两次都是同样的错误，但奇怪的是，同样这两个配置文件，在12月02日10点时编译居然通过了，但由于Actions最后提示空间不足无法上传固件而没有获得最终固件。所以才进行了后面两次编译。错误信息见（3）

### (2) 路由器型号和固件版本

A：X86-64

### (3) 详细日志

A： make[3] -C feeds/packages/net/haproxy compile
make: *** [world] Error 1
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
 make[1] world
 make[2] target/compile
 make[3] -C target/linux compile
 make[2] buildinfo
 make[3] diffconfig buildversion feedsversion
 make[2] package/cleanup
 make[2] package/compile
 make[3] -C package/libs/libjson-c host-compile

##这里还编译了很多内容##

make[3]: Entering directory '/workdir/openwrt/feeds/packages/net/haproxy'
touch /workdir/openwrt/build_dir/target-x86_64_musl/haproxy-ssl/haproxy-2.2.4/.prepared_bc03b8d7dfe5228eb102d2cca0cef341_6664517399ebbbc92a37c5bb081b5c53_check
. /workdir/openwrt/include/shell.sh; gzip -dc /workdir/openwrt/dl/haproxy-2.2.4.tar.gz | tar -C /workdir/openwrt/build_dir/target-x86_64_musl/haproxy-ssl/haproxy-2.2.4/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /workdir/openwrt/build_dir/target-x86_64_musl/haproxy-ssl/haproxy-2.2.4

Applying ./patches/000-OPENWRT-BUILD-make-dladdr1-depend-on-glibc-version-and-not-__USE_GNU.patch using plaintext: 
can't find file to patch at input line 15
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|commit 62af9c83f9ed2b25e0061798e29e3cccfce5fbdc
|Author: Willy Tarreau <w@1wt.eu>
|Date:   Tue Mar 10 07:51:48 2020 +0100
|
|    BUILD: make dladdr1 depend on glibc version and not __USE_GNU
|    
|    Technically speaking the call was implemented in glibc 2.3 so we must
|    rely on this and not on __USE_GNU which is an internal define of glibc
|    to track use of GNU_SOURCE.
|
|diff --git a/src/standard.c b/src/standard.c
|index e0ea8328e..d16eebfea 100644
|--- a/src/standard.c
|+++ b/src/standard.c
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch failed!  Please fix ./patches/000-OPENWRT-BUILD-make-dladdr1-depend-on-glibc-version-and-not-__USE_GNU.patch!
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/haproxy-ssl/haproxy-2.2.4/.prepared_bc03b8d7dfe5228eb102d2cca0cef341_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[2]: *** [package/feeds/packages/haproxy/compile] Error 2
Makefile:152: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/haproxy-ssl/haproxy-2.2.4/.prepared_bc03b8d7dfe5228eb102d2cca0cef341_6664517399ebbbc92a37c5bb081b5c53' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/haproxy'
make: *** [world] Error 2
time: package/feeds/packages/haproxy/ssl/compile#0.22#0.08#0.27
package/Makefile:111: recipe for target 'package/feeds/packages/haproxy/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
Error: Process completed with exit code 2.

WYC-2020, 2020-12-03:
del this path,because new version is fix bug

asushugo, 2020-12-03:
参考这个https://github.com/coolsnowwolf/packages/pull/102
更新了即可

alvazhang79, 2020-12-03:
> del this patch,because new version is fix bug

请问，具体要删除哪个？谢谢

alvazhang79, 2020-12-03:
> 参考这个[coolsnowwolf/packages#102](https://github.com/coolsnowwolf/packages/pull/102)
> 更新了即可

感谢！

WYC-2020, 2020-12-03:
删除补丁文件即可，但是必须保证版本再2.2.4上面 因为2.2.4上面已经没有那个stand.c文件 并且已经修复了这个bug 所以升不升级都无所谓，只要删除那个补丁文件即可

alvazhang79, 2020-12-03:
> 删除补丁文件即可，但是必须保证版本再2.2.4上面 因为2.2.4上面已经没有那个stand.c文件 并且已经修复了这个bug 所以升不升级都无所谓，只要删除那个补丁文件即可

感谢！一会试试！

closed 2020-12-03 by coolsnowwolf