coolsnowwolf/lede #5871: 是否应该使用  trojan-go 替代  trojan ？

coolsnowwolf, 2020-12-03:
trojan-go 的目前发展的速度，功能，和特性都已经完全超越 Trojan，唯一考虑的也就是 Golang 程序的体积了

是否应该开始启用 使用  trojan-go 替代  trojan 呢？

（ 16MB ROM 路由器手动再见吧？）

hcym, 2020-12-03:
速度快的话就取代吧，没用过，都是原版的，感觉c不应该比go的慢吧？

Beginner-Go, 2020-12-03:
虽然没用过trojan 你说trojan-go超越了  那就替换吧

ZqinKing, 2020-12-03:
> 速度快的话就取代吧，没用过，都是原版的，感觉c不应该比go的慢吧？

原版没有设计多线程

hackyes, 2020-12-03:
可以添加上去啊，要什么就手动编译，让用户自己选择

ywyha, 2020-12-03:
早就希望有这么一天，早就应该替换了

accors, 2020-12-03:
可以加一个，让用户选

lanfenglin, 2020-12-03:
可以加上，让用户选择。

Fyred, 2020-12-03:
当然用go了，我vps上都换go了，爽的一比

lifetyper, 2020-12-03:
加上去，然后让用户自己选择编译就行了。

Lrvineshaw, 2020-12-03:
低端硬路由还是别用go了吧，C语言还是优先

realsteam, 2020-12-03:
加上去默认不勾选，让编译的用户自己决定是否用

lqhorochi, 2020-12-03:
加上去,用户自主选择添加插件

guochan2019, 2020-12-04:
同意增加或替换

aueu, 2020-12-04:
已经采用多个go程序了
这下子，32M rom路由也要拜拜

lqhorochi, 2020-12-04:
OpenWrt的Trojan go插件项目
https://github.com/frainzy1477/luci-app-trojan

Myoko, 2020-12-04:
GO 可以替换 trojan

Cathgao, 2020-12-04:
go兼容trojan，应该替换

BOBINIUNIU, 2020-12-05:
看过一个内网环境测试，go的性能 比 trojan无论是单核还是多核都快了将近一倍。支持替换。
现在一个路由器才多少钱，抱着32M不放的是准备当传家宝吗

mcbbsandnz, 2020-12-05:
我觉得可以加上去让用户来自己选择,trojan-go的性能比原版强很多

dingpengyu, 2020-12-06:
> 我觉得可以加上去让用户来自己选择,trojan-go的性能比原版强很多

支持

eallion, 2020-12-07:
可以替换，可以共存，但应该要有 Trojan Go 的选项。

Ai1954, 2020-12-08:
同意替换，非常赞成

one02free, 2020-12-08:
同意替换，非常赞成

qqhpc, 2020-12-09:
支持替换

Marcotso, 2020-12-11:
支持替换

kenyloveg, 2020-12-14:
支持替换

mcucore, 2020-12-16:
换吧，老旧路由就默认别选上TROJAN-GO。

Gokou-Ruri, 2020-12-16:
支持替换，空间我觉得不是问题，可以overlay嘛，想要多大都行。

madogao, 2020-12-17:
支持替换，我现在是拿trojan-go直接替换的trojan可执行文件

31415926pai, 2020-12-24:
关键是GO可以用CDN，速度再快，网速就是那么多。1000M的宽带+1G的VPS，TROJAN也能跑满。主要是能用CDN

31415926pai, 2020-12-24:
其实大部分人不会纠结选择哪一个，用路由器的因为存储不够，会纠结选择哪个。

shu0802, 2020-12-29:
支持更新

lqhorochi, 2021-01-08:
直接支持更新Trojan Go吧....稳如狗...

ywyha, 2021-01-08:
大佬，快点替换啊，等好久了……

xmapst, 2021-01-11:
支持更新

pop2006, 2021-01-16:
为什么不支持呢？编译的时候完全可以自己选择啊。

wujunze, 2021-01-18:
可以加一个，让用户选

ywyha, 2021-01-19:
@coolsnowwolf  大佬，我看已经加入了trajon-go 可是 MUX选项怎么没有了？……

ywyha, 2021-01-19:
问下大家的trajon-go的MUX选项 还在吗……

closed 2021-06-20 by coolsnowwolf