coolsnowwolf/lede #5873: IPsec无法启动连接，老版没问题，5.4就有这问题。

kkxkx, 2020-12-03:
日志
Thu Dec  3 21:50:46 2020 daemon.info : 00[DMN] Starting IKE charon daemon (strongSwan 5.8.4, Linux 5.4.75, x86_64)
Thu Dec  3 21:50:46 2020 daemon.info : 00[LIB] failed to open /dev/net/tun: No such file or directory
Thu Dec  3 21:50:46 2020 daemon.info : 00[KNL] failed to create TUN device
Thu Dec  3 21:50:46 2020 daemon.info : 00[LIB] plugin 'kernel-libipsec': failed to load - kernel_libipsec_plugin_create returned NULL
Thu Dec  3 21:50:46 2020 daemon.info : 00[KNL] unable to create IPv4 routing table rule
Thu Dec  3 21:50:46 2020 daemon.info : 00[KNL] unable to create IPv6 routing table rule
Thu Dec  3 21:50:46 2020 daemon.info : 00[CFG] loading ca certificates from '/etc/ipsec.d/cacerts'
Thu Dec  3 21:50:46 2020 daemon.info : 00[CFG] loading aa certificates from '/etc/ipsec.d/aacerts'
Thu Dec  3 21:50:46 2020 daemon.info : 00[CFG] loading ocsp signer certificates from '/etc/ipsec.d/ocspcerts'
Thu Dec  3 21:50:46 2020 daemon.info : 00[CFG] loading attribute certificates from '/etc/ipsec.d/acerts'
Thu Dec  3 21:50:46 2020 daemon.info : 00[CFG] loading crls from '/etc/ipsec.d/crls'
Thu Dec  3 21:50:46 2020 daemon.info : 00[CFG] loading secrets from '/etc/ipsec.secrets'
Thu Dec  3 21:50:46 2020 daemon.info : 00[CFG]   loaded IKE secret for %any
Thu Dec  3 21:50:46 2020 daemon.info : 00[CFG]   loaded EAP secret for lean
Thu Dec  3 21:50:46 2020 daemon.info : 00[LIB] loaded plugins: charon aes sha1 random nonce x509 pubkey gmp xcbc hmac kernel-netlink socket-default stroke updown xauth-generic
Thu Dec  3 21:50:46 2020 daemon.info : 00[JOB] spawning 16 worker threads
Thu Dec  3 21:50:46 2020 daemon.info : 05[DMN] thread 5 received 11
Thu Dec  3 21:50:46 2020 daemon.info : 05[LIB] no support for capturing backtraces
Thu Dec  3 21:50:46 2020 daemon.info : 05[DMN] killing ourself, received critical signal
Thu Dec  3 21:50:46 2020 authpriv.info ipsec_starter[3425]: charon has died -- restart scheduled (5sec)
Thu Dec  3 21:50:46 2020 authpriv.info ipsec_starter[3425]: charon refused to be started

coolsnowwolf, 2020-12-04:
no support for capturing backtraces ？

kkxkx, 2020-12-04:
啥问题lean大佬，我用的你的源码11.17号编译的，升级到5.4以后ipsec基本就没用过了。什么都没动只改了改ip然后运行，就出现这些错误。每5s一次

aepub, 2021-01-06:
期待解决办法

closed 2021-01-16 by kkxkx