coolsnowwolf/lede #5878: 20201204 编译的K2P固件（内核更新到了5.4.79）无法启动

xylz0928, 2020-12-04:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：20201204 编译的四个固件，其中只有K2P更新后无法启动
尝试breed直接刷今日版本，仍然无法启动
尝试置空配置，刷上周版本固件，然后更新至今日固件，又无法启动

### (2) 路由器型号和固件版本

A：
K2P MT7621 5.4.79 不工作
R3P MT7621 5.4.79 正常
K2 MT7620 5.4.79 正常
R1CM MT7620 5.4.79 正常

### (3) 详细日志

A：暂无

kkxkx, 2020-12-05:
我12.03号编译的没问题，我精简了 当交换机。没有很多插件。可运行

closed 2020-12-05 by coolsnowwolf

soundghost, 2020-12-14:
跟你一样，我恰好也是编译了12.4号的固件之后刷机无法启动，不过我是选择的32MB，以前我都是手动改dts和mt7621.mk这两个文件来适配32MB的k2p，在修改时我发现coolsnowwolf新增加的32MB分区大小跟我以前的大小不一样，后来我重新改为我以前改的大小32448之后，还是无法启动。
于是我继续git pull，更新到更新的版本后，修改分区容量为32448，刷入后依然无法启动，但是刷入initramfs-kernel固件又可以启动，在luci里升级到squashfs-sysupgrade又无法启动，故障状态为亮红灯。
于是我在breed里刷入，这次选择分区格式为斐讯（以前一直都选公版），刷入的是squashfs-sysupgrade版本固件，这下可以启动了，但也不是没有问题，无法软重启。只要软重启，又亮红灯（包括luci里恢复出厂设置，或者autoreboot）。
希望这些信息能对 @coolsnowwolf 排查有帮助。

xylz0928, 2020-12-14:
> 跟你一样，我恰好也是编译了12.4号的固件之后刷机无法启动，不过我是选择的32MB，以前我都是手动改dts和mt7621.mk这两个文件来适配32MB的k2p，在修改时我发现coolsnowwolf新增加的32MB分区大小跟我以前的大小不一样，后来我重新改为我以前改的大小32448之后，还是无法启动。
>
> 于是我继续git [...]

一样的现象，breed下刷kernel能够启动，一旦更新sysupgrade就无法启动

soundghost, 2020-12-14:
> 一样的现象，breed下刷kernel能够启动，一旦更新sysupgrade就无法启动

那你试试在breed里选择分区格式为斐讯刷入squashfs-sysupgrade试试看，是否跟我一样能启动，但是无法软重启？

ltactor, 2020-12-15:
> 那你试试在breed里选择分区格式为斐讯刷入squashfs-sysupgrade试试看，是否跟我一样能启动，但是无法软重启？

一样的问题。 我没试过冷启动。