coolsnowwolf/lede #5882: 京东云RE-SP-01B  Flash实际可用不足32M 导致编译包含插件多了不出刷机固件

langliang3241, 2020-12-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最近京东云收益越来越少，以前保留了breed,就自己云编译了个固件使用，个人添加我常使用的插件后编译出来的固件没有刷   机固件，只有：openwrt-ramips-mt7621-jdcloud_re-sp-01b-initramfs-kernel.bin ，Google搜索了下找到恩山论坛一个帖子 ：https://www.right.com.cn/forum/thread-2846333-1-1.html  
===================================================================================
   1、为什么Flash实际可用不足32M？
        此问题已于[2020-03-01]的版本修复，详细请看更新日志。以下为原先的描述
    原因：除u-boot、config两个必要分区占用了少量空间外，官方固件还有自定义的mini、oem两个分区。目前形势下个人偏向于保留这两个分区的数据，其他Openwrt也是这样做的
===================================================================================

我备份了原厂编程器固件，想挖豆子随时可以恢复回去，现在想删除mini ome 两个分区，完全使用Flash的32M 空间以便安装更多的插件，请问各位大佬我要怎么才能办到？

### (2) 路由器型号和固件版本

A：路由器型号：JDCloud RE-SP-01B
      固件版本：OpenWrt R20.10.20 / LuCI Master (git-20.256.12360-1a54222) 

### (3) 详细日志

mt7621_jdcloud_re-sp-01b.dts文件

================================================================

            flash@0 {
                    compatible = "jedec,spi-nor";
                    reg = <0>;
                    spi-max-frequency = <50000000>;

                    partitions {
                            compatible = "fixed-partitions";
                            #address-cells = <1>;
                            #size-cells = <1>;

                            partition@0 {
                                    label = "u-boot";
                                    reg = <0x0 0x30000>;
                                    read-only;
                            };

                            config: partition@30000 {
                                    label = "config";
                                    reg = <0x30000 0x10000>;
                                    read-only;
                            };

                            factory: partition@40000 {
                                    label = "factory";
                                    reg = <0x40000 0x10000>;
                                    read-only;
                            };

                            partition[url=home.php?mod=space&uid=485662]@50000[/url] {
                                    compatible = "denx,uimage";
                                    label = "firmware";
                                    reg = <0x50000 0x1ab0000>;
                            };

                            partition@1b00000 {
                                    label = "mini";
                                    reg = <0x1b00000 0x400000>;
                                    read-only;
                            };

                            partition@1f00000 {
                                    label = "oem";
                                    reg = <0x1f00000 0x100000>;
                                    read-only;
                            };
                    };
            };

================================================================

mt7621.mk文件

================================================================

    define Device/jdcloud_re-sp-01b
      IMAGE_SIZE := 27328k
      DEVICE_VENDOR := JDCloud
      DEVICE_MODEL := RE-SP-01B
      DEVICE_PACKAGES := kmod-fs-ext4 kmod-mt7603 kmod-mt7615e kmod-sdhci-mt7620 \
            kmod-usb3 wpad-openssl
    endef

================================================================

Aes64X, 2020-12-06:
.....目测go部分占用了不少

AmadeusGhost, 2020-12-06:
dts 改成这样：
```
partitions {
	compatible = "fixed-partitions";
	#address-cells = <1>;
	#size-cells = <1>;

	partition@0 {
		label = "u-boot";
		reg = <0x0 0x30000>;
		read-only;
	};

	config: partition@30000 {
		label = "config";
		reg = <0x30000 0x10000>;
		read-only;
	};

	factory: partition@40000 {
		label = "factory";
		reg = <0x40000 0x10000>;
		read-only;
	};

	partition@50000 {
		compatible = "denx,uimage";
		label = "firmware";
		reg = <0x50000 0x1fb0000>;
	};
};
```
然后 mt7621.mk 的 IMAGE_SIZE 这一行改为 `IMAGE_SIZE := 32448k`

langliang3241, 2020-12-07:
经测试，修改上述两个地方后32M空间可以充分利用

closed 2020-12-07 by langliang3241