coolsnowwolf/lede #5887: 树莓派docker跑openwrt问题

elarkasi, 2020-12-06:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)

* [ ] 没有类似的issue

## 2. 详细叙述

编译好的rootfs.tar.gz 导入docker后，运行报错，container无法启动
删掉container之后重新再跑一遍，还报错，但是container能够运行起来

### (1) 具体问题

A：
`docker import openwrt-bcm27xx-bcm2709-rpi-2-rootfs.tar.gz openwrt`
`docker run --name openwrt --network macnet --privileged openwrt /sbin/init`

### (2) 路由器型号和固件版本

树莓派4 
Linux rpi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux

### (3) 详细日志

Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
Failed to find the folder holding the modules
Failed to find the folder holding the modules
Failed to find the folder holding the modules
Failed to find the folder holding the modules

NagaseKouichi, 2020-12-08:
docker的话，应该编译armvirt架构的吧？

kddxnb, 2020-12-08:
看固件名字，你这树莓派4用树莓派3的固件，咋可能跑的起来？

elarkasi, 2020-12-08:
> 看固件名字，你这树莓派4用树莓派3的固件，咋可能跑的起来？

32位的raspberry pi OS只能支持32位的固件, 在menuconfig那一步已经全部选了rpi1,rp2,rp3,rp4所有选项了。编译出来的固件名字是rp2的

elarkasi, 2020-12-08:
> docker的话，应该编译armvirt架构的吧？

固件配置里面有树莓派的选项

NagaseKouichi, 2020-12-08:
> > docker的话，应该编译armvirt架构的吧？
> 
> 固件配置里面有树莓派的选项

那是你直接安装的情况下用树莓派的固件，在docker下面装的话，应该是要用armvirt的

kddxnb, 2020-12-08:
> > 看固件名字，你这树莓派4用树莓派3的固件，咋可能跑的起来？
> 
> 32位的raspberry pi OS只能支持32位的固件, 在menuconfig那一步已经全部选了rpi1,rp2,rp3,rp4所有选项了。编译出来的固件名字是rp2的

哦，我只试过64位的，跑起来没问题，ubuntu 64位arm系统。

kddxnb, 2020-12-08:
> > 看固件名字，你这树莓派4用树莓派3的固件，咋可能跑的起来？
> 
> 32位的raspberry pi OS只能支持32位的固件, 在menuconfig那一步已经全部选了rpi1,rp2,rp3,rp4所有选项了。编译出来的固件名字是rp2的

哦，我只试过64位的，也是在64位系统下docker里跑，没什么问题

elarkasi, 2020-12-08:
> > > docker的话，应该编译armvirt架构的吧？
> > 
> > 
> > 固件配置里面有树莓派的选项
> 
> 那是你直接安装的情况下用树莓派的固件，在docker下面装的话，应该是要用armvirt的

好的，我试试，谢谢

SuLingGG, 2020-12-11:
如果不想折腾的话可以移步：
<https://github.com/SuLingGG/OpenWrt-Rpi-Docker>
如果用这个项目的话，因为你的宿主机内核是 32 位，所以只能用 latest 或者 rpi2 标签。

elarkasi, 2020-12-12:
> 如果不想折腾的话可以移步：
> https://github.com/SuLingGG/OpenWrt-Rpi-Docker
> 如果用这个项目的话，因为你的宿主机内核是 32 位，所以只能用 latest 或者 rpi2 标签。

上面是使用的32位系统做的测试，会有上面那个报错，百思不得其解。
后来编译了64位的镜像，放在64位系统上面跑的话只报下面的这个warning，没有error了，可以正常运行不会退出。
没查到原因
```
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
```

closed 2021-06-20 by coolsnowwolf