coolsnowwolf/lede #5890: Xray v1.1.2更新后无法启动

ZqinKing, 2020-12-07:
-----------start-----------
2020-12-07 13:46:52 Main node:Xray 1.1.2 (OpenWrt) Custom (go1.15.5 linux/amd64) Started!
------------end------------
2020-12-07 13:47:29 ssr redir tcp error.restart!

测试只有关闭xtls才能连上，配置和服务器均沿用v1.0.0的。v1.0.0时功能正常。

jxph1123, 2020-12-07:
更新，应该修复了。

Portan, 2020-12-07:
> 
> 
> 更新，应该修复了。

我更新了还是这个错误，不加xtls就能运行

jxph1123, 2020-12-07:
https://github.com/fw876/helloworld/commit/f195b8f6bf6dc2ae213025714acaf21dc7a7f2d7
这个应该就是？

Portan, 2020-12-07:
> 
> 
> > 更新，应该修复了。
> 
> 我更新了还是这个错误，不加xtls就能运行

找到原因了，我的配置里tls host为空，所以xtls无法使用。填上tls host就可以了

ZqinKing, 2020-12-07:
> > > 更新，应该修复了。
> > 
> > 
> > 我更新了还是这个错误，不加xtls就能运行
> 
> 找到原因了，我的配置里tls host为空，所以xtls无法使用。填上tls host就可以了

还真是，这个参数以前一直没填过。。。

ZqinKing, 2020-12-07:
issue先开着吧。等修复了再关。

aaronchin, 2020-12-08:
我发现的状况是设置xtls的时候 mux的设定会出现错误导致无法运行 选择xtls后 mux的在设定档里会是 "mux": false, 导致xray无法运行

Dimitri2020007, 2020-12-08:
ws伪装是用不了xtls的，支持cdn。只有tcp，domain socket，quic可以使用xtls和xtls流控，不支持cdn。

shadowswpu, 2020-12-10:
我最近也被ssr redir tcp error这个错误困扰，我用v2ray节点
以前都忽略了tls host这个选项，
1，勾选“允许不安全的连接”时，可以不用设置这一项
2，取消勾选，也就是需要验证时，需要设置tls host（跟证书域名一致吧）
以上两种情况个人遇到这个问题的解决方案，供大家参考

ixxmu, 2020-12-10:
看来tls参数改变了，之前就算开启tls验证，也可以忽略tls参数，现在必须填写tls参数才能正常拉起ssr，但是填写这个参数后延时是否增加了？ 之前电脑端出现过tls非空延时会增加。

huangyingjie, 2020-12-11:
谢谢，新鲜出炉的坑，终于填好了

aaronchin, 2020-12-11:
"mux": false, mux的设定还是有问题 会造成xray无法启动
     "mux": false,
     "protocol": "vless",
     "streamSettings": {
       "network": "tcp",
       "tlsSettings": {

root@OpenWrt:/etc/init.d# /usr/bin/xray -config /var/etc/v2-ssr-retcp.json
Xray 1.1.2 (OpenWrt) Custom (go1.15.5 linux/amd64)
A unified platform for anti-censorship.
panic: infra/conf/serial: failed to read config file at line 18 char 17 > json: cannot unmarshal bool into Go struct field OutboundDetourConfig.outbound.mux of type conf.MuxConfig

goroutine 1 [running]:
github.com/xtls/xray-core/common.Must(...)
        github.com/xtls/xray-core/common/common.go:26
github.com/xtls/xray-core/main/jsonem.init.0.func1(0xd727e0, 0xc0001b4c80, 0x7ffd7964eec7, 0x4, 0xc00007f8f0)
        github.com/xtls/xray-core/main/jsonem/jsonem.go:27 +0x473
github.com/xtls/xray-core/core.LoadConfig(0xed7f4d, 0x4, 0x7ffd7964eeb1, 0x1a, 0xd727e0, 0xc0001b4c80, 0xc00013fd30, 0xccfb86, 0xffd1a0)
        github.com/xtls/xray-core/core/config.go:67 +0x262
main.startXray(0x6708af, 0x0, 0xc0001bc180, 0x164)
        github.com/xtls/xray-core/main/run.go:158 +0x12e
main.executeRun(0x14fd100, 0xc0001ba4b0, 0x0, 0x0)
        github.com/xtls/xray-core/main/run.go:65 +0x4a
github.com/xtls/xray-core/main/commands/base.Execute()
        github.com/xtls/xray-core/main/commands/base/execute.go:64 +0x285
main.main()
        github.com/xtls/xray-core/main/main.go:22 +0x16e

closed 2020-12-17 by ZqinKing