coolsnowwolf/lede #5896: 可以在ssr plus中加入dns over https的解析模式吗？

Dimitri2020007, 2020-12-08:
同标题，如果建议不好可以忽略。谢谢！

garypang13, 2020-12-08:
可惜dns over https没法过滤aaaa记录, 不然比dnspd,smartdns等都好用.

jjit, 2020-12-08:
dns https proxy的端口设置为 5335
ssr plus的 DNS解析方式 设置为 使用本机端口为5335的DNS服务

Dimitri2020007, 2020-12-08:
dns https proxy的监听端口是不是设置成0.0.0.0，设成127.0.0.1的监听端口会错。

Dimitri2020007, 2020-12-08:
我拓扑是双lede路由，主lede负责多拨，旁lede负责ssr doh解析国外域名。网关是旁lede，dns是主lede。

主lede dns加速填isp dns。

旁lede dns是一样的。

而且都打开了sfe。

旁lede开启了dns https proxy 监听0.0.0.0:5335

旁lede ssr plus从pdnsd tcp解析模式改成了使用5335端口服务。

旁lede已经禁用了所有防火墙。

测试结果是ssr 谷歌链接测试通过了。
github暂时上不了，不知道过多长时间就又恢复了。

Dimitri2020007, 2020-12-08:
> 可惜dns over https没法过滤aaaa记录, 不然比dnspd,smartdns等都好用.

pdnsd是原来用的，smart dns没用过，unbound 缓存+tls转发，速度慢的离谱。头回解析慢，二回刷新贼快（不设置tls转发）。过一段时间又回到头回解析的情况了。

后来改sfe内的dns加速了，国内域名速度秒开。国外的配合doh也是很快就开。

Dimitri2020007, 2020-12-09:
> 试一下这个方法，dns https proxy的端口设置为 5335
> ssr plus的 DNS解析方式 设置为 使用本机端口为5335的DNS服务

你这个方法已经成功了，ssr plus国外解析已经开始走doh模式了。

只是github release和github raw的解析有问题，release下载和raw纯文本解析还是不行。

ZqinKing, 2020-12-09:
> 可惜dns over https没法过滤aaaa记录, 不然比dnspd,smartdns等都好用.

DOH跟smartdns有关系么？两个维度的东西，使用smartdns用DOH加上有节点，再过滤AAA记录就好。

Dimitri2020007, 2020-12-09:
smart dns没用过（源码已经收录但未出现在luci-app）

另外smartdns可以实现国内外域名秒开吗？

smartdns可以设置doh并和ssr plus关联吗？

smartdns重启了会掉缓存吗？

Dimitri2020007, 2020-12-09:
> 可惜dns over https没法过滤aaaa记录, 不然比dnspd,smartdns等都好用.

dns https proxy确实有这个问题，不知道怎么删除ipv6的dns。

而且我在的地方没有ipv6

ZqinKing, 2020-12-09:
> smart dns没用过（源码已经收录但未出现在luci-app）
> 
> 另外smartdns可以实现国内外域名秒开吗？
> 
> smartdns可以设置doh并和ssr plus关联吗？
> 
> smartdns重启了会掉缓存吗？
> 
> 而且lean openwrt删除了smartdns的luci-app，怎么找回？

L大去掉smartdns是被气的，你可以先git下来放在一个目录下（luci 和 smartdns两个），再feeds.conf.default加一行src-link把本地路径填上。

Dimitri2020007, 2020-12-09:
啊是这样的。

看来要重新action了。

ZqinKing, 2020-12-10:
> 啊是这样的。
> 
> 看来要重新action了。
> 
> smartdns端口是不是还是和ssr plus一样都在5335？
> 
> 若是这样我就可以取消unbound的编译了。

一般第一组默认不用动，第二组将端口设置成5335，第二组上游全部添加国外DNS走DOT或者DOH。第二区单独给SSR plus使用

Dimitri2020007, 2020-12-10:
是否还需要设置好snat伪装

ZqinKing, 2020-12-10:
> 是否还需要设置好snat伪装?
> 
> 另外smartdns在主lede上打开还是在旁lede上打开？
> 
> 或者都打开？

不需要snat，在你终端使用的网关IP上打开，有旁路由的话，应该是旁路由

Dimitri2020007, 2020-12-10:
谢谢你了！

closed 2020-12-10 by Dimitri2020007