coolsnowwolf/lede #5897: How to use mt driver instead of mt76?

Espionage724, 2020-12-08:
I have a NETGEAR R6260 (`MT7621`). It has a `MT7615E`.

I want to use the `mt` driver ([here](https://github.com/coolsnowwolf/lede/tree/master/package/lean/mt)) instead of `mt76` ([here](https://github.com/coolsnowwolf/lede/tree/master/package/kernel/mt76)). How would I do this?

As I understand, `mt` in this repo is the proprietary MediaTek wireless driver, where `mt76` is the open-source version.

AmadeusGhost, 2020-12-09:
Deselect open source driver, and then choose kmod-mt7615d and luci-app-mtwifi.

Espionage724, 2020-12-09:
Thanks; got that and it seems to load up according to `dmesg`, but I can't seem to actually get the wireless up (neither 2.4GHz nor 5GHz APs show up on devices).

I tried playing around with settings yesterday from luci and had no luck. Is there a command or setting I need to set to enable the wifi?

langit7, 2020-12-09:
mir3p 
CONFIG_PACKAGE_kmod-mt76-core=m
# CONFIG_PACKAGE_kmod-mt7601u is not set
# CONFIG_PACKAGE_kmod-mt7603 is not set
# CONFIG_PACKAGE_kmod-mt7603e is not set
CONFIG_PACKAGE_kmod-mt7615-common=m
CONFIG_PACKAGE_kmod-mt7615-firmware=y
CONFIG_PACKAGE_kmod-mt7615d=y
CONFIG_MTK_SUPPORT_OPENWRT=y
CONFIG_MTK_WIFI_DRIVER=y
CONFIG_MTK_FIRST_IF_MT7615E=y
# CONFIG_MTK_FIRST_IF_MT7622 is not set
# CONFIG_MTK_FIRST_IF_MT7626 is not set
# CONFIG_MTK_FIRST_IF_NONE is not set
# CONFIG_MTK_SECOND_IF_NONE is not set
CONFIG_MTK_SECOND_IF_MT7615E=y
CONFIG_MTK_THIRD_IF_NONE=y
# CONFIG_MTK_THIRD_IF_MT7615E is not set
CONFIG_MTK_RT_FIRST_CARD=7615
CONFIG_MTK_RT_SECOND_CARD=7615
CONFIG_MTK_RT_FIRST_IF_RF_OFFSET=0xc0000
CONFIG_MTK_RT_SECOND_IF_RF_OFFSET=0xc8000
CONFIG_MTK_MT_WIFI=y
CONFIG_MTK_MT_WIFI_PATH="mt_wifi"

root@Xiaomi-MiR3P:/etc/config# opkg list-installed|grep mt76
kmod-mt7615-firmware - 5.4.79+2020-09-23-b22977c2-2
kmod-mt7615d - 5.4.79+5.0.4.0-1
kmod-mt7615d_dbdc - 5.4.79+5.0.4.0-1

5Ghz so poor no only 1 bar for 1 meter range. and unable to change channel and country

AmadeusGhost, 2020-12-10:
> opkg list-installed|grep mt76
> kmod-mt7615-firmware - 5.4.79+2020-09-23-b22977c2-2
> kmod-mt7615d - 5.4.79+5.0.4.0-1
> kmod-mt7615d_dbdc - 5.4.79+5.0.4.0-1

If you want to use this private driver, you should not choose kmod-mt7615-common and kmod-mt7615-firmware (these are open source drivers), nor should you choose kmod-mt7615d_dbdc (this only applies to mt7615dn). If you are affected by the related commit of kmod-mt7615d_dbdc, you can try with `git revert 13ec6bcfd6e9d7f3aeb115fb2786d249c7a9e95a`.

langit7, 2020-12-11:
> If you want to use this private driver, you should not choose kmod-mt7615-common and kmod-mt7615-firmware (these are open source drivers), nor should you choose kmod-mt7615d_dbdc (this only applies [...]

Thanks its working , but on my mir3p 5Ghz slower than 2.4 G.. 
```
root@Xiaomi-MiR3P:~# opkg list-installed |grep mt
kmod-mt7615d - 5.4.79+5.0.4.0-1
luci-app-mtwifi - 1-16
mt_wifi - 1-1
```
2.4G speed
```
Accepted connection from 192.168.7.53, port 46742
[  5] local 192.168.7.1 port 5201 connected to 192.168.7.53 port 46744
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  12.7 MBytes   106 Mbits/sec                  
[  5]   1.00-2.00   sec  18.3 MBytes   153 Mbits/sec                  
[  5]   2.00-3.00   sec  16.3 MBytes   137 Mbits/sec                  
[  5]   3.00-3.08   sec  1.02 MBytes   114 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-3.08   sec  48.3 MBytes   132 Mbits/sec                  receiver
-----------------------------------------------------------
```
5G speed
```
Accepted connection from 192.168.7.53, port 46766
[  5] local 192.168.7.1 port 5201 connected to 192.168.7.53 port 46768
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  6.66 MBytes  55.7 Mbits/sec                  
[  5]   1.00-2.00   sec  6.78 MBytes  57.1 Mbits/sec                  
[  5]   2.00-3.00   sec  14.6 MBytes   122 Mbits/sec                  
[  5]   3.00-3.08   sec  1.18 MBytes   118 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-3.08   sec  29.2 MBytes  79.4 Mbits/sec                  receiver
-----------------------------------------------------------
```

closed 2021-06-20 by coolsnowwolf