coolsnowwolf/lede #5900: 请问2.4Gwifi，5Gwifi，网口驱动分别是哪几个安装包？

logernLee, 2020-12-09:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：请问2.4Gwifi，5Gwifi，网口驱动分别是源码里面的哪几个安装包？

### (2) 路由器型号和固件版本

A：竟斗云 IPQ4019

### (3) 详细日志

A：

AmadeusGhost, 2020-12-09:
无线是 kmod-ath10k-ct 和 ath10k-firmware-qca4019-ct，有线的不是软件包

logernLee, 2020-12-10:
好的    感谢回复

logernLee, 2020-12-10:
请问这个wifi没有加载上是什么情况呢？
[image]

AmadeusGhost, 2020-12-10:
> 请问这个wifi没有加载上是什么情况呢？

选上 ipq-wifi-p2w_r619ac

logernLee, 2020-12-10:
你好，这个选项已经是选上的。
[image]

AmadeusGhost, 2020-12-10:
看了下日志，你这机子不是竞斗云吧？

logernLee, 2020-12-10:
板子确实是竟斗云的，外壳上都写着竟斗云，里面默认固件登录进去也是竟斗云的界面。但是就是不知道为啥会出现这个问题--

AmadeusGhost, 2020-12-10:
你试试我编译的固件：https://ws28.cn/f/46ug3eh07se

logernLee, 2020-12-10:
这个还是有点小问题。。
[image]

logernLee, 2020-12-10:
fetch board file的时候开始报错，我的是一开始加载就报错了 。。

closed 2021-06-20 by coolsnowwolf