coolsnowwolf/lede #5901: 【咨询】新的wifi luci管理 如何在redmi ac2100 上开启

MonwF, 2020-12-09:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：现在只支持k2p，想在我的ac2100上开启，除了kernel选中mt7615d_dbdc，还需要做些什么，编译后wifi设置一直是未启用状态 

### (2) 路由器型号和固件版本

A：红米 ac2100

报错如下

```
Wed Dec  9 18:58:04 2020 daemon.err uhttpd[1517]: assoclist ioctl fails
Wed Dec  9 18:58:04 2020 daemon.err uhttpd[1517]: assoclist ioctl fails
Wed Dec  9 18:58:04 2020 daemon.err uhttpd[1517]: assoclist ioctl fails
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): killall: -q: no process killed
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): killall: apcli_5g: no process killed
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): killall: -q: no process killed
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): killall: apcli_2g: no process killed
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): killall: -q: no process killed
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): killall: apcli_5g: no process killed
Wed Dec  9 18:58:07 2020 kern.warn kernel: [ 3300.321066] ra0: ===> main_virtual_if_close
Wed Dec  9 18:58:07 2020 kern.warn kernel: [ 3300.325373] wifi_sys_close(), wdev idx = 0
Wed Dec  9 18:58:07 2020 daemon.notice netifd: Network device 'ra0' link is down
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): ifconfig: SIOCGIFFLAGS: No such device
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): ifconfig: SIOCGIFFLAGS: No such device
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): ifconfig: SIOCGIFFLAGS: No such device
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): ifconfig: SIOCGIFFLAGS: No such device
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): ifconfig: SIOCGIFFLAGS: No such device
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): ifconfig: SIOCGIFFLAGS: No such device
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): ifconfig: SIOCGIFFLAGS: No such device
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): ifconfig: SIOCGIFFLAGS: No such device
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): ifconfig: SIOCGIFFLAGS: No such device
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): ifconfig: SIOCGIFFLAGS: No such device
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): ifconfig: SIOCGIFFLAGS: No such device
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): ifconfig: SIOCGIFFLAGS: No such device
Wed Dec  9 18:58:07 2020 daemon.notice netifd: rax0 (32308): MT_DBDC:ra0:up.Later we'll restart ra0
```

AmadeusGhost, 2020-12-09:
ac2100 还不支持

MonwF, 2020-12-09:
> ac2100 还不支持

感谢回复，那我先还原回去，再等等

closed 2021-06-20 by coolsnowwolf