coolsnowwolf/lede #5911: rockchip支持r4s的补丁有错误

wushenghn, 2020-12-11:
日志
Applying /home/wsh/lede/target/linux/rockchip/patches-5.4/200-rockchip-add-support-for-NanoPi-R4S.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/Makefile
patching file arch/arm64/boot/dts/rockchip/rk3399-nanopi-r4s.dts
patch unexpectedly ends in middle of line
patch: **** malformed patch at line 155:

Patch failed!  Please fix /home/wsh/lede/target/linux/rockchip/patches-5.4/200-rockchip-add-support-for-NanoPi-R4S.patch!
Makefile:26: recipe for target '/home/wsh/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.79/.prepared_13216ae8bf261790bb300ade74370374' failed
make[4]: *** [/home/wsh/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.79/.prepared_13216ae8bf261790bb300ade74370374] Error 1
make[4]: Leaving directory '/home/wsh/lede/target/linux/rockchip'
Makefile:13: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/wsh/lede/target/linux'
time: target/linux/compile#11.18#99.17#282.30
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 2
make[2]: Leaving directory '/home/wsh/lede'
target/Makefile:18: recipe for target '/home/wsh/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile' failed
make[1]: *** [/home/wsh/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/wsh/lede'
/home/wsh/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
wsh@wsh:~/lede$

onions00, 2020-12-11:
缺少rk3399-nanopi-r4s.dts
拷贝一个到目录就OK了

hotislandn, 2020-12-11:
那个patch文件的155行删除，再手动输入一行，然后重新编译试试。

closed 2020-12-11 by wushenghn

lifetyper, 2020-12-11:
> 
> 
> 缺少rk3399-nanopi-r4s.dts
> 拷贝一个到目录就OK了

从哪里拷到哪里？

wushenghn, 2020-12-11:
200.patch文件第155行格式有问题删掉重写一下，u-boot 版本有bug回退上一个版本