coolsnowwolf/lede #5912: 有两个kcptun，可以考虑删除任意其中一个

szoyj, 2020-12-11:
在package/lean/下有一个kcptun
在feeds/packages/net/下也有一个kcptun
完全是多余的，而且会产生各种混乱和混淆，二选一就可以了。

可以参考官方的，建议kcptun只保留feeds下的

WYC-2020, 2020-12-11:
@szoyj 完全没必要，别人自动会优选选用lean的 那个不影响

szoyj, 2020-12-11:
@WYC-2020 你所谓的优选就是，feeds里的那个函数is_core_src的功劳。
我还是说下我的情况吧，我打算在原版openwrt里移植helloworld进去。
我拿官方原本openwrt（tag19.07.5），额外加了feed（helloworld），然后执行feeds update/install，此时会报kcptun-client找不到，然后把kcptun复制到了官方openwrt/package/lean/*下，然后再次执行feeds update/install，但是还是会报kcptun-client找不到，原因include/scan.awk，的作用是两个都存在时反而优选feeds下的

szoyj, 2020-12-11:
@WYC-2020 当然你可以在完全干净的根目录的情况下，先复制所有的依赖，再执行feeds update/install，这样就会是使用package下的，而不是feeds下的。
这就是我所谓的混乱和混淆，不同的操作过程因为tmp下各种缓存导致结果不一致。

WYC-2020, 2020-12-11:
@szoyj官方源码不会又lean目录，如果只是单独新增hell只要找好他依赖的库都作为feed载入不就行了，不要到处去放，lean的源码是做了优先级的 可以优先载入lean目录下的

szoyj, 2020-12-11:
@WYC-2020 你所谓的优先级是不是feeds里的is_core_src函数？

szoyj, 2020-12-11:
@WYC-2020 你可以这么试
完全干净的lean的openwrt项目
1. 移除package/lean/kcptun
2. 执行feeds update 和install
3. 还原package/lean/kcptun
4.执行feeds update 和install
5.在文件/tmp/info/.overrides-packageinfo-*里会有什么内容，你可以去看下，是不是有lean/kcptun，是不是意味着被override了？

WYC-2020, 2020-12-11:
@szoyj ./scripts/feeds clean

szoyj, 2020-12-11:
@WYC-2020 clean我当然知道啊。算了，不说了，爱删不删吧。

closed 2020-12-11 by szoyj

szoyj, 2020-12-12:
a:  这里路上有个坑，恐怕会有人会踩到。
b: 我们都是在走边上的。
a: 你这么这么走，一定掉坑里。
b: 你可以跨过去呀。
a: 对对对，你说的都对