coolsnowwolf/lede #5920: 请教：怎么将B50其中一个 lan 口转换成 wan 口

Hey-Leo, 2020-12-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：想要将B50的一个lan口转换成wan口，用于双宽带双线接入

> 尝试过程：

已经尝试过：试过在lan口取消桥接lan3，然后重新添加新的接口，设置名lan3，物理接口lan3，DHCP客户端，防火墙为WAN
设置完成后：无法取得IPv4地址，直接将默认带有的DHCPv6那个改DHCP是能获得正确地址的

> 网络环境：

校园网，里面有DHCP协议和PPPoE协议，这两者并存，在这里我是用DHCP协议检测设置是否正确，正确后再启用PPPoE进行网络拨号（DHCP协议好像是给校园网客户端Dr.com交流通讯的？可以正常访问内网的网络自助管理系统，无法访问外网的，，PPPoE是可以访问外网的）

> 另：

没有进行过vlan的配置，因为自带的luci里面没有这一选项，

### (2) 路由器型号和固件版本

A：B50，R20.10.20

Her0R0cket, 2020-12-12:
你这个不需要使用新的物理接口,你这个可以直接添加新接口,然后包含原来的wan口,协议选dhcp就可以了

Hey-Leo, 2020-12-12:
> 你这个不需要使用新的物理接口，你这个可以直接添加新接口，然后包含原来的wan口，协议选dhcp就可以了

我是要双线一起的啊，就是需要物理接口，因为单口是100M的网速（没办法，学校交换机老了），单口就没意义了；双口才能最大化叠加到200M

Ljzkirito, 2020-12-12:
设置完成后重启路由试试

Her0R0cket, 2020-12-12:
因为家里普通路由没有用openwrt系统了,所以不能确定,我记得要在交换机里先untagged,还要新建个vlan的吧.具体可以网上找下

Hey-Leo, 2020-12-12:
> 因为家里普通路由没有用openwrt系统了,所以不能确定,我记得要在交换机里先untagged,还要新建个vlan的吧.具体可以网上找下

luci没有带vlan的设置，又不知道怎么划分

Her0R0cket, 2020-12-12:
> > 因为家里普通路由没有用openwrt系统了,所以不能确定,我记得要在交换机里先untagged,还要新建个vlan的吧.具体可以网上找下
> 
> luci没有带vlan的设置，又不知道怎么划分

我印象中openwrt的vlan也在交换机那边的

Hey-Leo, 2020-12-12:
> > > 因为家里普通路由没有用openwrt系统了,所以不能确定,我记得要在交换机里先untagged,还要新建个vlan的吧.具体可以网上找下
> > 
> > 
> > luci没有带vlan的设置，又不知道怎么划分
> 
> 我印象中openwrt的vlan也在交换机那边的

[image]
没

Her0R0cket, 2020-12-14:
[image]
没有这个路由,现编译了一个路由宝的固件,是有交换机设置的,请检查你自己的编译选项

Hey-Leo, 2020-12-16:
> [image]
> 没有这个路由,现编译了一个路由宝的固件,是有交换机设置的,请检查你自己的编译选项

```
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_hiwifi_hc5962=y

# CONFIG_PACKAGE_kmod-xmurp-ua=y
CONFIG_PACKAGE_nginx-mod-luci=y

CONFIG_PACKAGE_luci-app-adbyby-plus=y

CONFIG_PACKAGE_UnblockNeteaseMusic=y
CONFIG_PACKAGE_UnblockNeteaseMusicGo=y
CONFIG_UnblockNeteaseMusic_Go=y
CONFIG_UnblockNeteaseMusic_NodeJS=y

# CONFIG_PACKAGE_luci-app-adguardhome=y
# CONFIG_PACKAGE_luci-app-ipsec-vpnd=y
# CONFIG_PACKAGE_luci-app-jd-dailybonus=y
# CONFIG_PACKAGE_luci-app-mwan3=y
CONFIG_PACKAGE_luci-app-syncdial=y
# CONFIG_PACKAGE_luci-app-softethervpn=y

CONFIG_PACKAGE_luci-theme-argon=y
CONFIG_PACKAGE_luci-app-argon-config=y
# CONFIG_PACKAGE_luci-theme-atmaterial=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_luci-theme-infinityfreedom=y
# CONFIG_PACKAGE_luci-theme-purple=y
# CONFIG_PACKAGE_luci-theme-rosy=y

# CONFIG_PACKAGE_luci-app-vssr=y

CONFIG_PACKAGE_luci-app-ssr-plus=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Kcptun=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Redsocks2=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Trojan=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray_plugin=y

CONFIG_PACKAGE_automount=y
CONFIG_PACKAGE_luci-app-aria2=y
# CONFIG_PACKAGE_luci-app-hd-idle=y
# CONFIG_PACKAGE_luci-app-minidlna=y
# CONFIG_PACKAGE_luci-app-samba4=y
# CONFIG_PACKAGE_webui-aria2=y
# CONFIG_PACKAGE_yaaw=y

# CONFIG_PACKAGE_luci-app-familycloud=y
# CONFIG_PACKAGE_luci-app-frpc=y
# CONFIG_PACKAGE_luci-app-frps=y
# CONFIG_PACKAGE_luci-app-kodexplorer=y
# CONFIG_PACKAGE_luci-app-music-remote-center=y
# CONFIG_PACKAGE_luci-app-oaf=y
# CONFIG_PACKAGE_luci-app-rclone=y
# CONFIG_PACKAGE_luci-app-wrtbwmon=y

# CONFIG_PACKAGE_luci-app-transmission=y
# CONFIG_PACKAGE_luci-app-qbittorrent=y
# CONFIG_PACKAGE_luci-app-e2guardian=y
# 
CONFIG_PACKAGE_luci-app-ttyd=y
# CONFIG_PACKAGE_luci-app-watchcat=y
# CONFIG_PACKAGE_luci-app-meshwizard=y
```

这都没关什么

yulinsoft, 2020-12-22:
好像在交换机中设置一下，再添加接口就可以了。

Hey-Leo, 2020-12-22:
就是，默认的没有带交换机的配置界面，，所以不晓得怎么弄了，，，不知道要添加哪个插件

LGA1150, 2020-12-23:
> 已经尝试过：试过在lan口取消桥接lan3，然后重新添加新的接口，设置名lan3，物理接口lan3，DHCP客户端，防火墙为WAN
> 设置完成后：无法取得IPv4地址，直接将默认带有的DHCPv6那个改DHCP是能获得正确地址的

你有没有手动固定一下 lan3 的 MAC 地址？理论上这样设置没有问题

Hey-Leo, 2020-12-23:
> > 已经尝试过：试过在lan口取消桥接lan3，然后重新添加新的接口，设置名lan3，物理接口lan3，DHCP客户端，防火墙为WAN
> > 设置完成后：无法取得IPv4地址，直接将默认带有的DHCPv6那个改DHCP是能获得正确地址的
> 
> 你有没有手动固定一下 lan3 的 MAC 地址？理论上这样设置没有问题

没懂

LGA1150, 2020-12-23:
> 没懂

确保 LAN3 和 WAN 的 MAC 地址不一样，否则接在同一广播域下会冲突

Hey-Leo, 2020-12-24:
> 确保 LAN3 和 WAN 的 MAC 地址不一样，否则接在同一广播域下会冲突

他本身就是不一样的，而且直接改wan6协议dhcp是可以拿到IP地址的，wan和wan6口MAC是一样的都没问题（dhcp协议）

Hey-Leo, 2020-12-24:
我又搜了一下交换机，似乎，5.4内核上投过票，去掉了交换机的那个东西 ？  同步官方？？？

无能为力了，找不到方法  /摊手

Hey-Leo, 2020-12-24:
有说可以改dts/dtsi ？ 但是没说具体需要怎么改，，， 我太难了

LGA1150, 2020-12-24:
给我看看你的配置 /etc/config/network，和内核日志

Hey-Leo, 2020-12-24:
周末看看，现在要上班，，，我没能设置成功就重置了网络

Hey-Leo, 2020-12-25:
```
[    0.000000] Linux version 5.4.84 (Hey-Leo@GitHub Actions) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0-af6a5cc)) #0 SMP Wed Dec 23 04:31:24 2020
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is HiWiFi HC5962
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s27184 r8192 d21968 u57344
[    0.000000] pcpu-alloc: s27184 r8192 d21968 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 250600K/262144K available (5756K kernel code, 248K rwdata, 1240K rodata, 1256K init, 238K bss, 11544K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x340/0x550 with crng_init=0
[    0.000000] CPU Clock: 880MHz, start overclocking
[    0.000000] CPU Clock: 1000MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 3822520893 ns
[    0.000008] sched_clock: 32 bits at 500MHz, resolution 2ns, wraps every 4294967295ns
[    0.007794] Calibrating delay loop... 663.55 BogoMIPS (lpj=1327104)
[    0.046005] pid_max: default: 32768 minimum: 301
[    0.050758] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.057979] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.068277] rcu: Hierarchical SRCU implementation.
[    0.073682] smp: Bringing up secondary CPUs ...
[    6.537897] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    6.537905] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    6.537917] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    6.538016] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.110443] Synchronize counters for CPU 1: done.
[    6.547187] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    6.547195] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    6.547202] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    6.547255] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.179093] Synchronize counters for CPU 2: done.
[    6.614502] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    6.614509] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    6.614516] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    6.614570] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.240949] Synchronize counters for CPU 3: done.
[    0.270805] smp: Brought up 1 node, 4 CPUs
[    0.278782] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.288462] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.295610] pinctrl core: initialized pinctrl subsystem
[    0.302268] NET: Registered protocol family 16
[    0.317827] FPU Affinity set after 5328 emulations
[... 407 lines cut ...]
config interface 'wan'
	option ifname 'wan'
	option _orig_ifname 'wan'
	option _orig_bridge 'false'
	option proto 'pppoe'
	option username '31'
	option password '12'
	option ipv6 'auto'
	option peerdns '0'
	option dns '119.29.29.29 222.200.115.251 223.5.5.5 222.200.115.252'
	option keepalive '0'

config device 'wan_wan_dev'
	option name 'wan'
	option macaddr 'd4:ee:00:00:9f:f7'

config interface 'wan6'
	option ifname 'wan'
	option proto 'dhcpv6'

config interface 'lan3'
	option proto 'dhcp'
	option ifname 'lan3'

```

LGA1150, 2020-12-25:
现在 LAN3 可以 DHCP 吗？

Hey-Leo, 2020-12-25:
好像又可以了

Hey-Leo, 2020-12-25:
改pppoe还是接不上去，正常的那个wan口是可以的
```
[  673.729265] mt7530 mdio-bus:1f lan3: Link is Down
[  673.752819] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  673.764870] 8021q: adding VLAN 0 to HW filter on device lan3
[  674.274323] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  674.281555] 8021q: adding VLAN 0 to HW filter on device lan3
[  676.343472] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[  676.351039] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
[  678.141798] pppoe-wan: renamed from ppp0
[  691.678283] mt7530 mdio-bus:1f lan3: Link is Down
[  691.700378] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  691.708395] 8021q: adding VLAN 0 to HW filter on device lan3
[  691.853479] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  691.860679] 8021q: adding VLAN 0 to HW filter on device lan3
[  693.911124] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[  693.918650] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
[  709.245242] mt7530 mdio-bus:1f lan3: Link is Down
[  709.264767] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  709.271960] 8021q: adding VLAN 0 to HW filter on device lan3
[  709.415488] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  709.423134] 8021q: adding VLAN 0 to HW filter on device lan3
[  711.478984] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[  711.486511] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
[  726.782641] mt7530 mdio-bus:1f lan3: Link is Down
[  726.798310] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  726.805678] 8021q: adding VLAN 0 to HW filter on device lan3
[  726.914152] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  726.921219] 8021q: adding VLAN 0 to HW filter on device lan3
[  728.982844] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[  728.990363] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
[  744.290656] mt7530 mdio-bus:1f lan3: Link is Down
[  744.305906] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  744.313206] 8021q: adding VLAN 0 to HW filter on device lan3
[  744.424892] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  744.431911] 8021q: adding VLAN 0 to HW filter on device lan3
[  746.486594] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[  746.494116] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
[  761.791317] mt7530 mdio-bus:1f lan3: Link is Down
[  761.806072] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  761.814385] 8021q: adding VLAN 0 to HW filter on device lan3
[  761.927543] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  761.934610] 8021q: adding VLAN 0 to HW filter on device lan3
[  763.990372] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[  763.997909] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
[  779.282726] mt7530 mdio-bus:1f lan3: Link is Down
[  779.296979] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  779.304809] 8021q: adding VLAN 0 to HW filter on device lan3
[  779.419591] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  779.426644] 8021q: adding VLAN 0 to HW filter on device lan3
[  781.462115] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[  781.469650] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
[  796.781645] mt7530 mdio-bus:1f lan3: Link is Down
[  796.799924] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  796.807237] 8021q: adding VLAN 0 to HW filter on device lan3
[  796.973658] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  796.980647] 8021q: adding VLAN 0 to HW filter on device lan3
[  799.029926] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[  799.037468] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
[  814.347272] mt7530 mdio-bus:1f lan3: Link is Down
[  814.364746] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  814.372023] 8021q: adding VLAN 0 to HW filter on device lan3
[  814.486507] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  814.493805] 8021q: adding VLAN 0 to HW filter on device lan3
[  816.533668] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[  816.541195] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
[  831.829406] mt7530 mdio-bus:1f lan3: Link is Down
[  831.844149] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  831.851621] 8021q: adding VLAN 0 to HW filter on device lan3
[  831.974919] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  831.982516] 8021q: adding VLAN 0 to HW filter on device lan3
[  834.037545] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[  834.045132] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
[  849.345156] mt7530 mdio-bus:1f lan3: Link is Down
[  849.363396] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  849.370865] 8021q: adding VLAN 0 to HW filter on device lan3
[  849.484870] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  849.491867] 8021q: adding VLAN 0 to HW filter on device lan3
[  851.541146] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[  851.548681] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
[  866.833621] mt7530 mdio-bus:1f lan3: Link is Down
[  866.851495] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  866.858896] 8021q: adding VLAN 0 to HW filter on device lan3
[  866.963511] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[  866.970556] 8021q: adding VLAN 0 to HW filter on device lan3
[  869.012763] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[  869.020374] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
```

Hey-Leo, 2020-12-25:
系统日志重复报下面这段
```
Fri Dec 25 22:29:59 2020 kern.info kernel: [ 1272.236941] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
Fri Dec 25 22:29:59 2020 kern.info kernel: [ 1272.244186] 8021q: adding VLAN 0 to HW filter on device lan3
Fri Dec 25 22:29:59 2020 daemon.notice netifd: Interface 'lan3' is enabled
Fri Dec 25 22:30:01 2020 kern.info kernel: [ 1274.283309] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
Fri Dec 25 22:30:01 2020 kern.info kernel: [ 1274.290990] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
Fri Dec 25 22:30:01 2020 daemon.notice netifd: Network device 'lan3' link is up
Fri Dec 25 22:30:01 2020 daemon.notice netifd: Interface 'lan3' has link connectivity
Fri Dec 25 22:30:01 2020 daemon.notice netifd: Interface 'lan3' is setting up now
Fri Dec 25 22:30:01 2020 daemon.err insmod: module is already loaded - slhc
Fri Dec 25 22:30:01 2020 daemon.err insmod: module is already loaded - ppp_generic
Fri Dec 25 22:30:01 2020 daemon.err insmod: module is already loaded - pppox
Fri Dec 25 22:30:01 2020 daemon.err insmod: module is already loaded - pppoe
Fri Dec 25 22:30:01 2020 daemon.info pppd[8299]: Plugin rp-pppoe.so loaded.
Fri Dec 25 22:30:01 2020 daemon.info pppd[8299]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Fri Dec 25 22:30:01 2020 daemon.notice pppd[8299]: pppd 2.4.8 started by root, uid 0
Fri Dec 25 22:30:16 2020 daemon.warn pppd[8299]: Timeout waiting for PADO packets
Fri Dec 25 22:30:16 2020 daemon.err pppd[8299]: Unable to complete PPPoE Discovery
Fri Dec 25 22:30:16 2020 daemon.info pppd[8299]: Exit.
Fri Dec 25 22:30:16 2020 daemon.notice netifd: Interface 'lan3' is now down
Fri Dec 25 22:30:16 2020 kern.info kernel: [ 1289.623509] mt7530 mdio-bus:1f lan3: Link is Down
Fri Dec 25 22:30:16 2020 daemon.notice netifd: Interface 'lan3' is disabled
Fri Dec 25 22:30:16 2020 kern.info kernel: [ 1289.642764] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
Fri Dec 25 22:30:16 2020 kern.info kernel: [ 1289.650304] 8021q: adding VLAN 0 to HW filter on device lan3
Fri Dec 25 22:30:16 2020 daemon.notice netifd: Interface 'lan3' is enabled
Fri Dec 25 22:30:16 2020 daemon.notice netifd: Interface 'lan3' is setting up now
Fri Dec 25 22:30:16 2020 daemon.notice netifd: Network device 'lan3' link is down
Fri Dec 25 22:30:16 2020 daemon.notice netifd: Interface 'lan3' has link connectivity loss
Fri Dec 25 22:30:16 2020 daemon.err insmod: module is already loaded - slhc
Fri Dec 25 22:30:16 2020 daemon.err insmod: module is already loaded - ppp_generic
Fri Dec 25 22:30:16 2020 daemon.err insmod: module is already loaded - pppox
Fri Dec 25 22:30:16 2020 daemon.err insmod: module is already loaded - pppoe
Fri Dec 25 22:30:16 2020 daemon.notice netifd: lan3 (8450): Command failed: Permission denied
Fri Dec 25 22:30:17 2020 daemon.notice netifd: Interface 'lan3' is now down
Fri Dec 25 22:30:17 2020 daemon.notice netifd: Interface 'lan3' is disabled
Fri Dec 25 22:30:17 2020 kern.info kernel: [ 1290.166071] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
Fri Dec 25 22:30:17 2020 kern.info kernel: [ 1290.173449] 8021q: adding VLAN 0 to HW filter on device lan3
Fri Dec 25 22:30:17 2020 daemon.notice netifd: Interface 'lan3' is enabled
Fri Dec 25 22:30:19 2020 kern.info kernel: [ 1292.235179] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
Fri Dec 25 22:30:19 2020 kern.info kernel: [ 1292.243908] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
```

LGA1150, 2020-12-25:
> Timeout waiting for PADO packets

这个就是找不到 PPPoE 服务器的意思，也有可能是 MAC 地址 `d4:ee:00:00:9f:f6` 被服务器封了

你用台电脑改 MAC 为 `d4:ee:00:00:9f:f6` ，连接 LAN3 所在的网口，可以正常 PPPoE 拨号吗？

Hey-Leo, 2020-12-25:
改回dhcp后，又连不了了，系统日志
```
Fri Dec 25 22:34:32 2020 kern.info kernel: [ 1545.747133] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
Fri Dec 25 22:34:32 2020 kern.info kernel: [ 1545.755624] 8021q: adding VLAN 0 to HW filter on device lan3
Fri Dec 25 22:34:32 2020 daemon.notice netifd: Interface 'lan3' is enabled
Fri Dec 25 22:34:32 2020 daemon.notice netifd: Interface 'lan3' has link connectivity
Fri Dec 25 22:34:32 2020 daemon.notice netifd: Interface 'lan3' is setting up now
Fri Dec 25 22:34:32 2020 daemon.notice netifd: Network device 'lan3' link is down
Fri Dec 25 22:34:32 2020 daemon.notice netifd: Interface 'lan3' has link connectivity loss
Fri Dec 25 22:34:32 2020 daemon.notice netifd: lan3 (14631): Command failed: Permission denied
Fri Dec 25 22:34:32 2020 daemon.notice netifd: Interface 'lan3' is now down
Fri Dec 25 22:34:33 2020 daemon.notice netifd: Interface 'lan3' is disabled
Fri Dec 25 22:34:33 2020 kern.info kernel: [ 1546.072982] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
Fri Dec 25 22:34:33 2020 kern.info kernel: [ 1546.086463] 8021q: adding VLAN 0 to HW filter on device lan3
Fri Dec 25 22:34:33 2020 daemon.notice netifd: Interface 'lan3' is enabled
Fri Dec 25 22:34:33 2020 daemon.info dnsmasq[14621]: Connected to system UBus
Fri Dec 25 22:34:33 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Dec 25 22:34:33 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Dec 25 22:34:35 2020 kern.info kernel: [ 1548.132484] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
Fri Dec 25 22:34:35 2020 kern.info kernel: [ 1548.140158] IPv6: ADDRCONF(NETDEV_CHANGE): lan3: link becomes ready
Fri Dec 25 22:34:35 2020 daemon.notice netifd: Network device 'lan3' link is up
Fri Dec 25 22:34:35 2020 daemon.notice netifd: Interface 'lan3' has link connectivity
Fri Dec 25 22:34:35 2020 daemon.notice netifd: Interface 'lan3' is setting up now
Fri Dec 25 22:34:35 2020 daemon.notice netifd: lan3 (14943): udhcpc: started, v1.31.1
Fri Dec 25 22:34:35 2020 daemon.notice netifd: lan3 (14943): udhcpc: sending discover
Fri Dec 25 22:34:37 2020 daemon.info dnsmasq[14990]: Connected to system UBus
Fri Dec 25 22:34:37 2020 cron.err crond[15183]: crond (busybox 1.31.1) started, log level 8
Fri Dec 25 22:34:38 2020 daemon.notice netifd: lan3 (14943): udhcpc: sending discover
Fri Dec 25 22:34:38 2020 cron.err crond[15310]: crond (busybox 1.31.1) started, log level 8
Fri Dec 25 22:34:41 2020 daemon.notice netifd: lan3 (14943): udhcpc: sending discover
```

LGA1150, 2020-12-25:
重启可以连上吗？
如果把 LAN3 的 MAC 地址
```
config device 'lan_lan3_dev'
	option name 'lan3'
	option macaddr 'd4:ee:00:00:9f:f6'
```
改成别的，如 `d4:ee:00:00:9f:f0`，再重启可以连上吗？

Hey-Leo, 2020-12-25:
不行，连正常的wan都会掉
```
Fri Dec 25 22:46:46 2020 daemon.notice netifd: Interface 'wan' is now down
Fri Dec 25 22:46:46 2020 daemon.notice netifd: Interface 'wan' has link connectivity loss
Fri Dec 25 22:46:46 2020 daemon.notice netifd: Interface 'wan' has link connectivity
Fri Dec 25 22:46:46 2020 daemon.notice netifd: Interface 'wan' is setting up now
Fri Dec 25 22:46:46 2020 daemon.err insmod: module is already loaded - slhc
Fri Dec 25 22:46:46 2020 daemon.err insmod: module is already loaded - ppp_generic
Fri Dec 25 22:46:46 2020 daemon.err insmod: module is already loaded - pppox
Fri Dec 25 22:46:46 2020 daemon.err insmod: module is already loaded - pppoe
Fri Dec 25 22:46:46 2020 daemon.info pppd[7284]: Plugin rp-pppoe.so loaded.
Fri Dec 25 22:46:46 2020 daemon.info pppd[7284]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Fri Dec 25 22:46:46 2020 daemon.notice pppd[7284]: pppd 2.4.8 started by root, uid 0
Fri Dec 25 22:46:47 2020 daemon.info dnsmasq[7316]: Connected to system UBus
Fri Dec 25 22:46:47 2020 user.notice mwan3[7246]: Execute ifdown event on interface wan (unknown)
Fri Dec 25 22:46:47 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Dec 25 22:46:47 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Dec 25 22:46:49 2020 cron.err crond[7672]: crond (busybox 1.31.1) started, log level 8
Fri Dec 25 22:46:49 2020 daemon.info dnsmasq[7635]: Connected to system UBus
Fri Dec 25 22:46:50 2020 cron.err crond[7799]: crond (busybox 1.31.1) started, log level 8
Fri Dec 25 22:46:52 2020 cron.err crond[8161]: crond (busybox 1.31.1) started, log level 8
Fri Dec 25 22:46:54 2020 cron.err crond[8289]: crond (busybox 1.31.1) started, log level 8
Fri Dec 25 22:47:01 2020 daemon.warn pppd[7284]: Timeout waiting for PADO packets
Fri Dec 25 22:47:01 2020 daemon.err pppd[7284]: Unable to complete PPPoE Discovery
Fri Dec 25 22:47:01 2020 daemon.info pppd[7284]: Exit.
Fri Dec 25 22:47:01 2020 daemon.notice netifd: Interface 'wan' is now down
Fri Dec 25 22:47:01 2020 daemon.notice netifd: Interface 'wan' is setting up now
Fri Dec 25 22:47:01 2020 daemon.err insmod: module is already loaded - slhc
Fri Dec 25 22:47:02 2020 daemon.err insmod: module is already loaded - ppp_generic
Fri Dec 25 22:47:02 2020 daemon.err insmod: module is already loaded - pppox
Fri Dec 25 22:47:02 2020 daemon.err insmod: module is already loaded - pppoe
Fri Dec 25 22:47:02 2020 daemon.info pppd[8511]: Plugin rp-pppoe.so loaded.
Fri Dec 25 22:47:02 2020 daemon.info pppd[8511]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Fri Dec 25 22:47:02 2020 daemon.notice pppd[8511]: pppd 2.4.8 started by root, uid 0
```

Hey-Leo, 2020-12-25:
我重置了路由器，原本的正常network配置wan是秒上线的，
摸不着头脑，还是算了，，，

closed 2020-12-27 by Hey-Leo