coolsnowwolf/lede #5921: 关于NanoPi R2S的rtl8153网卡

tl768, 2020-12-12:
### rtl8153网卡挂载不上
**相应有驱动文件**
/lib/modules/5.4.79/r8152.ko
好像是驱动不上
固件是默认选项
那位高手指点下！

tl768, 2020-12-12:
lsusb
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

tl768, 2020-12-12:
dmesg
[    1.396904] usb usb3-port1: Cannot enable. Maybe the USB cable is bad?
[    2.296849] usb usb3-port1: Cannot enable. Maybe the USB cable is bad?
[    2.297742] usb usb3-port1: attempt power cycle
[    3.512956] usb usb3-port1: Cannot enable. Maybe the USB cable is bad?
[    4.412824] usb usb3-port1: Cannot enable. Maybe the USB cable is bad?
[    4.413702] usb usb3-port1: unable to enumerate USB device

tl768, 2020-12-12:
ls /lib/modules/`uname -r`/r815*
/lib/modules/5.4.79/r8152.ko

tl768, 2020-12-12:
dmesg | grep r8152
[    7.572490] usbcore: registered new interface driver r8152

programmejava, 2020-12-16:
按提示来看，很有可能是cpu引出的usb网口硬件问题，拆开看看网口附近有个芯片，用放大镜看看是不是有引脚虚焊，要么下载官方固件试试，

tl768, 2020-12-16:
> 按提示来看，很有可能是cpu引出的usb网口硬件问题，拆开看看网口附近有个芯片，用放大镜看看是不是有扭曲虚焊，可以下载官方固件试试，

FriendlyWRT固件使用正常
lean7月21号的固件使用正常
openwrt官方和最新lean的固件，lan口的rtl8153网卡不识别。不知道啥原因呀。郁闷。

programmejava, 2020-12-16:
那就拆开看看，在网口附近有个IC 四边很多引脚，检查是否有虚焊现象。电路虚焊有时候在不同的电压电流下工作就有异常

tl768, 2020-12-16:
> 那就拆开看看，在网口附近有个IC四边很多串联，检查是否有虚焊现象。电路虚焊有时候在不同的电压电流下工作就有异常

IC芯片看着太费劲了

FriendlyWRT固件网卡都正常
r8512驱动也正常

minlang112, 2020-12-18:
> > 那就拆开看看，在网口附近有个IC四边很多串联，检查是否有虚焊现象。电路虚焊有时候在不同的电压电流下工作就有异常
> 
> IC芯片看着太费劲了
> 
> FriendlyWRT固件网卡都正常
> r8512驱动也正常

用最新的代码库编译吧，已经把R4S的支持取消，uboot退回了

tl768, 2020-12-19:
> > > 那就拆开看看，在网口附近有个IC四边很多串联，检查是否有虚焊现象。电路虚焊有时候在不同的电压电流下工作就有异常
> > 
> > 
> > IC芯片看着太费劲了
> > FriendlyWRT固件网卡都正常
> > r8512驱动也正常
> 
> 用最新的代码库编译吧，已经把R4S的支持取消，uboot退回
好的，我试试。

closed 2021-02-23 by tl768

longbowapache, 2021-07-15:
@tl768  请问这个问题怎么解决的，我也遇到了

tl768, 2021-08-05:
> @tl768 请问 这个问题怎么解决的，我也遇到过

没解决呀！