coolsnowwolf/lede #5927: VMware ESXi x86 客户机死机

He-Peter-Huang, 2020-12-13:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：大概每两到三周虚拟机死机一次。Console 没反应，Ping不通。HOST机CPU显示高占用。

[image]

### (2) 路由器型号和固件版本

A：
HOST INFO:
ESXi-6.5.0-20180502001-standard (VMware, Inc.) （x86)

GUEST INFO:
Guest OS | Other 3.x or later Linux (64-bit)
Compatibility | ESXi 6.5 and later (VM version 13)
VMware Tools | No
CPUs | 4
Memory | 4 GB

OPENWRT INFO:
主机型号 | Intel(R) Xeon(R) CPU E31275 @ 3.40GHz : 4 Core 4 Thread (CpuMark : 81950.419996 Scores)
CPU 信息 | 3392.294 MHz
固件版本 | OpenWrt R20.10.20 / LuCI Master (git-20.256.12360-1a54222)
内核版本 | 5.4.75

### (3) 详细日志

A：
[image]

He-Peter-Huang, 2020-12-13:
经过搜索更多的Issue，不知是否和一下问题有关：
https://github.com/coolsnowwolf/lede/issues/5349#issuecomment-708977359

minlang112, 2020-12-14:
esxi6.7u3环境下未见崩溃现象，正常我一个月才重启一次……

Fddh2012, 2020-12-14:
我之前也遇见了，先是后台各种爆内存，然后web进不去，可以SSH连接，服务全上不了。 好像是看节点数量，要是节点太多，还都离线了，在测速时就会爆炸。

minlang112, 2020-12-14:
> 我之前也遇见了，先是后台各种爆内存，然后web进不去，可以SSH连接，服务全上不了。 好像是看节点数量，要是节点太多，还都离线了，在测速时就会爆炸。

……玩玩没想到居然有这种问题，看来我的鸡场不够高级（误

tcsr200722, 2020-12-14:
同样出现该问题  CPU占用100%

Dimitri2020007, 2020-12-14:
是不是同时编译了sfe和flowoffload？

看你截图的情况像是sfe和flowoffload起冲突了。

IRN-Kawakaze, 2020-12-15:
之前碰到过类似情况，web访问卡死不加载内容，ssh能进，内存和cpu跑满，一用命令行重启网络就彻底死机。怀疑是直通的usb网卡有问题，pppoe到期重播就死机，而且控制台有rtl网卡相关的错误（不过后来也没去解决）。可以看一下esxi的客户机控制台有没有网卡相关的错误。

axiaoni, 2021-01-07:
我每天都会这样，esxi6.7环境，尝试过更换内核，虚拟网卡关掉，仍然卡死

crypto75, 2021-01-11:
+1，死机很随机，有时候几天一次，有时候一天几次。几乎是默认配置，就选了luci里面几个常用的包。这几天编译过一次打开了死锁检测的内核，发现几乎每次启动lede的时候都会有dead lock提示，不知道和导致死机的原因有没有关系。以前4.19的内核用了一年也没死过一次机。

axiaoni, 2021-01-20:
好像是SFE的问题，我关掉后正常运行3天了，以前基本一天会死一次

9k001, 2021-02-03:
> 好像是SFE的问题，我关掉后正常运行3天了，以前基本一天会死一次

我跟你的情况一样
是Turbo ACC 网络加速带的 sfe
我现在关闭试试看

closed 2021-06-20 by coolsnowwolf