coolsnowwolf/lede #6029: 歌华链 用最新源码编译的固件无法启动

langliang3333, 2020-12-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
歌华链 用最新源码编译出的固件无法启动，插上电源开机后只有 POWER灯亮，隔一会USB/Internet 灯闪烁一下然后熄灭（可能自动重启了），12月10号 前的源码编译出的固件是可以正常启动的。breed下刷入我10号编译的固件  OpenWrt R20.10.20 / LuCI Master (git-20.343.54716-6fc079f) 可以正常启动

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
路由器型号：GeHua GHL-R-001  
A：

### (3) 详细日志

A：

asushugo, 2020-12-15:
对于这个问题，我准备今晚重新clone编译测试一下，因为前几天我也遇到了一样的问题，在红米AC2100上，就是在不保留配置文件的情况下刷最新版系统，系统看起来可以启动，但是不会分配IP地址，也不能通过在系统设置好静态地址来访问，之前的版本没问题。

asushugo, 2020-12-15:
我觉得要问一下大佬 @AmadeusGhost

AmadeusGhost, 2020-12-16:
> 我觉得要问一下 @AmadeusGhost

我萌新，试试 #6040

langliang3333, 2020-12-16:
我按 #6040 这个方法编译一个固件试试看

langliang3333, 2020-12-16:
> 
> 
> 我按 #6040 这个方法编译一个固件试试看

重新编译了固件测试还是不能启动

AmadeusGhost, 2020-12-16:
@asushugo 大佬有具体一点的时间说明是从哪个 commit 开始出问题的吗？不然只能拿到 ttl 启动 log 才能知道问题在哪了。

asushugo, 2020-12-17:
> @asushugo 大佬有具体一点的时间说明是从哪个 commit 开始出问题的吗？不然只能拿到 ttl 启动 log 才能知道问题在哪了。

我今晚回去把TTL输出一下吧，我的AC2100还没拆过呢哈哈哈哈

asushugo, 2020-12-17:
看TTL发现这里报错
[image]
发现文件里面这里不正确
[image]
但是之前已经merge了 #5869 编译出来的固件依然存在此问题
于是执行rm -rf build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/后重新编译，重新刷机后成功启动

langliang3333, 2020-12-20:
今天用最新源码，删除原来的源码后重新编译 测试成功启动

closed 2020-12-20 by langliang3333