coolsnowwolf/lede #6030: 莫名其妙下午六点左右重启

aape01, 2020-12-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
具体问题：莫名其妙下午六点左右重启

### (2) 路由器型号和固件版本

A：
路由器型号：Intel(R) Celeron(R) J4125 CPU @ 2.00GHz : 4 Core 4 Thread (CpuMark : 59857.837636 Scores)
固件版本：OpenWrt R20.10.20 / LuCI Master (git-20.343.54716-6fc079f)

### (3) 详细日志

A：系统日志
Mon Dec 14 18:12:53 2020 daemon.info pppd[9448]: Using interface pppoe-wan
Mon Dec 14 18:12:53 2020 daemon.notice pppd[9448]: Connect: pppoe-wan <--> eth0
Mon Dec 14 18:12:53 2020 daemon.info pppd[9448]: syncppp not active
Mon Dec 14 18:12:53 2020 daemon.info pppd[9448]: Remote message: Authentication failed!
Mon Dec 14 18:12:53 2020 daemon.err pppd[9448]: PAP authentication failed
Mon Dec 14 18:12:53 2020 daemon.notice pppd[9448]: Connection terminated.
Mon Dec 14 18:12:53 2020 daemon.info pppd[9448]: Sent PADT
Mon Dec 14 18:12:53 2020 daemon.info pppd[9448]: Exit.
Mon Dec 14 18:12:53 2020 daemon.notice netifd: Interface 'wan' is now down
Mon Dec 14 18:12:53 2020 daemon.notice netifd: Interface 'wan' is setting up now
Mon Dec 14 18:12:53 2020 daemon.err insmod: module is already loaded - slhc
Mon Dec 14 18:12:53 2020 daemon.err insmod: module is already loaded - ppp_generic
Mon Dec 14 18:12:53 2020 daemon.err insmod: module is already loaded - pppox
Mon Dec 14 18:12:53 2020 daemon.err insmod: module is already loaded - pppoe
Mon Dec 14 18:12:53 2020 daemon.notice netifd: wan (9478): uci: Entry not found
Mon Dec 14 18:12:53 2020 daemon.notice netifd: wan (9478): sh: out of range
Mon Dec 14 18:12:53 2020 daemon.info pppd[9507]: Plugin rp-pppoe.so loaded.
Mon Dec 14 18:12:53 2020 daemon.info pppd[9507]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Mon Dec 14 18:12:53 2020 daemon.notice pppd[9507]: pppd 2.4.8 started by root, uid 0
Mon Dec 14 18:12:53 2020 daemon.info pppd[9507]: PPP session is 51698
Mon Dec 14 18:12:53 2020 daemon.warn pppd[9507]: Connected to cc:1a:fa:ed:55:80 via interface eth0
Mon Dec 14 18:12:53 2020 kern.info kernel: [   68.529191] pppoe-wan: renamed from ppp0
Mon Dec 14 18:12:53 2020 daemon.info pppd[9507]: Renamed interface ppp0 to pppoe-wan
Mon Dec 14 18:12:53 2020 daemon.info pppd[9507]: Using interface pppoe-wan
Mon Dec 14 18:12:53 2020 daemon.notice pppd[9507]: Connect: pppoe-wan <--> eth0
Mon Dec 14 18:12:53 2020 daemon.info pppd[9507]: syncppp not active
Mon Dec 14 18:12:53 2020 daemon.info pppd[9507]: Remote message: Authentication failed!
Mon Dec 14 18:12:53 2020 daemon.err pppd[9507]: PAP authentication failed
Mon Dec 14 18:12:53 2020 daemon.notice pppd[9507]: Connection terminated.
Mon Dec 14 18:12:53 2020 daemon.info pppd[9507]: Sent PADT
Mon Dec 14 18:12:53 2020 daemon.info pppd[9507]: Exit.
Mon Dec 14 18:12:53 2020 daemon.notice netifd: Interface 'wan' is now down
Mon Dec 14 18:12:53 2020 daemon.notice netifd: Interface 'wan' is setting up now
Mon Dec 14 18:12:53 2020 daemon.err insmod: module is already loaded - slhc
Mon Dec 14 18:12:53 2020 daemon.err insmod: module is already loaded - ppp_generic
Mon Dec 14 18:12:53 2020 daemon.err insmod: module is already loaded - pppox
Mon Dec 14 18:12:53 2020 daemon.err insmod: module is already loaded - pppoe
Mon Dec 14 18:12:53 2020 daemon.notice netifd: wan (9538): uci: Entry not found
Mon Dec 14 18:12:53 2020 daemon.notice netifd: wan (9538): sh: out of range
Mon Dec 14 18:12:53 2020 daemon.info pppd[9567]: Plugin rp-pppoe.so loaded.
Mon Dec 14 18:12:53 2020 daemon.info pppd[9567]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Mon Dec 14 18:12:53 2020 daemon.notice pppd[9567]: pppd 2.4.8 started by root, uid 0
Mon Dec 14 18:12:54 2020 daemon.info pppd[9567]: PPP session is 51703
Mon Dec 14 18:12:54 2020 daemon.warn pppd[9567]: Connected to cc:1a:fa:ed:55:80 via interface eth0
Mon Dec 14 18:12:54 2020 kern.info kernel: [   69.031317] pppoe-wan: renamed from ppp0
Mon Dec 14 18:12:54 2020 daemon.info pppd[9567]: Renamed interface ppp0 to pppoe-wan
Mon Dec 14 18:12:54 2020 daemon.info pppd[9567]: Using interface pppoe-wan
[... 1658 lines cut ...]
[   73.978428] pppoe-wan: renamed from ppp0
[   74.376848] pppoe-wan: renamed from ppp0
[   74.728282] pppoe-wan: renamed from ppp0
[   75.177412] pppoe-wan: renamed from ppp0
[   75.576766] pppoe-wan: renamed from ppp0
[   76.028945] pppoe-wan: renamed from ppp0
[   76.477070] pppoe-wan: renamed from ppp0
[   76.876524] pppoe-wan: renamed from ppp0
[   77.276707] pppoe-wan: renamed from ppp0
[   77.678092] pppoe-wan: renamed from ppp0
[   78.077799] pppoe-wan: renamed from ppp0
[   78.477175] pppoe-wan: renamed from ppp0
[   78.877667] pppoe-wan: renamed from ppp0
[   79.280990] pppoe-wan: renamed from ppp0
[   79.627187] pppoe-wan: renamed from ppp0
[   80.077921] pppoe-wan: renamed from ppp0
[   80.478172] pppoe-wan: renamed from ppp0
[   80.827479] pppoe-wan: renamed from ppp0
[   81.256890] pppoe-wan: renamed from ppp0
[   81.657385] pppoe-wan: renamed from ppp0
[   82.108341] pppoe-wan: renamed from ppp0
[   82.507612] pppoe-wan: renamed from ppp0
[   82.912782] pppoe-wan: renamed from ppp0
[   83.378460] pppoe-wan: renamed from ppp0
[   83.776448] pppoe-wan: renamed from ppp0

caoyipeng2008, 2020-12-18:
是不是 watchcat设置的问题？

aape01, 2020-12-19:
> 是不是 watchcat设置的问题？

我没设置过watchcat，还有自己编译的X86固件打开百度都有0.5秒左右的延迟

aape01, 2020-12-19:
2020-12-19更新日志

helianling, 2020-12-21:
同样的J4125+4口RTK8125B 2.5G成品软路由，自编译的最新固件完全没有问题。

closed 2021-06-20 by coolsnowwolf