coolsnowwolf/lede #6031: K2P必须断电重启，5Gwifi无法修改信道！！

qjiang888, 2020-12-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：K2P必须断电几次才能启动好，web里点重启基本无法启动，5Gwifi无法修改信道！！

### (2) 路由器型号和固件版本

A：K2P

### (3) 详细日志

A：Mon Dec 14 19:17:52 2020 kern.notice kernel: [    0.000000] Linux version 5.4.79 (qjiang@ubuntu) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2973-59fe7c1ac)) #0 SMP Fri Dec 11 02:36:13 2020
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] MIPS: machine is Phicomm K2P
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Mon Dec 14 19:17:52 2020 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Mon Dec 14 19:17:52 2020 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Mon Dec 14 19:17:52 2020 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] Zone ranges:
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000]   HighMem  empty
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] Movable zone start for each node
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] Early memory node ranges
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Mon Dec 14 19:17:52 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Mon Dec 14 19:17:52 2020 kern.debug kernel: [    0.000000]   Normal zone: 288 pages used for memmap
Mon Dec 14 19:17:52 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Mon Dec 14 19:17:52 2020 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s27184 r8192 d21968 u57344
Mon Dec 14 19:17:52 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s27184 r8192 d21968 u57344 alloc=14*4096
Mon Dec 14 19:17:52 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Mon Dec 14 19:17:52 2020 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] Writing ErrCtl register=00019200
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] Readback ErrCtl register=00019200
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] Memory: 120776K/131072K available (5722K kernel code, 249K rwdata, 1240K rodata, 1288K init, 238K bss, 10296K reserved, 0K cma-reserved, 0K highmem)
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] NR_IRQS: 256
Mon Dec 14 19:17:52 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x340/0x550 with crng_init=0
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.015479] Calibrating delay loop... 583.68 BogoMIPS (lpj=1167360)
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.055819] pid_max: default: 32768 minimum: 301
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.065161] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.079563] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.097623] rcu: Hierarchical SRCU implementation.
Mon Dec 14 19:17:52 2020 kern.info kernel: [    0.107719] smp: Bringing up secondary CPUs ...
Mon Dec 14 19:17:52 2020 kern.warn kernel: [    3.242943] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Mon Dec 14 19:17:52 2020 kern.warn kernel: [    3.242952] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Mon Dec 14 19:17:52 2020 kern.warn kernel: [    3.242964] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Mon Dec 14 19:17:52 2020 kern.info kernel: [    3.243065] CPU1 revision is: 0001992f (MIPS 1004Kc)
[... 505 lines cut ...]
Mon Dec 14 19:18:06 2020 kern.warn kernel: [   33.909097] 1
Mon Dec 14 19:18:06 2020 kern.warn kernel: [   33.912221] 5
Mon Dec 14 19:18:06 2020 kern.warn kernel: [   33.915371] 6
Mon Dec 14 19:18:06 2020 kern.warn kernel: [   33.918527] 0
Mon Dec 14 19:18:06 2020 kern.warn kernel: [   33.921681] 3
Mon Dec 14 19:18:21 2020 daemon.notice netifd: rax0 (1331): Interface ra0 now up.
Mon Dec 14 19:18:21 2020 daemon.notice netifd: rax0 (1331): /tmp/profiles/mt_dbdc_cmd_2g.sh: line 4: iwpriv: not found
Mon Dec 14 19:18:21 2020 daemon.notice netifd: rax0 (1331): /tmp/profiles/mt_dbdc_cmd_2g.sh: line 5: iwpriv: not found
Mon Dec 14 19:18:21 2020 daemon.notice netifd: rax0 (1331): /tmp/profiles/mt_dbdc_cmd_2g.sh: line 6: iwpriv: not found
Mon Dec 14 19:18:21 2020 daemon.err dnscache[2541]: * 12/14 19:18:21| pdnsd: info: pdnsd-1.2.9b-par starting.
Mon Dec 14 19:18:24 2020 daemon.notice procd: /etc/rc.d/S90kms: KMS Server has started.
Mon Dec 14 19:18:24 2020 daemon.notice procd: /etc/rc.d/S91wrtbwmon: Warning: using existing /tmp/usage.db
Mon Dec 14 19:18:25 2020 user.info watchcat: starting all tasks
Mon Dec 14 19:18:25 2020 user.info watchcat: started task (mode=ping;period=21600;pinghosts=8.8.8.8;pingperiod=1080;forcedelay=30)
Mon Dec 14 19:18:25 2020 user.info watchcat: all tasks started
Mon Dec 14 19:18:25 2020 daemon.notice netifd: rax0 (1331): Interface rax0 now up.
Mon Dec 14 19:18:25 2020 daemon.notice netifd: rax0 (1331): /tmp/profiles/mt_dbdc_cmd_5g.sh: line 4: iwpriv: not found
Mon Dec 14 19:18:25 2020 daemon.notice netifd: rax0 (1331): /tmp/profiles/mt_dbdc_cmd_5g.sh: line 5: iwpriv: not found
Mon Dec 14 19:18:25 2020 daemon.notice netifd: rax0 (1331): /tmp/profiles/mt_dbdc_cmd_5g.sh: line 6: iwpriv: not found
Mon Dec 14 19:18:25 2020 daemon.notice netifd: rax0 (1331): /tmp/profiles/mt_dbdc_cmd_5g.sh: line 7: iwpriv: not found
Mon Dec 14 19:18:26 2020 daemon.notice netifd: Network device 'rax0' link is up
Mon Dec 14 19:18:26 2020 daemon.notice netifd: Network device 'ra0' link is up
Mon Dec 14 19:18:26 2020 daemon.info procd: - init complete -
Mon Dec 14 19:51:41 2020 daemon.info pppd[1422]: System time change detected.
Mon Dec 14 19:51:44 2020 daemon.info dnsmasq[3543]: Connected to system UBus

SilverHunterLib, 2020-12-14:
我的K2P刷完最新12.12版固件，重启之后无法连接路由器，反复重复了3、4次还是不行，只能刷回原来的，不知道啥原因

ltactor, 2020-12-15:
> 我的K2P刷完最新12.12版固件，重启之后无法连接路由器，反复重复了3、4次还是不行，只能刷回原来的，不知道啥原因

我也是。我以为是我的问题。我只选默认配置也不可以。

wsspsb, 2020-12-17:
我去  搞我一周多了 我还以为自己问题没编译对, 不过我试过默认配置多次,可以重启正常.
我现在也是重启后一直蓝灯,但是路由器后台进不去,wifi没有.
我勾选的插件会有重启问题:
[image]

勾选这几个插件就能正常启动(默认配置我试过也行)
[image]

wsspsb, 2020-12-17:
顺便问下你这个日志没法开机的话是怎么看到的,概率性开机后进路由器后台看到的吗

wsspsb, 2020-12-18:
我k2p更新2020-12-17的那次提交,可以正常启动了

qjiang888, 2020-12-18:
我最新的也测试了，breed下先刷openwrt-ramips-mt7621-phicomm_k2p-initramfs-kernel 再刷openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade 一切正常， 只要改完IP 无线SSID 什么的  再次重启就 启动不了， 不停的插拔电源 就有可能可以启动！！

qjiang888, 2020-12-18:
无线信道修改是无效的

ltactor, 2020-12-19:
> 我最新的也测试了，breed下先刷openwrt-ramips-mt7621-phicomm_k2p-initramfs-kernel 再刷openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade 一切正常， 只要改完IP 无线SSID 什么的 再次重启就 启动不了， 不停的插拔电源 就有可能可以启动！！

k2p  硬改32M   选ipv6 wireguard  v2ray     编译成功，可以刷入，重启失败。    12月18日固件。

closed 2020-12-23 by coolsnowwolf

qjiang888, 2020-12-23:
谢谢大佬们的回复，我最新的再测试下！！！

PR #6085 (2020-12-23): ramips: fix k2p 32m reboot problems
fix #6031 , fix #6056 , fix #6077 .