coolsnowwolf/lede #6032: 请教树莓派docker转发不正常

yah-yuan, 2020-12-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

使用 macvlan 的方法在树莓派4设置了L大的docker版本作为旁路由使用，主路由所有配置不变，需要审计的设备手动配置到旁路由上。

然而发包无法正常被转发，所有包均trace到旁路由为止。奇怪的是使用了passwall后可以正常访问被代理的地址（google.com），不被代理的仍然无法访问（baidu.com）。

路由表正常，0.0.0.0指向主路由。

### (2) 路由器型号和固件版本

A：

Raspberry Pi 4 Model B Rev 1.1

Linux pi 5.4.0-1025-raspi #28-Ubuntu SMP PREEMPT

OpenWrt R20.12.12 / LuCI Master (git-20.343.54716-6fc079f)

### (3) 详细日志

A：

sakuyamaij, 2020-12-15:
我的使用环境和方法和楼主完全相同，同时使用了openclash作为翻墙的网关和dns，一切正常。
楼主参考
https://juejin.cn/post/6844904083409600526
https://mlapp.cn/376.html
这两篇文章看看自己步骤是否有没有问题吧

yah-yuan, 2020-12-15:
> 我的使用环境和方法和楼主完全相同，同时使用了openclash作为翻墙的网关和dns，一切正常。
> 楼主参考
> https://juejin.cn/post/6844904083409600526
> https://mlapp.cn/376.html
> 这两篇文章看看自己步骤是否有没有问题吧

感谢回复。设置过程参考了文2的方法。
矛盾点在于：文2的步骤我完全照做了，而文2中没有做代理就可以让openwrt做网关了。然而我的情况是不设置代理就不能由openwrt路由访问任何外网域名或是ip，设置了代理却能正常访问被代理的域名或ip。所以我的推测是网络没问题，但是直接路由转发没有加密的都被某个网卡丢弃了？
另外，我在文1的评论区也发现了跟我同样的情况，他的方法是全局代理……很尴尬

> 小屁孩_H lv-1
> 你好，我设置完发现网关设置的openwrt上了网，电脑开科*学*上*网*代理时候能进入国外网址。国内的不能上是什么原因

求赐教

j999yp, 2020-12-18:
环境不同但是故障类似：我是x86软路由做主路由，k2p刷老毛子做AP，电脑可以正常上网，但是除了电脑以外的一切设备都无法正常上网。排除故障的过程中发现除电脑以外的设备在speedtest中下载速度正常但上传速度为0，使用traceroute发现所有包全部停在了软路由上。

saltfishh, 2020-12-27:
很奇怪，我和你环境一样，raspi4 ubuntu server，docker镜像是自己编译完导入进去的。
trace都是先经旁路由再过主路由到公网。
经clash代理的域名规则才能出去这个就很迷…是不是openwrt防火墙的问题？

yah-yuan, 2021-01-05:
> openwrt

防火墙我这边全部放行了...

yah-yuan, 2021-03-11:
似乎是Ubuntu的锅，教程是使用Debian的。。。换了Raspbian就好了

closed 2021-03-11 by yah-yuan