coolsnowwolf/lede #6034: SSR+在GFW列表模式下，只有GFW列表中的网站会用国外DNS服务器解析吗

NagaseKouichi, 2020-12-15:
请问各位大佬，SSR+在GFW列表模式下，只有GFW列表中的网站会用国外DNS服务器解析吗？

现在发现SSR+在GFW列表模式下，某些使用了cloudflare CDN的网站会出现DNS污染，比如u2.dmhy.org。

有没有办法让SSR+在GFW列表模式下，所有的国外域名都通过国外DNS服务器解析（使用DOH或者DOT），国内域名则通过自带的dnsmasq转发到Turbo ACC的DNS加速。但是科学上网还是只有GFW列表中被墙的网站才爬墙，而不是所有国外网站都爬墙。

感谢大佬指教。

NagaseKouichi, 2020-12-15:
或者我是不是可以这样解决，主路由的SSR+使用GFW列表模式用于爬墙，在额外网关（旁路由）上的SSR+使用“绕过大陆IP模式”用来解析国外的所有域名。然后在主路由的DHCP设置中让所有设备都从旁路由上获取DNS

Dimitri2020007, 2020-12-16:
ssr plus dns解析使用5335端口，装smartdns，开两个smartdns进程，一路专走国内解析，一路专走国外5335（doh或者dot）。

这个对所有网站不一定有效。

NagaseKouichi, 2020-12-16:
> ssr plus dns解析使用5335端口，装smartdns，开两个smartdns进程，一路专走国内解析，一路专走国外5335（doh或者dot）。
> 
> 这个对所有网站不一定有效。

这个我研究了下，应该也是不能解决这种情况。因为首先会由SSR+的GFW列表来判断这个域名是不是在规则内，满足规则才会走国外5335解析，但是有些被污染的域名并不在GFW列表里面，所以还是会走国内解析，导致DNS污染。

我暂时安装了chinadns解决了，会自动抛弃国内的dns服务器返回的国外IP，缺点是没有ipv6的dns解析结果了

Dimitri2020007, 2020-12-16:
chinadns在lean openwrt源码里吗？

还是需要单独编译？

hsts错误真是好烦人

NagaseKouichi, 2020-12-16:
> chinadns在lean openwrt源码里吗？
> 
> 还是需要单独编译？
> 
> 国外域名hsts错误真是好烦人

https://github.com/aa65535/openwrt-chinadns
https://github.com/aa65535/openwrt-dist-luci
从这里下ipk安装就可以

Dimitri2020007, 2020-12-16:
当然是添加到源码编译喽

话说你虚拟机是哪个？esxi、pve、unraid还是hyper-v

NagaseKouichi, 2020-12-16:
> 当然是添加到源码编译喽
> 
> 话说你虚拟机是哪个？esxi、pve、unraid还是hyper-v

我的旁路由和测试用的op，有运行在PVE下面的也有运行在威联通NAS的KVM虚拟机下面的，不过运行在PVE的旁路由不是用虚拟机运行，是用LXC容器运行的

Dimitri2020007, 2020-12-16:
我运行的是双lean openwrt（虚拟机）

NagaseKouichi, 2020-12-16:
> 我运行的是双lean openwrt（虚拟机）
> 
> lxc设置是不是比虚拟机复杂些？

安装要麻烦点，需要懂lxc的一些操作

Dimitri2020007, 2020-12-16:
你可以把旁路由装到虚拟机下试试

NagaseKouichi, 2020-12-16:
> 你可以把旁路由装到虚拟机下试试

用LXC装的这台旁路由安装chinadns之后启动没问题，有问题的是一台虚拟机上的测试用的OP，可能那个OP编译的时候选的有些内核设置有冲突，今天重新编译了一个升级之后启动chinadns没问题了

Dimitri2020007, 2020-12-16:
lxc容器可以装双软路由吗？

NagaseKouichi, 2020-12-17:
> lxc容器可以装双软路由吗？

运行几个都可以，每个容器是单独的IP，但是不建议做主路由

Dimitri2020007, 2020-12-17:
你现在这个配置ssr plus怎么做到国内外域名分流解析的？

如果可以需要什么插件？

可以实现国内外分流解析而互不干扰吗？

NagaseKouichi, 2020-12-17:
> 你现在这个配置ssr plus怎么做到国内外域名分流解析的？
> 
> 如果可以需要什么插件？
> 
> 可以实现国内外分流解析而互不干扰吗？

就是用的chinadns啊，具体的设置我写了个博客，可以参考下：https://iwan.ga/archives/368

Dimitri2020007, 2020-12-17:
好吧我测试下，稍晚些时候看看效果如何

libinanshu, 2020-12-17:
> ssr plus dns解析使用5335端口，装smartdns，开两个smartdns进程，一路专走国内解析，一路专走国外5335（doh或者dot）。
> 
> 这个对所有网站不一定有效。

我一直好奇一个问题，国外网址DNS查询需要节点去查询不是本机区去查询，ssr+是节点查询的。smartdns怎么做到使用节点去查询？

NagaseKouichi, 2020-12-17:
> > ssr plus dns解析使用5335端口，装smartdns，开两个smartdns进程，一路专走国内解析，一路专走国外5335（doh或者dot）。
> > 这个对所有网站不一定有效。
> 
> 我一直好奇一个问题，国外网址DNS查询需要节点去查询不是本机区去查询，ssr+是节点查询的。smartdns怎么做到使用节点去查询？

把smartdns的外国组DNS服务器放到SSR+的强制代理列表里面就可以吧。不过我不用smartdns，smartdns还是需要搭配chinadns才能做国内外域名解析分流，基本是多此一举。

Dimitri2020007, 2020-12-17:
我已经在lean openwrt里编译好了passwall包含chinadns-ng是不是可以和lean新的https-dns-proxy执行国内外域名分流解析

hyyz17200, 2020-12-22:
> > > ssr plus dns解析使用5335端口，装smartdns，开两个smartdns进程，一路专走国内解析，一路专走国外5335（doh或者dot）。
> > > 这个对所有网站不一定有效。
> > 
> > 
> > 我一直好奇一个问题，国外网址DNS查询需要节点去查询不是本机区去查询，ssr+是节点查询的。smartdns怎么做到使用节点去查询？
> 
> 把smartdns的外国组DNS服务器放到SSR+的强制代理列表里面就可以吧。不过我不用smartdns，smartdns还是需要搭配chinadns才能做国内外域名解析分流，基本是多此一举。

smartdns搭配ssr+就可以国内外域名分流，我一直这样用。ssr+启用以后好像就会区分的。

Dimitri2020007, 2020-12-22:
> smartdns搭配ssr+就可以国内外域名分流，我一直这样用。ssr+启用以后好像就会区分的。

smartdns怎么设置才能实现国内外域名分流，如果是双软路由该咋做？

hyyz17200, 2020-12-22:
> smartdns怎么设置才能实现国内外域名分流，如果是双软路由该咋做？

smartdns luci界面里有第二dns服务器，打开，端口设置5335给ssr+用。然后第二dns服务器使用非默认组，你可以自己起一个名字。

在dns里加一个或多个你需要的用的国外dns，添加额外的服务器参数 -exclude-default-group。 

就可以了。

Dimitri2020007, 2020-12-22:
好的，若是主路由开启了smartdns，旁路由是不是就可以直接把dns地址指向开了smartdns的路由器？

可以pdnsd指到smartdns解析吗？

hyyz17200, 2020-12-22:
> 好的，若是主路由开启了smartdns，旁路由是不是就可以直接把dns地址指向开了smartdns的路由器？
> 
> 可以pdnsd指到smartdns解析吗？

当然可以的，我在公司里就是用虚拟机开的旁路由，手机dns网关指向该旁路由，就可以用它科学上网的。

pdnsd可以指向，但是smartdns的两个服务器是分别解析国内和国外的。如果你要自动分流恐怕不行啊，还是用路由自己的127.0.0.1比较好。

我也是开smartdns作为dnsmasq的上游，没有问题。

国外我是正常的，刚忘了说你国外的dns要用tcp方式请求。（默认情况下v2ray这些透明代理只代理tcp流量，udp一定会直连）。我猜你是不是用了默认udp访问。求保险的话，在ssr+里把你国外dns再加入强制列表。

我用gfw模式和绕过大陆都是可以的。但是绕过大陆我个人一般不爱用，我自己的ISP访问非墙国外速度还可以的。

Dimitri2020007, 2020-12-22:
好吧我再试下

NagaseKouichi, 2020-12-23:
> smartdns搭配ssr+就可以国内外域名分流，我一直这样用。ssr+启用以后好像就会区分的。

你没看我前面写的吗？SSR+的分流只会让在GFW列表中的域名通过国外DNS服务器去解析，而不是所有国外域名都通过国外DNS服务器解析。有些国外的域名没有在GFW列表中，一样会被污染。用chinadns就可以解决这些域名污染的问题

Dimitri2020007, 2020-12-23:
这个应该和你所在的地区、运营商线路有关。

但不管你用的是啥软件分流，只要它有用，那就是最好的。

closed 2021-06-20 by coolsnowwolf