coolsnowwolf/lede #6035: cifs-mount开机时无法正常挂载其他设备的局域网共享

EnnawYang, 2020-12-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：cifs-mount开机时无法正常挂载其他设备的局域网共享，但从luci里面点击应用可以挂载，尝试装了nfs-utils无效

### (2) 路由器型号和固件版本

A：x86_64 EFI  2020-12-15编译

### (3) 详细日志

A：
系统日志:
```
Tue Dec 15 10:43:33 2020 kern.info kernel: [   25.713612] CIFS: Attempting to mount //192.168.10.132/视频
Tue Dec 15 10:43:33 2020 daemon.notice procd: /etc/rc.d/S97cifs: mount: /mnt/pc-video: bad option; for several filesystems (e.g. nfs, cifs) you might need a /sbin/mount.<type> helper program.
```
ssh:
```
root@OpenWrt_x86:~# /etc/init.d/cifs stop
root@OpenWrt_x86:~# /etc/init.d/cifs start
mount: /mnt/pcvideo: bad option; for several filesystems (e.g. nfs, cifs) you might need a /sbin/mount.<type> helper program.
```

yusurvival, 2021-04-08:
先检查软件包中是否有
kmod-fs-cifs kmod-nls-utf8 kmod-nls-base kmod-crypto-hmac kmod-crypto-md5 kmod-crypto-misc
这些需要单独编译安装（以上为网上其他资料，因我的lede有这些软件包，不确定是否为必须，下面是本人解决方法）
如软件包中有cifsmount直接第二步。
1、
opkg update
opkg install cifsmount
2、
cp /usr/sbin/mount.cifs /sbin/mount.cifs
cp /usr/sbin/mount.smb3 /sbin/mount.smb3
即安装cifsmount（就是cifs-utils），但是这个ipk会吧mount.cifs生成到/usr/sbin文件夹下，直接调用的话就把这个文件拷贝到根目录的/sbin下,拷贝完以后应该就可以了。

closed 2021-06-20 by coolsnowwolf

LarryLi-cn, 2022-12-23:
我也遇到同样的问题， 请问解决了吗？

LarryLi-cn, 2022-12-23:
> 先检查软件包中是否有 kmod-fs-cifs kmod-nls-utf8 kmod-nls-base kmod-crypto-hmac kmod-crypto-md5 kmod-crypto-misc 这些需要单独编译安装（以上为网上其他资料，因我的lede有这些软件包，不确定是否为必须，下面是本人解决方法） 如软件包中有cifsmount直接第二步。 1、 opkg update opkg [...]

我也遇到同样的问题， 只有web端手动可以挂载。 
kmod-fs-cifs 有安装的
但sbin目录下没有mount.cifs

LarryLi-cn, 2022-12-25:
已解决， 方法： 
安装cifsmount后， 将原来的mount -t cifs 指令替换为 mount.cifs就可以了