coolsnowwolf/lede #6036: 最新版本更新订阅之后无法启动ssr

zhudan, 2020-12-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
最新版本更新订阅之后无法启动ssr
程序日志： 2020-12-15 12:37:18 Main node:Can't find  program, can't start!
Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：2020-12-15 12:37:18 Main node:Can't find  program, can't start!

amosly, 2020-12-15:
同样发现了这个问题

amosly, 2020-12-15:
发现节点类型默认是ssr,手动改回vmess可以运行了

hcym, 2020-12-15:
瓦工机场没事，早上编译的版本

jxph1123, 2020-12-15:
是今天才编译的吗？好像是修正了。不知道你是好久编译的版本。

zhudan, 2020-12-15:
> 是今天才编译的吗？好像是修正了。不知道你是好久编译的版本。

这个版本 https://github.com/coolsnowwolf/lede/commit/37836e0c3db69dd64cd30e7a4a415365ee1f074e

zhudan, 2020-12-15:
> 是今天才编译的吗？好像是修正了。不知道你是好久编译的版本。

貌似有错误提示了全部节点都无效，然后就就报Main node:Can't find program, can't start!。 然后就死活无法启动xray了，只能重新刷机

Ljzkirito, 2020-12-15:
执行过`./scripts/feeds update -a && ./scripts/feeds install -a`吗？
ssr与coolsnowwolf/lede无关，与[fw876/helloworld](https://github.com/fw876/helloworld)有关
你看看ssr版本是不是181-4

zhudan, 2020-12-15:
> 执行过`./scripts/feeds update -a && ./scripts/feeds install -a`吗？
> ssr与coolsnowwolf/lede无关，与[fw876/helloworld](https://github.com/fw876/helloworld)有关
> 你看看ssr版本是不是181-4

https://github.com/fw876/helloworld/commit/021fd8e6f8894269f9a27d44d6bf19362a282ae5

貌似隔壁解决了，感谢

closed 2020-12-15 by zhudan