coolsnowwolf/lede #6038: 现在支不支持R4S？

31415926pai, 2020-12-15:
请问大佬，什么时候支持R4S ?

NagaseKouichi, 2020-12-15:
已经支持了吧

AmadeusGhost, 2020-12-15:
别问，问就是不支持

programmejava, 2020-12-15:
可以参照杂交r2s的方法编译r4s   我已经编译成功了，并在官方固件ubuntu core的docker中也可以运行了，放到v2ex上各种说怕投毒的，说原版openwrt最好的，叨逼叨半天，，就不放固件了，自己编译

IamDLY, 2020-12-17:
最新已经移除了r4s的支持

lifetyper, 2020-12-17:
因为某商家只肯给某开发者包邮省6块钱的优惠？

AmadeusGhost, 2020-12-17:
@lifetyper 你说的某开发者不是说了为了支持 r4s 而升级 uboot 反而会导致 r2s 没之前稳定

d975xbx, 2020-12-18:
不支持是对的。智商检测软路由

DanielChen008, 2020-12-18:
啊…又不支持了么？我是之前看到支持，买了一个才到…这个也不叫智商检测软路由吧，可以塞进弱电箱就很方便了。R2S 我有3个，因为跑不满千兆，所以放在公司、父母家、女朋友屋子各一个。自己家是电信千兆目前用的 J1900搁在电视柜上，黑黑一坨太占地方…那我岂不是只有退货了…

programmejava, 2020-12-19:
> 啊…又不支持了么？我是之前看到支持，买了一个才到…这个也不叫智商检测软路由吧，可以塞进弱电箱就很方便了。R2S 我有3个，因为跑不满千兆，所以放在公司、父母家、女朋友屋子各一个。自己家是电信千兆目前用的 J1900搁在电视柜上，黑黑一坨太占地方…那我岂不是只有退货了…

我编译的R4S的杂交固件，给你测试测试
https://mega.nz/file/8EVihDzI#3oX1v94I4QTvjoAzdSK-18m1isPKjE2RmEpqhxSd3dY

[image]

31415926pai, 2020-12-20:
> 可以参照杂交r2s的方法编译r4s 我已经编译成功了，并在官方固件ubuntu core的docker中也可以运行了，放到v2ex上各种说怕投毒的，说原版openwrt最好的，叨逼叨半天，，就不放固件了，自己编译

谢谢分享。

31415926pai, 2020-12-20:
> > 啊…又不支持了么？我是之前看到支持，买了一个才到…这个也不叫智商检测软路由吧，可以塞进弱电箱就很方便了。R2S 我有3个，因为跑不满千兆，所以放在公司、父母家、女朋友屋子各一个。自己家是电信千兆目前用的 J1900搁在电视柜上，黑黑一坨太占地方…那我岂不是只有退货了…
> 
> 我编译的R4S的杂交固件，给你测试测试
> https://mega.nz/file/8EVihDzI#3oX1v94I4QTvjoAzdSK-18m1isPKjE2RmEpqhxSd3dY
> 
> [image]
> 
> 图片是我在ubuntu core中的docker运行的这个杂交固件，官方的ubuntu core现在全线是4.9的内核,我编译的是5.4内核 ，因在docker中运行所以显示4.9的内核

谢谢分享。

mcucore, 2020-12-21:
配置不上不下，全配后价格也不便宜，整个小主机他不香吗？
如果AX5 AX6之流有驱动，也就没它啥事了。

DanielChen008, 2020-12-22:
可以关闭此 ticket 了，源码已支持…
同为开发者，理解并支持此项目中各位大大的决定，辛苦了！

XGHeaven, 2020-12-28:
@mcucore 讲道理配置可以顶 j1900 了，但是 j1900 准系统接近 500 块大洋，一套下来也要差不多 800，都能买个全新满配 r4s + 8 口交换机了，甚至还能换个 16 口，整体功耗（r4s 光做路由空载使用最大 3w 左右，交换机一般满载 6w，整体不到 10w）还比 j1900（待机就要 10w）低。

Michael728, 2021-04-15:
> @mcucore 讲道理配置可以顶 j1900 了，但是 j1900 准系统接近 500 块大洋，一套下来也要差不多 800，都能买个全新满配 r4s + 8 口交换机了，甚至还能换个 16 口，整体功耗（r4s 光做路由空载使用最大 3w 左右，交换机一般满载 6w，整体不到 10w）还比 j1900（待机就要 10w）低。

R4S的固件怎么编译呀？

closed 2021-06-20 by coolsnowwolf