coolsnowwolf/lede #6039: boot挂载bug

WYC-2020, 2020-12-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
[image]
boot分区挂载出现重复挂载
挂载生成配置boot无法识别
A：

### (2) 路由器型号和固件版本
软路由j4105，最新版编译
A：

### (3) 详细日志

A：

EnnawYang, 2020-12-16:
正常，忽略即可

jiangzm, 2020-12-17:
> 正常，忽略即可

这正常在哪里？linux没这问题，openwrt 非x86也没这问题

EnnawYang, 2020-12-17:
。。。

jiangzm, 2020-12-18:
@WYC-2020  翻源码找到了重复挂载的原因

代码在：https://github.com/coolsnowwolf/lede/blob/995724bffbcb942b487cb894aa60f6f84c381de5/target/linux/x86/base-files/lib/preinit/79_move_config#L9-L15

可以看到有两次挂载，第一次挂载内核分区`/dev/sda1`到`/boot`，第二次把`/boot/boot`挂载到`/boot` ，相当于把子目录覆盖到了父目录。 这样做的原因是为了kexec载入内核，所以这并不是一个BUG。

但强迫症用户看着确实别扭，可以修改下把`/dev/sda1`挂到`/tmp/boot`，然后`/boot`软链接到`/tmp/boot`即可。

修改代码如下：
``` shell
# Copyright (C) 2012-2015 OpenWrt.org

move_config() {
        local partdev parttype=ext4

        . /lib/upgrade/common.sh
        if export_bootdevice && export_partdevice partdev 1; then
                mkdir -p /tmp/boot
                part_magic_fat "/dev/$partdev" && parttype=vfat
                mount -t $parttype -o rw,noatime "/dev/$partdev" /tmp/boot
                if [ -f "/tmp/boot/$BACKUP_FILE" ]; then
                        mv -f "/tmp/boot/$BACKUP_FILE" /
                fi
                #mount --bind /boot/boot /boot
                ln -s /tmp/boot/boot /boot
        fi
}

boot_hook_add preinit_mount_root move_config
```
重新编译后目录：
```
root@OpenWrt:~# ls -l /
drwxr-xr-x    2 root     root           822 Dec 16 02:36 bin
lrwxrwxrwx    1 root     root            14 Dec 19 03:40 boot -> /tmp/boot/boot
drwxr-xr-x    6 root     root          2240 Dec 19 03:42 dev
drwxr-xr-x    1 root     root          3488 Dec 19 04:02 etc
drwxr-xr-x    3 root     root          3488 Dec 19 03:40 home
drwxr-xr-x    1 root     root          3488 Dec 19 03:40 lib
lrwxrwxrwx    1 root     root             3 Dec 16 02:36 lib64 -> lib
drwxr-xr-x    2 root     root             3 Dec 16 02:36 mnt
drwxr-xr-x    4 root     root          4096 Dec 19 03:40 overlay
dr-xr-xr-x   96 root     root             0 Dec 19 03:42 proc
drwxr-xr-x   16 root     root           236 Dec 16 02:36 rom
drwxr-xr-x    1 root     root          3488 Dec 19 04:17 root
drwxr-xr-x    2 root     root           870 Dec 16 02:36 sbin
dr-xr-xr-x   16 root     root             0 Dec 19 03:42 sys
drwxrwxrwt   24 root     root           680 Dec 19 04:00 tmp
drwxr-xr-x    1 root     root          3488 Dec 19 03:40 usr
lrwxrwxrwx    1 root     root             3 Dec 16 02:36 var -> tmp
drwxr-xr-x    4 root     root            67 Dec 16 02:36 www
```
```
root@OpenWrt:~# ls -l /boot/
lrwxrwxrwx    1 root     root            14 Dec 19 03:42 boot -> /tmp/boot/boot
drwxr-xr-x    2 root     root          4096 Dec 19 03:40 grub
-rw-r--r--    1 root     root       4956832 Dec 17 02:12 vmlinuz
```
系统截图：
[image]

WYC-2020, 2021-02-23:
@jiangzm ok已经修改

closed 2021-02-23 by WYC-2020