coolsnowwolf/lede #6042: 网卡状态显示的ipv6地址后缀不会跟随设置改变

IRN-Kawakaze, 2020-12-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：修改lan口的ipv6后缀之后（比如改为::ab:cc或者::192.168.1.1），网卡状态显示的ipv6地址仍为xxx:xxx:xxx::1/64，不会跟随设置改变。从ip addr和下发的地址来看，实际后缀已经改变并生效。
不管是公网地址（来自wan口的ipv6-pd）还是内网的fdxx地址，均是如此。

因为ipv4是显示实际地址而不是显示x.x.x.1，所以我认为这是个显示bug。如果是默认如此，还请大佬把这条issue当成建议。

### (2) 路由器型号和固件版本

A：OpenWrt R20.12.12 / LuCI Master (git-20.343.54716-6fc079f)

### (3) 详细日志

A：
[image]
[image]
[image]

hyird, 2020-12-18:
看来你不了解IPV6  ：：后面　表示中间为０　只剩最后一段了　　也就是说：：后面只能跟一段　再来一个：是不行的

IRN-Kawakaze, 2020-12-18:
> 看来你不了解IPV6 ：：后面　表示中间为０　只剩最后一段了　　也就是说：：后面只能跟一段　再来一个：是不行的

看来你对`::`有误解，`::`只是这个地址里连续最多的0的缩写，并不能说明后面只有最后一段。

底下的提示就写着“(例如：‘::1’或‘::1:2’)”，这个`::1:2`不就有两段？

别的例子就更多了，比如fe80开头的本地地址、根据eui-64生成的地址和`cloudflare-dns.com`的AAAA记录，这些地址的`::`后面都不止一段。

hyird, 2020-12-18:
抱歉 我记错了

hyird, 2020-12-18:
[image]
 连eui64都不行了 eui64我以前到时用过 是可以的 现在也不行了

closed 2021-04-06 by IRN-Kawakaze