coolsnowwolf/lede #6043: 为Netgear R7800路由器编译此固件时，报告v2ray-core-4.33.0 编译出错失败

moralrebuild, 2020-12-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
抽取最新的源代码，编译网件路由器R7800失败，V2Ray模块无法通过编译

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
在Ubuntu 18.04 LTS Docker container下进行编译。Newifi D2顺利编译成功。但同一个环境下对目标路由器Netgear R7800进行相似单线程编译工作，但始终被阻挡在V2Ray模块。

### (1) 具体问题
编译命令行：

make -j1 V=s

### (2) 路由器型号和固件版本

Netgear R7800

### (3) 详细日志
```
make[3]: Entering directory '/home/alice/lede/package/lean/v2ray'                                                                                                                                                                           rm -f /home/alice/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.33.0/.built                                                                                                                                        touch /home/alice/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.33.0/.built_check                                                                                                                                  CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/alice/lede/tmp/go-build" GO_MOD_CACHE_DIR="/home/alice/lede/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="v2ray.com/core" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO$SOURCE_ONLY="" GO_BUILD_PKG="v2ray.com/core/main" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/alice/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.33.0" GO_BUILD_DIR="/home/l$cke/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.33.0/.go_work/build" GO_BUILD_BIN_DIR="/home/alice/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.33.0/.go_work/build/bin/linux_arm" GO_$UILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/alice/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/share/gocode/src" GOOS="linux" GOARCH="arm" GO386="" GOARM="7" GOMIPS="" GOMIPS64="" CGO_ENABLED=1 $C="arm-openwrt-linux-muslgnueabi-gcc" CXX="arm-openwrt-linux-muslgnueabi-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-ma$=/home/alice/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.33.0=v2ray-core-4.33.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/alice/lede$staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -I/home/alice/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/include -I/home/alice/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gc$-8.4.0_musl_eabi/include/fortify -I/home/alice/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-$rror=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/alice/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.33.0=v2ray-core-4.33.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,$z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/alice/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/home/alice/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/home/alice/lede/staging_dir/toolchain-a$m_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/lib -L/home/alice/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib -znow -zrelro" GOPATH="/home/alice/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi$v2ray-core-4.33.0/.go_work/build" GOCACHE="/home/alice/lede/tmp/go-build" GOMODCACHE="/home/alice/lede/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/alice/lede/feeds/packages/lang/golang//golang-build.sh build -v -trimpath -ldflag$ "all=-buildid '1607744006' -linkmode external -extldflags '-L/home/alice/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/home/alice/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/home/alice/l$de/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/lib -L/home/alice/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib -Wl,-z,now -Wl,-z,relro'"    -installsuffix "v7"    -ldflags "-s $w -X v2ray.com/core.version=4.33.0 -X v2ray.com/core.codename=OpenWrt -buildid '1607744006' -linkmode external -extldflags '-L/home/alice/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/home/alice/lede/staging_di$/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/home/alice/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/lib -L/home/alice/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib -$l,-z,now -Wl,-z,relro'"
Finding targets

Building targets
google.golang.org/protobuf/internal/flags
golang.org/x/sys/internal/unsafeheader
go build golang.org/x/sys/internal/unsafeheader: no non-test Go files in /home/alice/lede/dl/go-mod-cache/golang.org/x/sys@v0.0.0-20201107080550-4d91cf3a1aaf/internal/unsafeheader                                                         github.com/lucas-clemente/quic-go/internal/logutils
go build github.com/lucas-clemente/quic-go/internal/logutils: no non-test Go files in /home/alice/lede/dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.19.2/internal/logutils
# google.golang.org/protobuf/internal/flags
../../../../../dl/go-mod-cache/google.golang.org/protobuf@v1.25.0/internal/flags/flags.go:17:21: undefined: protoLegacy
golang.org/x/crypto/chacha20
golang.org/x/crypto/poly1305
golang.org/x/crypto/sha3
# golang.org/x/crypto/poly1305                                                                                                                                                                                                              ../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/poly1305/poly1305.go:64:2: undefined: mac
# golang.org/x/crypto/chacha20
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:46:7: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:198:22: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:227:30: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:229:4: s.xorKeyStreamBlocks undefined (type *Cipher has no field or method xorKeyStreamBlocks)
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:235:23: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:237:12: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:239:16: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:249:12: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:251:4: s.xorKeyStreamBlocks undefined (type *Cipher has no field or method xorKeyStreamBlocks)
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:252:11: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:252:11: too many errors
golang.org/x/text/secure/bidirule
github.com/seiflotfy/cuckoofilter
github.com/lucas-clemente/quic-go/internal/protocol                                                                                                                                                                                         go.starlark.net/starlark
# golang.org/x/crypto/sha3
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/hashes.go:19:10: undefined: new224Asm
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/hashes.go:29:10: undefined: new256Asm
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/hashes.go:39:10: undefined: new384Asm
[... cut ...]
../../../../../dl/go-mod-cache/go.starlark.net@v0.0.0-20201014215153-dff0ae5b4820/starlark/int.go:91:19: i.get undefined (type Int has no field or method get)
../../../../../dl/go-mod-cache/go.starlark.net@v0.0.0-20201014215153-dff0ae5b4820/starlark/int.go:91:19: too many errors

Makefile:151: recipe for target '/home/alice/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.33.0/.built' failed
make[3]: *** [/home/alice/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/v2ray-core-4.33.0/.built] Error 2
make[3]: Leaving directory '/home/alice/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#1.45#1.60#0.99
```

AmadeusGhost, 2020-12-17:
网络问题，实在不行试试先在终端输入 `export GOPROXY=https://goproxy.cn` 再编译

AmadeusGhost, 2020-12-17:
update 下 feeds 看看 golang 相关的代码有没有更新

moralrebuild, 2020-12-18:
> 
> 
> update 下 feeds 看看 golang 相关的代码有没有更新

谢谢帮助。仅仅有golang之外的其它helloworld的更新。执行update以后依然同样错误。
后来删除掉git clone的整个工作目录，从头开始竟然顺利编译成功了。最近刚刚从编译官方openwrt的正式版转过来，接触到没有版本节点的编译，估计这种情况是经常发生的。

closed 2020-12-18 by moralrebuild